openscad/openscad #5791: Calculation error

AnReWi, 2025-03-31:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]
 - System: [e.g. Intel PC 64-bit, Raspberry PI 4]
 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

I wrote a small program to see how the wooden legs have to be shaped when 4 legs meet like a X under the table.
When I rotate the legs they do not meet exactly in the center.

$fn=100;

Height = 720;
Width = 650;

module Bone () {
  intersection () {
    H = Height/2;
    H2 = Height;
    W = Width/2;
    W2 = W * 0.25;
    A = 24;
    translate ([-W/2,W/2,H/2])
      cube ([W,W,H], center=true);
    translate ([-W2,W2,H/2])
      rotate ([A-.5,A,0])
        rotate ([0,0,45])
          cube ([45,45,H2], center=true);
  }
}

  for (X=[0:90:359]) {
    rotate ([0,0,X]) Bone ();
  }

[image]

kintel, 2025-03-31:
I suggest using the mailing list or IRC to discuss or get help on design issues: https://openscad.org/community.html

closed 2025-03-31 by kintel

jordanbrown0, 2025-03-31:
Yeah, e-mail or IRC is better for this sort of thing.  OpenSCAD isn't doing anything wrong here - you're just stacking up rotates and translates that don't quite add up the way you expect.  The way to diagnose this is to, start at your innermost cube().  Take a look at it.  Then apply a rotate (the rotate around Z) to it, and take a look at that.  Then the next rotate (the rotate around X), and so on.  If at any step it doesn't look like what you expect, stop and figure out why.  In particular, I suspect that you're expecting a rotation around X and a rotation around Y to stack to produce a diagonal rotation, and they won't.  You probably want to rotate around X and then rotate around Z.  Two-axis rotates are very hard to visualize; it's much easier to visualize one rotate at a time.

AnReWi, 2025-03-31:
Thanks a lot!
I have never had a probling like this before

I will test the way you write

Best regard
Anders
________________________________
Från: Jordan Brown
Skickat: Måndag, 31 mars 2025 19:37
Till: openscad/openscad
Kopia: AnReWi; Author
Ämne: Re: [openscad/openscad] Calculation error (Issue #5791)

[jordanbrown0]jordanbrown0 left a comment (openscad/openscad#5791)<https://github.com/openscad/openscad/issues/5791#issuecomment-2766921209>

Yeah, e-mail or IRC is better for this sort of thing. OpenSCAD isn't doing anything wrong here - you're just stacking up rotates and translates that don't quite add up the way you expect. The way to diagnose this is to, start at your innermost cube(). Take a look at it. Then apply a rotate (the rotate around Z) to it, and take a look at that. Then the next rotate (the rotate around X), and so on. If at any step it doesn't look like what you expect, stop and figure out why. In particular, I suspect that you're expecting a rotation around X and a rotation around Y to stack to produce a diagonal rotation, and they won't. You probably want to rotate around X and then rotate around Z. Two-axis rotates are very hard to visualize; it's much easier to visualize one rotate at a time.