openscad/openscad #5795: Request "Automatic Reload and Preview [on Save]" vs "Automatic Reload and Preview [on Keypress]"

DreadPirateShawn, 2025-04-02:
**Is your feature request related to a problem? Please describe.**

In a recent update (see additional context below for range), the "automatic reload and preview" behavior changed:
* previously = save triggered reload and preview
* now = any edit triggers reload and preview

My frustration is, I will often type several related lines, and then hit "save" to see the net change from the "before" version to the "after" version. However, with this new behavior, the preview thrashes around mid-typing.

Additionally, the console constantly spews errors on partially-complete lines of code as I'm typing them. Since I know "active typing" != "errors", this risks training users (me) to ignore errors.

**Describe the solution you'd like**

I would very much like to regain the option to automatically reload and preview **on save** rather than **on every keypress**.

**Describe alternatives you've considered**

I tried to disable "Automatic Reload and Preview" entirely, but this means that the preview NEVER updates by itself, not even on save.

**Additional context**

Version: `2025.04.01`
Specifically: `openscad-git r11467.7bdaeda-1` (I use Arch btw)
Behavior change occurred somewhere in this commit range: https://github.com/openscad/openscad/compare/681fff1..7bdaeda

kintel, 2025-04-02:
Tagging it as a bug for now, as it sounds like a regression

DreadPirateShawn, 2025-04-02:
Thanks, I didn't want to assume.

t-paul, 2025-04-02:
I think that this strictly is "reload on external file change", meant mostly for external editors on save. It's just not suppressed when saving from the internal editor.

jordanbrown0, 2025-05-14:
I think that logic may have originally been intended to trigger the first run after the file was loaded, but looking at the various sizes doesn't seem like the right way to do that.

closed 2025-05-22 by jordanbrown0