openscad/openscad #5802: Inner hole inside a cube is not visible in STL

asapir, 2025-04-03:
**Describe the bug**
Trying to generate a void inside a cube. When exporting the model into stl and open it in slicer program I'm getting solid cube w/o a void. 

**To Reproduce**
1. Use the following code
`
difference() {
    cube(10, true);
    cube(2, true);
}
`
2. export to stl
3. open in slicer program (e.g. Prusa slicer)
4. review the model in the slicer

**Expected behavior**
I expected to see the void of the inner cube when examine the different layers.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2021.01

t-paul, 2025-04-03:
Works fine here

[image]

Please attach the generated STL (zip it, or just name it `*.txt` so github upload accepts it).

gsohler, 2025-04-03:
of course you would have define !one! viewing angle for beeing able to sort the triangles,
but i failed to find the location where triangle set is passed to the opengl engine. maybe you can show me without having to track that down.

asapir, 2025-04-03:
I guess this is a bug in Prusa Slicer I'm trying to slice for SLA printer (Anycubic Photon)

[image]

t-paul, 2025-04-03:
Right, seem like a SLA mode issue then. Screenshot above is from PrusaSlicer-2.9.1 with MK4 InputShaper 0.8mm nozzle profile.

Closing, no OpenSCAD issue.

closed 2025-04-03 by t-paul

kintel, 2025-04-03:
> of course you would have define !one! viewing angle for beeing able to sort the triangles, but i failed to find the location where triangle set is passed to the opengl engine. maybe you can show me without having to track that down.

https://github.com/openscad/openscad/blob/7bdaeda6e05076637d9d5e8a2e591427cb2f8b19/src/glview/VBOBuilder.cc#L417

That code is still more convoluted that I'd like, but it's what we got so far.