openscad/openscad #5803: Color Problem in Rendered Object when object has a "difference" Hole

daackm, 2025-04-03:
**Describe the bug**
I am pleased to see color specifications propagated into rendered objects, but the color is not correct if the object has a hole.  See the code below, and the rectangle inside the difference.  It should be red, and it is before rendering, but after rendering it reverts to yellow.

**To Reproduce**
Steps to reproduce the behavior:
run code below

**Expected behavior**
Color in post-rendered object is rather new, and this is probably just a bug in recent code, but when I create documentation it is convenient to have individual parts color coded.

**Code reproducing the issue**
translate ([10, 0, 0])
   color([1,0,0])
cube([3, 30, 3.00]); 

difference ()
{
translate ([0, 0, 0])
   color([1,0,0])
cube([45, 30, 3.00]);  
translate ([0.75, 0.75, 0])
cube([43.50, 28.50, 3.00]);
}.

**Screenshots**

[image]

**Environment and Version info (please complete the following information):**
 -MS Windows 10,   2025.03.19  downloaded from openscad.org

**Library & Graphics card information**
OpenSCAD Version: 2025.03.19 (git 2c9f43c00)
System information: Microsoft Windows 10 (10.0.19045) x86_64 4 CPUs 7.94 GB RAM
User Agent: OpenSCAD/2025.03.19 (git 2c9f43c00) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof*, input-driver-dbus*, lazy-union*, vertex-object-renderers-indexing*, textmetrics*, import-function*, predictible-output
Application Path: C:/Program Files/OpenSCAD (Nightly)
Documents Path: C:\Users\Ackmann Family\Documents
User Documents Path: C:\Users\Ackmann Family\Documents
Resource Path: C:/Program Files/OpenSCAD (Nightly)
User Library Path: C:/Users/Ackmann Family/Documents/OpenSCAD/libraries
User Config Path: C:\Users\Ackmann Family\AppData\Local/OpenSCAD
Backup Path: C:/Users/Ackmann Family/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/Ackmann Family/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD (Nightly)\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
[... 340 lines cut ...]
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

**Additional context**
Add any other context about the problem here.

kintel, 2025-04-03:
Color rendering is only supported when using the Manifold backend: `Preferences->Advanced->3D Rendering->Backend`

closed 2025-04-03 by kintel

daackm, 2025-04-03:
> Even with Manifold, the behavior on difference is not what one might expect. One might think that drilling a hole through a colored object might leave the hole being the same color as the object, but that is not the case for either preview or Manifold. Instead, any "new" faces are the color of the negative object, which defaults to sort of a green.
> Tweaking the original example a little to get rid of the preview Z-fighting (see the FAQ <https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?> ), preview and render with Manifold both yield:
> image.png (view on web) <https://github.com/user-attachments/assets/e47537f8-a351-43ce-bd42-b7438e415b8e>
> If you want the hole to be red, you need to subtract a red object.
> Even with Manifold, the behavior on difference is not what one might expect. One might think that drilling a hole through a colored object might leave the hole being the same color as the object, but that is not the case for either preview or Manifold. Instead, any "new" faces are the color of the negative object, which defaults to sort of a green.
> Tweaking the original example a little to get rid of the preview Z-fighting (see the FAQ <https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?> ), preview and render with Manifold both yield:
> image.png (view on web) <https://github.com/user-attachments/assets/e47537f8-a351-43ce-bd42-b7438e415b8e>
> If you want the hole to be red, you need to subtract a red object.

Thanks, Jordan.  I’ll play with it the next few days.
 
Do you have a favorite description of what “Manifold” is?  If not, I can always search for information myself.
 
My “thing” is to write MS Windows software the help model railroaders design buildings.  They fill out forms and my executables write OpenSCAD code to deliver STLs of what they want, things like walls, roofing, windows and such.  Model railroaders are typically older guys adverse to technology, and I am trying to get them to elevate their game by lowering the learning curve.  Visit http://daackm.github.io to see what I do.  If Manifold is something that my users will have to load into their copy of OpenSCAD, that might be a deal breaker for my user base, and I’ll have to think of something other than color to differentiate the different pieces of windows I am trying to draw.
 
I go back to the early days of computer graphics and 3D printing reminds me so much of writing software drivers for pen plotters and 35mm slide film recorders.  Maybe I’ll see 24 bit color in 3D printing before I hang it all up.  Déjà vu.
 
Dave Ackmann

jordanbrown0, 2025-04-03:
Manifold is a rendering library that OpenSCAD now uses as its primary rendering engine.  It replaces the older and much slower CGAL.  It is not a separate component.

Mostly the change is transparent to the user, except that Manifold is much faster.  The Manifold support retains color through the rendering process, where CGAL lost color information.

daackm, 2025-04-03:
Thanks, that was a pretty clear explanation.  I’m running 2025.03.19, but when I changed the colors in a difference so that both objects were red, the object still was rendered in yellow.  I am sure I am missing something . . .
 
Dave Ackmann

kintel, 2025-04-03:
You currently have to enable the Manifold backend by selecting "Manifold" in Preferences->Advanced->3D Rendering->Backend.

This will eventually become the default setting, but we're still ironing out the last few issues.

daackm, 2025-04-04:
Well, look at that!  This is going to be quite useful.
 
I remember doing something about a year ago to speed up rendering and it must have been getting rid of GCAL, but performance dropped several months ago after I downloaded a new version from the Overnights, and you got me moving along faster and at the same time fixing my color issue!  Thanks for spending the time on my problem.
 
Dave Ackmann

jordanbrown0, 2025-04-04:
Sorry, I thought we'd switched the default.