openscad/openscad #5805: Error and a Worning

ymolcho, 2025-04-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]
 - System: [e.g. Intel PC 64-bit, Raspberry PI 4]
 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.
After I make any change to the program these lines appear on the console:
"ERROR: boost::filesystem::create_directory: The system cannot find the path specified 
WARNING: Cannot create backup path:"
After saving and reopening the program, no errors appeared.

kintel, 2025-04-03:
Possibly related to #5425

t-paul, 2025-04-03:
@ymolcho Please actually fill out the form with related information, it's fine to remove questions like "screenshot" that do not apply but version and library info are mandatory.

ymolcho, 2025-04-04:
I'm sorry. I don't know how to use a VPN drive, so I attached the Dummy files here.
J. Molcho

ymolcho, 2025-04-14:
A

[Errors demo1.pdf](https://github.com/user-attachments/files/19730467/Errors.demo1.pdf)

[Errors demo2.pdf](https://github.com/user-attachments/files/19730482/Errors.demo2.pdf)

[Errors demo3.pdf](https://github.com/user-attachments/files/19730483/Errors.demo3.pdf)

Attached 3 screenshots showing the bug. 
Errors occurred after I changed the r=2 to r=2.5 and pressed F5.

J. Molcho

t-paul, 2025-04-14:
From the comment, I assume this was not supposed to be closed... reopening.

ymolcho, 2025-04-14:
Attached Lib Info
J. Molcho  

 
 

Reopened #5805.

jordanbrown0, 2026-05-08:
The reason that the error didn't happen earlier is that the file had been loaded and hadn't been modified, so there was no reason to make a backup copy.

jordanbrown0, 2026-05-08:
I don't see Library Info attached.

Please open Help/Library Info, and copy-and-paste the output into a comment on this issue.  It should work to reply to the e-mail, but Github seems to generally behave better if you use the web user interface:  https://github.com/openscad/openscad/issues/5805

The message suggests that there is something wrong with Documents\OpenSCAD\backups,  (It would be good if it did a better job of saying what path it was **trying** to use.)

I see that you've put your desktop into C:\OneDriveTemp\OneDrive\Desktop, suggesting that you have mucked with the standard Windows configuration for your "home" folders.  If you haven't appropriately adjusted all of the Windows settings associated with those folders, it's unsurprising that bad things happen.  We believe what Windows says when we ask it where your Documents folder is.