openscad/openscad #5808: Assertion Failure when doing intersection with roof of diffed object

julvr123, 2025-04-05:
**Describe the bug**
Getting assertion failure when rendering diffed circles, with roof:

ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: ss_circle.has_on(sp_0)
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h
Line: 286

**To Reproduce**
The code below diffs a circle with three others.  Only the last of the diffed circles seems to generate the problem.   If you comment out the line `translate([cos(240)*36, sin(240)*36,0]) circle(r=24);`, it will render an output.  If you leave the line in, it does not.   It seems to occur when the angle is > 215..   Also, if you remove the cube_coords(..) line in the _bevel_cap_section, then it also produces output.

**Expected behavior**
Expecting render to produce output similar to preview.  (Render creates assertion failure then exits)

**Code reproducing the issue**

module cube_coords(tlf=[0,0,0], blb=[10,10,10]) {
    pos = ( [ min(blb[0],tlf[0]), min(blb[1],tlf[1]), min(blb[2],tlf[2]) ] );
    size = ( [ abs(blb[0]-tlf[0]), abs(blb[1]-tlf[1]), abs(blb[2]-tlf[2]) ] );
    translate(pos) cube(size=size);
}

module _bevel_cap_section(h, w, method="straight")
{
    dsz=1000;
    intersection() {
        scale([1,1,h/w]) roof(convexity=6,method=method) children();
        cube_coords([-dsz,-dsz,-0.01],[dsz,dsz,h]);
    }
} 

_bevel_cap_section(2,2)
{
    difference() {
        circle(r=24);
        translate([cos(0)*36, sin(0)*36,0]) circle(r=24);
        translate([cos(120)*36, sin(120)*36,0]) circle(r=24);
        // if I comment out this line, it works... it seems angles greater than 215 are causing problems:
        translate([cos(240)*36, sin(240)*36,0]) circle(r=24);
    }
}
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: AMD
 - OpenSCAD = Version 2025.01.06 (git 92a13b4f0)

kintel, 2025-04-05:
Note: roof() has fallen out of maintenance and needs significant love to survive: https://github.com/openscad/openscad/labels/Feature%3A%20Roof

coryrc, 2025-11-13:
Err, the appropriate label is "Fixed by manifold"