openscad/openscad #5810: Main window title isn't updated on save.

kintel, 2025-04-07:
**Describe the bug**

1. Launch OpenSCAD
2. Create a minimal design, e.g. `sphere();`
3. Save -> enter a new filename, e.g. `sphere.scad`
4. Observe that the tab changes its title to `sphere.scad`
5. Bug: The main window does _not_ change its title, and still shows `Untitled`

**Screenshots**

[image]

**Environment and Version info (please complete the following information):**
 - OS: macOS 14 @ M1
 - Latest master

Sparsh-N, 2025-04-08:
Error also occurs on Windows

[image]

damienmarchal, 2025-04-08:
I noticed the title window bug not being updated yesterday, 

The last commit of https://github.com/openscad/openscad/pull/5809 contains a fix, I will extract and open a dedicated PR.

damienmarchal, 2025-04-08:
Ok, so the segfault in FileDialog seems related to mimalloc, is this something known ?

closed 2026-05-08 by jordanbrown0