openscad/openscad #5819: arithmetic on ranges?

jordanbrown0, 2025-04-12:
**Is your feature request related to a problem? Please describe.**
I find myself wanting to multiple a range by a scalar.  (I want to describe the range in scale units, and then multiply by a scale.)

**Describe the solution you'd like**
```
range * number -> range : [ number*range.begin : number*range.step : number*range.end ]
```

**Describe alternatives you've considered**
```
function rmul(r, m) = [ m*r.begin : m*r.step : m*r.end ];
```

**Additional context**
Division similarly.  Addition and subtraction would modify the start and end, but not the step.

t-paul, 2025-04-12:
The description does not match the formula regarding `step`.

jordanbrown0, 2025-04-12:
How not?  (Was my made-up syntax for describing an arithmetic operation a problem?  The intent was to say that multiplying a range by a number would yield a range, using the range specification in the []s.)

Multiplying `[1:2:9]` (1, 3, 5, 7, 9) by 3 should yield `[3: 6: 27]` (3, 9, 15, 21, 27).

Another way to represent it would be:
```
function rmul(r, m) = [ for (v = r) r*m ];
```

t-paul, 2025-04-12:
Oh, my fault. I missed the "Add/Sub" in that sentence mentioning "modify the start and end, but not the step".

jordanbrown0, 2025-04-12:
To be clear:  this can easily be done in user space, but as a "natural" interaction between the two types it seems plausible as a built-in feature.