openscad/openscad #5824: syntax extension for variables used in libraries

yahbluez, 2025-04-13:
The feature is not a solution for a problem but an simple extension of the syntax to make the use of variables in included libs that can be changed before the include easier for typing and reading.

The standard way to declare a variable, global in the scope of an imported lib, but over writable in the script that did the include looks like this:

var_name = is_undef(var_name) ? value : var_name;

My idea to make that more readable is the following syntax:

var_name := value;

This solution would not collide with any existing part of the syntax but reduce the script size and the redundancy.

No big deal just an idea.

kintel, 2025-04-13:
What do you think about this existing suggestion to address the same issue? https://github.com/openscad/openscad/pull/5383

jordanbrown0, 2025-04-13:
Note that the designed-in way to have a variable where an included library sets a default and the main program can override the default is for the library to set the variable, and the main program to reset the variable *after* the include<>.

(Yes, this really works, including when the library has other variables derived from this one.  Try it.)

jordanbrown0, 2025-04-13:
Please clarify how you expect this
```
var_name = is_undef(var_name) ? value : var_name;
```
to be used.  Is this what the library does?  Is it what the main program does?  Are you expecting the main program's assignment to be before or after the include?

jordanbrown0, 2026-05-08:
I think this is addressed by the designed-in mechanism that allows the main program to override variable settings in an `include<>` file.