openscad/openscad #5827: backtrace missing entries

adrianVmariano, 2025-04-14:
I recently tried doing some development with a nightly build,2025.02.25.snap, and have been running into a significant annoyance that when there is an error, sometimes the backtrace omits the specific line where the error actually occurred. 

For example, in the dev version I got this:

ERROR: Assertion 'is_2d_transform(transform)' failed: "Transforms is 3D and acts on Z, but points are 2D" in file libraries/BOSL2/transforms.scad, line 1569
TRACE: called by '_apply' in file libraries/BOSL2/transforms.scad, line 1549
TRACE: called by 'path_sweep' in file libraries/BOSL2/skin.scad, line 1943
TRACE: assignment to "trans_scale" in file libraries/BOSL2/skin.scad, line 1943
TRACE: called by 'path_sweep' in file ttest1.scad, line 20

But the error was not on any listed line.   To find the error I ran the stable version which gave me this:

ERROR: Assertion 'is_2d_transform(transform)' failed: "Transforms is 3D and acts on Z, but points are 2D" in file libraries/BOSL2/transforms.scad, line 1569 
TRACE: called by '_apply' in file libraries/BOSL2/transforms.scad, line 1549 
TRACE: called by 'apply' in file libraries/BOSL2/skin.scad, line 2151 
TRACE: called by 'path_sweep' in file libraries/BOSL2/skin.scad, line 1943 
TRACE: called by 'path_sweep' in file ttest1.scad, line 20 
 
And yep, line 2151 was the problem.    

At the moment I don't have an example to demonstrate this.  (I was debugging my code, not trying to preserve the bad code...)   It's also not obvious how easy it will be to produce any kind of minimal example.

kintel, 2025-04-14:
Without an example, this isn't very actionable, but let's keep this open for a while, and hope that someone can come up with a reproducible example.

jordanbrown0, 2025-04-15:
Maybe tail call elimination?

jordanbrown0, 2025-04-15:
Are the affected components modules, or functions?

adrianVmariano, 2025-04-15:
Functions, I think.

adrianVmariano, 2025-04-15:
[BOSL2.zip](https://github.com/user-attachments/files/19757864/BOSL2.zip)

With the help of the undo history of my editor I managed to recreate the error case.   I've attached a buggy BOSL2 snapshot.  Code that triggers the behavior:

```
include<BOSL2/std.scad>
path_sweep(circle($fn=30,r=1), path3d(circle($fn=64,r=3)),closed=true, texture="bricks_vnf");
```

In the stable version I get the trace shown below where line 2151 (which is the line that causes the error) appears on the list:

```
ERROR: Assertion 'is_2d_transform(transform)' failed: "Transforms is 3D and acts on Z, but points are 2D" in file libraries/BOSL2/transforms.scad, line 1569 
TRACE: called by '_apply' in file libraries/BOSL2/transforms.scad, line 1549 
TRACE: called by 'apply' in file libraries/BOSL2/skin.scad, line 2151 
TRACE: called by 'path_sweep' in file libraries/BOSL2/skin.scad, line 1943 
TRACE: called by 'path_sweep' in file ttest1.scad, line 4 
```

but in the dev snapshot 2025.03.02 the erroneous line 2151 is missing from the trace:
```
ERROR: Assertion 'is_2d_transform(transform)' failed: "Transforms is 3D and acts on Z, but points are 2D" in file libraries/BOSL2/transforms.scad, line 1569
TRACE: called by '_apply' in file libraries/BOSL2/transforms.scad, line 1549
TRACE: called by 'path_sweep' in file libraries/BOSL2/skin.scad, line 1943
TRACE: assignment to "trans_scale" in file libraries/BOSL2/skin.scad, line 1943
TRACE: called by 'path_sweep' in file ttest1.scad, line 4

```

I'm pretty sure I've seen this also in 2025.02.25, but not 100% certain.    You can also run the example as a function instead of module like this:

```
include<BOSL2/std.scad>
x=path_sweep(circle($fn=30,r=1), path3d(circle($fn=64,r=3)),closed=true, texture="bricks_vnf");
```

and the output is a bit different, but line 2151, where the error is actually happening, is still missing.

jordanbrown0, 2025-04-15:
Thanks.  I'll take a look at this, but it probably won't be until Monday.

curtmcd, 2025-04-23:
I bisected the above test case to find this problem to be introduced by this commit not long after 2021.01 was released:

```
commit 2f56db5584c49b48ea57209169d6447593cc4810
Author: Ruud Koolen <redlizard@redlizard.nl>
Date:   Fri Apr 9 16:22:51 2021

    Separate context-dependent module lookup from module instantiation logic.

```

adrianVmariano, 2025-04-23:
Wow, so nobody has noticed this for 4 years.  Thanks for tracking that down.