openscad/openscad #5828: polygon() created 3D shape instead of 2D

akovez, 2025-04-17:
When I use polygon on a set of points, I get 3d shape instead of 2d shape.

Reproduce
1. Create and preview file:

```
module sector(radius, angles) {
  points = [
    for(a = [angles[0]:1:angles[1]]) [radius * cos(a), radius * sin(a)]
  ];
  polygon(points = concat([[0, 0]], points));
}

sector(radius = 4, angles = [0, 90]);
```
2. You will see 3d shape in preview.

[image]

**Expected behavior**
Expecting 2d shape polygon

Environment and Version info 
OpenSCAD version 2025.04.11 from aur
arch linux, intel pc 64bit

akovez, 2025-04-17:
Same with any other polygon 

[image]

gsohler, 2025-04-17:
This is default  behaviour. In Preview all 2D shapes become a thickness of
1.
It would make sense to reduce this width to easily distinguish the 2D
extent from the height.

akovez, 2025-04-17:
okay, thank you for prompt reply

jordanbrown0, 2025-04-17:
There are a number of ... odd ... behaviors associated with preview of 2D objects, because of the way that they are simulated using 1-unit-tall 3D objects.

Ideal would be to move to using "true" 2D objects, as can be observed in incomplete polyhedra, but they introduce other problems.