openscad/openscad #5831: Difference not working with complex shaper

basensax, 2025-04-18:
**Describe the bug**
I use the difference action in version 25.03.19 with a complex shape and it does nothing.

**To Reproduce**
Steps to reproduce the behavior:
1. Load the sample code
2. comment and uncomment the line difference in module reed in line 53
3. See the difference
In version 2023.24.05 it works OK

**Expected behavior**
The lowest picture added shows the result in version 2023.24.05

**Code reproducing the issue**
//-----------------------------------------------------------------------------------------
// RIETEN.SCAD
//-----------------------------------------------------------------------------------------

//-----------------------------------------------------------------------------------------
use <../library/naca4.scad>;
use <../library/ShortCuts.scad>;
use <../library/splines.scad>;
use <../library/Naca_sweep.scad> ;
use <../library/skin.scad> ;
//----------------------------------------------------------------------------------------
$fn			=180;
//------------------------------------------------------------------------------------------
reed = [ "CA", 			16.3,	18.9,	90.0,	3.2,		50,		3.7];
models = ["jazz clarinet",			0.3,	2,		0.3];
//------------------------------------------------------------------------------------------
txt_type				= 	reed[0];
txt_model				=	models[0];
w_btm 					=	reed[1];
w_tip					=	reed[2];
l_reed					= 	reed[3];
d_btm_mx				=	reed[4];			// maximale dikte van het riet
d_btm_mn				=	d_btm_mx/2;			// dikte aan zijkbat van het riet
tip_curve				=	reed[6];
l_snede				=	reed[5]*reed[3]/100;
d_tip					=	0.40;						//DIKTE VAN DE TIP
powery				=	models[2];			// factpr tussen 0.5 en 2 die de kromming van de snede wijzigt in y richting
powerx					= 	models[1];			// factor tussen 0.5 en 2 die de kromming van de snede wijzigt in z richting
mid_factor 			= 	models[3];			// factor die de zwaarte van het accent op het midden bepaalt
---------------------------------------------------------------------------------------------------

 render() reed();

module reed()
{
difference()
{
	 intersection()
	  {
		union()
		{
			basic_shape();		// step 1: 	combine rectangle
			tip_ellips();				// 			with tip ellips
		}
		top_curve();				// step 2	shape the topcurve
	  }
 	cutting_curve();
	tekst_uitsnede();				// step 4 	remove the text
}
}
//-------------------------------------------------------------------------------------------------------
// create side shape
//---------------------------
module basic_shape()
{
	 translate([0,0,0]) linear_extrude(2*d_btm_mx)
	 polygon(points=[[0,0.5*w_btm],[l_reed-tip_curve,0.5*w_tip],[l_reed-tip_curve,-0.5*w_tip],[0,-0.5*w_btm]]);
}
//------------------------------------------------------------------------------------------------------
// create cylinder for top curve
//-----------------------------
module top_curve()
{
  	S					= w_btm/2;
  	d					=d_btm_mx-d_btm_mn;
 	cane_diameter	= 2*S /sin(2* atan(d/S));
 	translate([0,0,-cane_diameter/2+d_btm_mx])
	rotate([0,90,0]) cylinder(l_reed, d=cane_diameter);
}
//----------------------------------------------------------------------------------------------------
// create tip ellipse
//-----------------------------
module tip_ellips()
{
  translate([l_reed-tip_curve,0,0]) 
  resize([tip_curve*2,w_tip,2*d_btm_mx]) cylinder(h=2*d_btm_mx,d=w_tip);
}
//---------------------------------------------------------------------------------------------------
// create cutting curve
//--------------------------
module cutting_curve()
{
 //---------------straight cutting applies to the sides-------------------------------
   h_snede						= d_btm_mx-d_tip;					// diepte van de snede
   r_snede						= straal(h_snede,l_snede);					// diameter of the cutting
//---------------lineair cutting applies for about 50% to the middle
   rc							= h_snede/l_snede;
   step 							= l_reed/20;
   cutting_shape = [ for (p= [ 90:2: 270])
		  let( l_snede	= l_snede)							// totale lengte van de snede
		  let (h_snede	= d_btm_mx-d_tip)					// diepte van de snede
		  //													 voor iedere p de x coordinaat
		  // de x richting wordt met een factor gelijk aan powerx vervormd
		  let (x_snede	= (p<180)? l_reed + l_snede*pow(abs(sin(p)),powerx):
		  						l_reed - l_snede*pow(abs(sin(p)),powerx))
     		  let( dx 	= abs(l_reed-x_snede))
		  //													voor iedere p de z coordinaat
   		  let (z_snede = d_btm_mx + h_snede*cos(p))
		  // maak een geleidijke overgang van een rechte lijn van tip naar midden 
		  // en de ellipsvormige snede
   	  	  let (mid_z			= max(z_snede+0.01,mid_factor*(d_tip + rc*dx) + (1-mid_factor)*z_snede))
 	        let (delta			= mid_z-z_snede)
		  let (Szc		      = (w_btm + (w_tip-w_btm)*x_snede/l_reed)/2)  
   		  let( rc	= h_snede/l_snede)							// relative distance in cutting   		
// --------------------------------------------------------------------------------------
		  [
			  x_snede, 				// x coordinate
			  z_snede,				// verticaal
			  Szc,				// breedte straal
			  delta,					//breedte
		  ]
		  ];	
		  knife = gen_cutting(cutting_shape,$fn);		// genereer de objectdata
		  sweep(knife);							// genereer object
}

function cutting_(r,h,z,N) = 
    [   // input:   radius r and the number of vectors 
        // output:  N vectors on edge of circle with radius r 
        for(i=[N-1:-1:0]) 
            // internal help variabel w = angle 
            let(w = i*360/N) 
	      let(x= (w>180)? 20: z+h*pow(sin(w),powery))
	  	//let(x=  z+h*pow(sin(w),powery))
	  	let(y =-r*cos(w))
            [x,y]
    ];

function gen_cutting(S, N) =
    [ for (i=[0:len(S)-1])
	let (x = S[i][0])		//stap over lengte van het riet
	let (z = S[i][1])		//beginpunt van snede aan zijkant
	let( r= S[i][2])		//breedte van de snede 
      let(h=S[i][3])		// hoogte van de boog
      let(dat = Ry_(90,vec3D(cutting_(r+0.1,h,z,N))))
    	   T_(x, 0, 0, dat)];
//------------------------------------------------------------------------------------------------------------
// STRAAL
//------------------------------------------------------------------------------------------------------------
function straal(d,s) = s/sin(2*atan(d/s));
//------------------------------------------------------------------------------------------------------------
// TEKST
//-----------------------------------
module tekst_uitsnede()
{
  tekst = str( txt_type,d_tip*1000,10*powerx,10*powery,10*mid_factor);
  translate([3,0.5,d_btm_mx-1]) linear_extrude(1.1, convexity =10) text(text=tekst,size=4);
  translate([3,-6,d_btm_mx-1]) linear_extrude(2, convexity=1) text(text=txt_model,size=5);

}

**Screenshots**

[image]

**Environment and Version info (please complete the following information):**
 - Windows 11
 - System: 2023 version Intel(R) Core(TM) i3-8145U CPU @ 2.10GHz   2.30 GHz
 - System: 2025 version instellingen Intel(R) Core(TM) i7-6700T CPU @ 2.80 GHz 
 - both versions installed from openscad.org

**Library & Graphics card information**
Nacasweep

**Additional context**
I could not find an installation for the 2023 version anymore.
The 2021 versions are too slow.

basensax, 2025-04-18:
Also cutting out the text from the shape is not perfect. Sometimes I miss a part of the text.

nophead, 2025-04-18:
Why have you got convexity=1 when extruding text, is it a typo?

kintel, 2025-04-18:
Could you provide a git repo or an archive with all the files and libraries needed?

basensax, 2025-04-18:
[bug report.zip](https://github.com/user-attachments/files/19815547/bug.report.zip)

kintel, 2025-04-18:
The reason the difference is broken is because the `cutting_curve()` object is "inside out". I'm not familiar with the libraries used, but you need the polygon winding to be inverted. This can be debugged in Throwntogether mode (F12). Any pink surfaces is a sign of issues.

basensax, 2025-04-26:
Hell Kintel, thanks for your reaction. I was not awate of the inside out problem. I was not familiar with that and will take a look at it. I assume that will fix the issue for me. Glad that is it not a bug in Openscad but my bug ;-)

closed 2025-05-05 by kintel