openscad/openscad #5834: Assertion in CGAL when computing a difference of offset linear extrusions of projections

mbuzdalov, 2025-04-19:
**Describe the bug**
The following error happens during rendering (but not preview):

```
ERROR: CGAL error in CGALUtils::applyBinaryOperator difference: CGAL ERROR: assertion violation!
Expr: itl != it->second.end()
File: /usr/include/CGAL/Nef_3/SNC_external_structure.h
Line: 1078
```

**To Reproduce**
Render the attached code.

**Expected behavior**
This should generate a shape identical to the preview.

**Code reproducing the issue**
```
module funnyCyl(off) {
    rotAngle = 10;
    rBwd = 252.5;
    rFwd = 17;
    h = 90;
    
    translate([62.4, 0, 14])
    rotate([0, rotAngle, 0])
    translate([rFwd, 0, -h - off / 100])
    cylinder(h = h + 2 * off / 100, 
            r1 = rBwd - off / sin(rotAngle), 
            r2 = rFwd - off / sin(rotAngle), 
             $fa = 1, $fs = 0.1);
}

module fwdBlockShape() {
    projection(cut = false)
        translate([6, 0, 0])
        rotate([0, 90, 0])
        intersection() {
            funnyCyl(0);
        translate([0, 0, -6])
            linear_extrude(20, convexity = 2)
            polygon([
                [30, -30.1],
                [60.4, -38.5],
                [62.4, -38.5],
                [62.4, +38.5],
                [60.4, +38.5],
                [30, +30.1]
            ]);
        }
}

module fwdBlock() {
    difference() {
        linear_extrude(30, scale=[18.2/20, 50/77])
            fwdBlockShape();
        translate([0, 0, -0.1])
            linear_extrude(30.2, scale=[(18.2-0.8)/(20-0.8), (50-0.8)/(77-0.8)])
            intersection() {
                offset(r = -0.4) fwdBlockShape();
                translate([1, -38.5 + 1.2]) square([20, 77 - 2.4]);
            }
    }
}

fwdBlock();

```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: Gentoo Linux
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2024.12.30, installed from Portage

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

The potentially relevant pieces of `openscad --info`:
```
OpenSCAD Version: 2024.12.30 (git 756e080c7)
System information: Linux 6.12.16-gentoo #1 SMP PREEMPT_DYNAMIC Tue Feb 25 15:32:48 GMT 2025 x86_64 Gentoo Linux 4 CPUs 15.33 GB RAM
User Agent: OpenSCAD/2024.12.30 (git 756e080c7) (Linux x86_64; Gentoo Linux)
Compiler: GCC "14.2.1 20241221" 64bit
MinGW build: No
Debug build: No
Boost version: 1_85
Eigen version: 3.4.0
CGAL version, kernels: 6.0, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.6.0
Manifold version: 3.0.1
```

**Additional context**
Add any other context about the problem here.

kintel, 2025-04-19:
Works for me on 2025.04.18 - try upgrading to a more recent snapshot and see if that helps.

kintel, 2025-04-19:
PS! If you switch to the Manifold backend, it will will also render significantly faster: Preferences->Advanced->3D Rendering->Backend

mbuzdalov, 2025-04-19:
Not too sure if updating would be easy enough for me right now, but switching to Manifold actually solves it!

kintel, 2025-04-19:
Sounds good! I’ll assume this is fixed. Feel free to reopen if this issue persists with a recent build.

closed 2025-04-19 by kintel