openscad/openscad #5846: Preview Missing Faces Under Certain Conditions

degroof, 2025-04-22:
**Describe the bug**
Certain combinations of CSG operations result in a preview that is incomplete. Faces are missing or transparent.
**NOTE: This happens only in preview. Render does not exhibit this issue.**

**To Reproduce**
Steps to reproduce the behavior:
1. Load the attached previewTest.scad
2. Preview (F5)

**Expected behavior**
The preview should show a square tube with a circular hole through it.

**Code reproducing the issue**
`difference()
{
  linear_extrude(80)
  difference()
  {
    square(80);
    offset(-2)
    square(80);
  }
  translate([0,0,40])
  rotate([0,90,45])
  cylinder(d=20,h=200);
}`

**Screenshots**

[image]

[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 11 Pro (build 26100.3476)
 - System: Lenovo P14s (AMD Ryzen 7 Pro w/ Radeon 780M)
 - OpenSCAD Version 2025.04.20 from https://openscad.org/downloads.html

**Library & Graphics card information**
OpenSCAD Version: 2025.04.20 (git 9d22f1fce)
System information: Microsoft Windows 11 (10.0.26100) x86_64 16 CPUs 27.67 GB RAM
User Agent: OpenSCAD/2025.04.20 (git 9d22f1fce) (Microsoft Windows 11 (10.0.26100) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: C:/Program Files/Openscad
Documents Path: C:\Users\steve\OneDrive\Documents
User Documents Path: C:\Users\steve\OneDrive\Documents
Resource Path: C:/Program Files/Openscad
User Library Path: C:/Users/steve/OneDrive/Documents/OpenSCAD/libraries
User Examples Path: C:/Users/steve/OneDrive/Documents/OpenSCAD/examples
User Config Path: C:\Users\steve\AppData\Local/OpenSCAD
[... 303 lines cut ...]
GL_KHR_shader_subgroup
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_timeline_semaphore
GL_OES_EGL_image
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
WGL_EXT_swap_control

**Additional context**
This seems to happen most often when a difference() operation is performed on an object with inner and outer walls (e.g. a tube, pipe or container).

t-paul, 2025-04-22:
Please use `linear_extrude(80, convexity = 2)`. See also this [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?).

closed 2025-04-22 by t-paul

degroof, 2025-04-22:
Ah, of course! It never occurred to me that it was a convexity issue. Thanks!