openscad/openscad #5848: Color support on export in 3MF format

FastFrench, 2025-04-23:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]
 - System: [e.g. Intel PC 64-bit, Raspberry PI 4]
 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

jordanbrown0, 2025-04-23:
This needs the form filled in to describe what it's actually asking for, but I suspect that it is a duplicate of #1608.

t-paul, 2025-04-23:
3MF has color support in the dev snapshots.

FastFrench, 2025-04-23:
Sorry for the confusion. I started this bug report, bug looks like I've done a mistake and post the empty template instead. As I tried to edit it, was interrupted and lost my data. 
So I've edited the content, so it can be as clear as possible. 
Thank you to reopen the bug please.

closed 2025-04-23 by jordanbrown0