openscad/openscad #5849: Color support on export in 3MF format (v2025.4.21)

FastFrench, 2025-04-23:
First I would like to apologize for my previous unfilled bug report. it was a mistake, and lost all the content as I was editing it. So here is the right filled content. 

**Describe the bug**
When I export a 3MF with last Nightly build (2025.04.21), but also with several previous versions, and import that 3MF file in OrcaSlicer or CrealityPrint, the full scene if just all grey. All the colors are lost in the process. 
Note that my object displays properly in OpenSCAD (with 4 colors in it), both with F5 and F6. 

**To Reproduce**
- OpenSCAD 2025.04.21
- Settings used: 

[image]

[image]
 
- As my working model was a bit complicate (850 lines for now), I tried with a very simple and basic one:
```
module partie(x, y, z) {
    translate([x, y, z])
    cube(10);
}

color("red") partie(0, 0, 0);
color("green") partie(15, 0, 0);
color("blue") partie(0, 15, 0);
```
[image]

- I export the .3MF file with the folowwing settings (same result with 'Export colors as Base Materials'): 
[image]

- I open the file in OrcaSlicer 2.3.0, the all scene is just grey, no color (same with CrealityPrint 6.1)
[image]

- To be sure that I don't have a problem with my OrcaSlicer configuration, I tried to open a color model from a .3ML file I dowloaded: it displays properly with all colors as expected (no matter the filaments really available at that time). 

**Expected behavior**
According to what I read from the community and the settings added in OpenSCAD, I expected to be able to get the colored model in OrcaSlicer, and not just grey objects (geometry only, with no colors). 

**Code reproducing the issue**
See up

**Screenshots**
See up

**Environment and Version info (please complete the following information):**
 - OS: Windows 10 Pro
 - PC 64-bit, AMD Ryzen 5 2600 Six-Core, 64GB
 - OpenSCAD Version: any recent versions. Last tried with:
[image]

**Library & Graphics card information**
No reason to think it may be a hardware issue. 

**Additional context**
N/A

t-paul, 2025-04-23:
Slicers don't use actual 3MF colors. They use different ways of handling colors that do not follow the official spec. Nothing OpenSCAD can do. Try uploading the generated 3MF file to Printables, it shows the colors there. 

Example: https://www.printables.com/model/1020666-openscad-color-3mf-test

It should still be possible to handle the different objects separately, assigning extruders.

t-paul, 2025-04-23:
See also https://github.com/bambulab/BambuStudio/issues/4292

closed 2025-04-23 by t-paul

FastFrench, 2025-04-23:
OK, I see. But then why can I find so many .3mf files on the Internet that have colors and import very well with OrcaSlicer ?

t-paul, 2025-04-23:
I assume those are saved from OrcaSlicer which obviously knows it's own special handling.

FastFrench, 2025-04-23:
OK, you must be right. 
Now I don't understand OpenSCAD position about that matter: why do you add and communicate about color support in OpenSCAD, if your position is "Can't work because of all the slicers". 
Is there much interest in having colors for for display and not printing ? 
Clearly the .3mf models you can find with colors do work with all slicers I've try, so it's hard to believe that slicers are wrong and are all able to read those wrong files but not the good one OpenSCAD produces...

t-paul, 2025-04-23:
> And the fact you closed the issue after only 6 minutes (took me much longer to try to make it clear) let me think it's maybe kind of taboo subject?

Look up who implemented 3MF support and who commented on that BambuSlicer issue.

Why do you think I have an example file uploaded to Printables ready to link?

Maybe consider that's a topic I've been fighting with for quite a while already?

And no, it's not a bug. OpenSCAD uses the official 3MF library for generating either color information or base materials. Both of which are defined in the 3MF specification. The library does not support reading proprietary extension currently. So there is no simple technical way of reading or writing any special data specific Slicers like to use.

OrcaSlicer is based on BambuSlicer, not a surprise they are compatible.

jordanbrown0, 2025-04-23:
>Is there much interest in having colors for display and not printing?

Yes.  Even if you have a single-color printer, it's useful to be able to visualize color, perhaps of parts that are printed independently, perhaps of parts that come from the store, or perhaps parts that you're going to paint.  Plus there are people who are using OpenSCAD for non-3D-printing purposes, generating illustrations or animations.

Full color printing support is best, but display-only support is still useful.

FastFrench, 2025-04-24:
Yep sure.
Now that I'm compared the .3mf files produced by OpenSCAD vs those rebuilded by OrcaSlicer (just open + save, or even open + assign filaments + save), I much better understand the issue. 
It's now clear that the slicers do not explicitely handle colors (well of course they do for display, but not in the files). They just assign filament, and the filament data holds the color.