openscad/openscad #5852: A suggestion to improve data sharing with the slicer

FastFrench, 2025-04-23:
**Is your feature request related to a problem? Please describe.**
I want to make multicolor printing, and right now it is a bit tricky. 
The way to support them is to understand how have several parts in your object, then do your best so that each such part correspond to a single filament (color and/or material). That way you have a chance to have a model compatible with color printing. 
But then you still have one more thing to do each time you make a change and import the .3mf in your slicer: you have to set the filament to be used for each part. And the name of the parts won't help you here: you can't define them explicitely in OpenSCAD, the name is hardcoded: OpenSCAD Model, OpenSCAD Model 1, OpenSCAD Model 2, and so on.
So if you have let say 4 real parts in your model, and 4 colours in each, then you will have 16 lines called 'OpenSCAD Model' up to 'OpenSCAD Model 15' and have to affect the right filament for each of them.

**Describe the solution you'd like**
Instead of relying on implicit (and somewhat foggy) mecanism to define and name the different parts that will be visible in the slicer, maybe we could have a new keyword (like **object(name)** ) for that purpose in OpenSCAD, working a bit like color() do. The idea here is that if we use _object("bluepart")_ in differents parts of the code, all the childs will be part of the same object, and so would be grouped as a single object in .3mf file, and visible as a single part in the slicers. 
I've downloaded the source code and checked how it feels. I think it should be doable with moderate impact. Of course those object() functions would be optional, and mecanisms kept unchanged if none of them is defined (for backward compatibility). 
I expected to try to code it by myself (C++ was my main coding language when I was young, using it every day for over 20 years), but I totally stopped since over 10 years, and the C++ language significantly improved in the while. Not to mention that I mainly use VS2022, and it seems a bit complicate to just build the solution with it. 

**Describe alternatives you've considered**
Alternatives for color printing are well known I think for the fans of OpenSCAD. 

**Additional context**
I started to compare the .3mf exported by OpenSCAD, with those generated by OrcaSlicer when I just open and save the .3mf, with no filament affected (very interesting BTW: I think the files produced by OrcaSlicer are better optimized and more readable. And then I checked the changes after filament affectation (meaning the item is colorized), and was surprised that only the Metadata were impacted. But I agree that the fact it relies on metadata for color/material/filament assignment looks like a temporary solution, as it is not very clean. And by the way that format was defined by BambuStudio, and is now reused by the others. 

thanks

jordanbrown0, 2025-04-24:
Probably a duplicate of #1608.

FastFrench, 2025-04-24:
Yep, you're right, it's pretty the same need. 

2016... it let me think that it won't ever be implemented.

jordanbrown0, 2025-04-24:
There's some hope - lots of people want it - but it isn't as easy as it might first appear.  The rendering mechanisms are fundamentally mesh processors - they process the surface of the objects, and these features require not only keeping attributes attached to parts as they go through the meat grinder, but also tracking the *insides* of the objects too, so that when you union two dissimilar objects something predictable happens to the overlapping volume.

closed 2025-04-24 by jordanbrown0

gsohler, 2025-04-25:
In my Fork you can specify an User defined Name to each object in a 3mf file