openscad/openscad #5854: Search field insertion point jumps to end after clicking earlier in field text and pressing any key

Zenmaster43, 2025-04-24:
**Bug Description**
When I clear the search field and type in it, then click somewhere before the end of the text in the field to set the insertion point there, then start typing to correct it, the first character gets inserted, then the insertion point automagically jumps to the end and the rest of the corrective characters are added at the end of the search field. To make the correction requires typing one character, then clicking right after it, then typing the next character, then clicking right after it, etc., etc., ad nausea. So, to insert several characters is more tedious than clearing the field and starting over.

**To Reproduce**
Steps to reproduce the behavior:
1. Type the keyboard shortcut command-F to show the search form.
2. Clear all text from the search field, if any.
3. Type some searchable text with a mistake in it into the search field.
4. Click on the gap between the mistake and the subsequent character in the search field.
5. Press the backspace key to delete the mistake.
6. Press character key(s) to insert the correct character(s).

**Expected behavior**
The correcting characters should all be inserted in sequence where the deleted text was.

**Code reproducing the issue**
N/A

**Screenshots**
Couldn't screenshot it to get the insertion point included, as it's flashing.
One image with multiple columns of Lib & Build window contents included, since copying the text in that window doesn't actually place it on the clipboard for pasting it, for example in this bug report.

**Environment and Version info (please complete the following information):**
 - OS: MacOS Sequoia 15.1
 - System: Apple M2 (MacBook Air 2022)
 - OpenSCAD Version: 2024.11.10 (git 6f5932feb)

**Library & Graphics card information**
See attached screenshot. (note: another bug prevents copied info in the Build & Info window from being placed in the clipboard for use in other applications.)

**Additional context**

[image]

No other context needed. Nothing else unusual noted.

jordanbrown0, 2025-04-24:
Not duplicated on Windows 11 OpenSCAD 2024.04.23.

kintel, 2025-04-24:
@Zenmaster43 2024.11.10 is pretty old - could you verify this on a recent snapshot?

t-paul, 2025-07-22:
Closing, no feedback.

closed 2025-07-22 by t-paul