openscad/openscad #5857: Discussion: Documentation heirarchy

AaronVerDow, 2025-04-25:
After getting started with contributing, I'd like to update and consolidate some documentation. Between the wikibook, new website, and git repo makrdown files, what is the preferred place for new documentation?

I want to add Compiling, Testing, and Contributing sections to README.md but there's too much information to be included in one file. If I wanted to consolidate documents into one location, should I stick to the WikiBook or do we want to migrate docs to the new website long term?  For instructions on building specifically, would markdown files in the source repo be a better place?

jordanbrown0, 2025-04-25:
My impression is that the Wikibooks document is the preferred place for most documentation.

In particular, there is a Building section at
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Building_OpenSCAD_from_Sources

t-paul, 2025-04-25:
As the build information is currently a bit duplicated, I think _very_ specific development stuff is better placed in the repo itself, everything else, especially anything user and language related should be in the Manuals at WikiBooks.

For the repo there is even some sort of convention to split things into multiple *.md files, like CONTRIBUTING.md and so on.

kintel, 2025-04-25:
There is also https://github.com/openscad/openscad/blob/master/doc/testing.txt, which could be cleaned up a bit, made into a .md etc.

I agree that anything dev-related best fits in the repo, and anything user-related in the doc or website.

jordanbrown0, 2025-04-25:
Should we move the "building from source" section out of the Wikibook?  Or would the distinction be building-to-use vs developing?

kintel, 2025-04-25:
I'd say ideally move out.

t-paul, 2025-04-25:
Yeah, I'd say so too. Originally I put the windows msys2 build instruction there thinking it might be easier for people to edit, but I suppose for actual developer stuff, changing on github is not much more complicated.

AaronVerDow, 2025-04-25:
This sounds good, I'll start collecting those docs in a branch.

closed 2025-04-25 by AaronVerDow