openscad/openscad #5862: ERROR: CGAL error in CGAL_Nef_polyhedron3()

dhyananbho, 2025-04-27:
**Describe the bug**
On MAC OS Sequoia 15.4.1 OpenSCAD-2025.03.20.dmg was still OK.
As of version OpenSCAD-2025.03.21.dmg, the following error occurs in connection with the latest BOLS2 library from 2025-04-27:
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: e_below != SHalfedge_handle()
File: /Users/distiller/libraries/install/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 418
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: e_below != SHalfedge_handle()
File: /Users/distiller/libraries/install/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 418
ERROR: CGAL error in CGALUtils::applyUnion3D: CGAL ERROR: assertion violation!
Expr: itl != it->second.end()
File: /Users/distiller/libraries/install/include/CGAL/Nef_3/SNC_external_structure.h
Line: 1078

**To Reproduce**
Steps to reproduce the behavior:
1. Open OpenSCAD-2025.03.21.dmg or newer
2. Open scad-document with this content:

include <BOSL2/std.scad>

$fa=1;
$fs=0.2;
Wand_D              = 2.4;
Rundung             = Wand_D;

module Flaschenring(DI=86, H=12, Topf_D=8) {
    union() {
        // grosser Ring
        cyl(d=DI+Wand_D, h=H, anchor=BOT);
        // Topfbefestigung
        back((DI+Wand_D+Topf_D)/2) cuboid([20,Topf_D+2*Wand_D,H], rounding=Rundung, teardrop=true, anchor=BOT);
    }
}
Flaschenring();

4. Click Preview
5. Wait very long ...
6. See error

**Expected behavior**
as in OpenSCAD Version 2025-03-20

**Environment and Version info (please complete the following information):**
 - MAC OS Sequoia 15.4.1
 - MacBook Pro M1 max
 - OpenSCAD Version 2025-03-21 or newer

**Library**
https://github.com/BelfrySCAD/BOSL2

kintel, 2025-11-02:
Smaller test case (from #6083):
```
include<BOSL2/std.scad>
cuboid([30,40,50], rounding=10, teardrop=true);
```

kintel, 2025-11-02:
The Manifold issue seems to be related to rounding from CGAL's Gmpq to Manifold's after a Minkowski convex decomposition; it causes some intermediate mesh to be considered self-intersecting or non-manifold. There might be opportunities for doing some basic mesh repair prior to converting to Manifold.

kintel, 2025-11-02:
Also note that the "wait very long" aspect of this ticket is due to #6297

stefano2734, 2026-09-05:
There are some open issues with polyhedron3 in CGAL.
Some are closed or will be closed in 6.3.
https://github.com/CGAL/cgal/issues?q=is%3Aissue%20state%3Aopen%20polyhedron3
Is there an issue to this here?

jordanbrown0, 2026-09-26:
I duplicate this with 2026.08.29 with CGAL, with a current BOSL2.

I do **not** duplicate it with Manifold.  Perhaps this can be marked fixed-by-Manifold.  (But see below!)

With Manifold, I do get this warning:
> WARNING: [manifold] Minkowski failed with error, falling back to Nef operation: bad any_cast
... which maybe deserves an issue of its own.

Note:  when reporting or analyzing a geometry failure with BOSL2, it's best to use a CSG dump to turn the BOSL2-isms into base OpenSCAD (and then simplify that if possible), in case a subsequent BOSL2 change stops it from triggering a still-present OpenSCAD bug.

Here's a pretty minimal base-OpenSCAD derivative of the test case that @kintel gave in https://github.com/openscad/openscad/issues/5862#issuecomment-3478202846 :
```
minkowski() {
    cube(1);
    rotate_extrude($fn = 6) {
        polygon(points = [
            [0, 10],
            [4, 10],
            [10, 0],
            [2, -9],
            [0, -10]
        ]);
    }
}
```
which (with CGAL) yields:
```
ERROR: CGAL error in CGALUtils::applyUnion3D: CGAL ERROR: assertion violation!
Expr: itl != it->second.end()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_external_structure.h
Line: 1084
```

However, when I went back to see if Manifold still generated the "bad any_cast" warning, I got this much more interesting warning:
```
WARNING: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: e_below != SHalfedge_handle()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 418. Attempting union...
WARNING: CGALUtils::convertSurfaceMeshToNef: Discarded 24 facets during Nef conversion.
```
That doesn't look like it's related to either the original error or the any_cast error, so maybe it too deserves an issue of its own.

jordanbrown0, 2026-09-26:
This seems to be a pretty minimal example of the any_cast error:
```
minkowski() {
    cube(1);
    mirror([1,0,0]) {
        rotate_extrude($fn = 4) {
            polygon(points = [
                [0, 10],
                [10, 0],
                [0, -10]
            ]);
        }
    }
}
```

jordanbrown0, 2026-09-26:
Removing the mirror flips it from the any_cast error to the SHalfedge_handle error.

It may be relevant to one or more of these errors that the rotate_extrude is generating a bunch of extra triangles, many of them degenerate.  That RE should be generating an octhedron, with 8 triangles, but is actually generating 24 triangles (as reported by the render report, and as observed in the STL).

That's not terribly surprising, given the segment that runs along the axis, but it still may be causing indigestion downstream.

jordanbrown0, 2026-09-27:
I tracked down the any_cast error to its source in CGAL, though at that point I largely got lost in a maze of twisty little templates, all different.

https://github.com/CGAL/cgal/blob/4abd208f2fa7295c0ef118579c4b30ba560dca5f/Nef_3/include/CGAL/Nef_3/SNC_FM_decorator.h#L416
```
      Edge_of[ std::any_cast<unsigned>(info(e_min->twin()->source()->twin()->source())) ];
```
It appears that the std::any value returned by info() does not contain an unsigned.  What it **does** contain... I'm not entirely sure.  I think it may be a Halfedge_handle, but what **that** is, I don't know.

It looks kind of like this is a caller-specified value that you can associate with a vertex.