openscad/openscad #5864: VirusTotal False Positives on 2025.4.27

kilgorezer, 2025-04-28:
Please first consider that general questions about using OpenSCAD or how to write code for specific models are better asked on the mailing list. See https://openscad.org/community.html for more details.

**Environment**
Windows Executable, Site

**Issue**
VirusTotal gives two false positives on the Windows x64 executable from Trapmine and Ikarus.

[https://www.virustotal.com/gui/file/7c489623b17852a9cdeeacd53349ff7630136b742607158a7f9a74c09a1f251f/detection](https://www.virustotal.com/gui/file/7c489623b17852a9cdeeacd53349ff7630136b742607158a7f9a74c09a1f251f/detection)

Just letting you know so it may be possible to address these.

kintel, 2025-04-28:
Thanks for letting us know! There isn't much we can do, except confirm here that the release in question is virus free.
You can always file a bug report with virustotal, as this is likely an problem with their detector.

closed 2025-04-28 by kintel

gsohler, 2025-04-28:
apparently these 2 false positives do exist and are know for a long time.
in case, that's not true, it would be possible to bisect the exact time
when these were added and see the code changes
which makes mxe trigger the "false positives"  and see if this code can be
written differently.

kintel, 2025-04-28:
Yes, it's always possible to work around bugs in virus scanners by trying to generate different binary patterns, if we feel like putting our limited time into that.

I'd say rather report the bug upstream, and if it's not fixed, switch to another virus scanner :)

nophead, 2025-04-28:
It seems to be the installer that looks like a virus, not the OpenSCAD executable. A program that installs another program probably looks like a Trojan to a naïve virus checker.