openscad/openscad #5867: OpenSCAD - ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.

J13ax, 2025-04-29:
Hi all,

I'm trying to create a model for a golf iron in OpenSCAD for a project, but running into mesh issues.

The issue seems to be when I add the 'hosel' to the model.
It previews fine, but then wont render due to open mesh.
It renders fine if I comment out the hosel.
The strange thing is it has worked before, I just tweaked a few things not even related to the hosel. 

I'm aware the below code is a bit long, crazy, and messy, but I'm hoping someone can help work out what's wrong?

```
use <MCAD/regular_shapes.scad>

length = 80;
height = 50;
width = 1;
loft = 18;
muscle_width_heel = 9;
muscle_width_toe = 18;
muscle_top_heel = -17;
muscle_top_toe = 2;
lie = 60;
hole_size = 9;

module face() {
		polyhedron(
           points = [
               [length / 2, width / 2, height / -2], //0
               [length / 2, width / 2, height / 2], //1
               [length / -2, width / 2, height / 2], //2
               [length / -2, width / 2, height / -2], //3
               [length / -2, width / -2, height / -2], //4
               [length / -2, width / -2, height / 2], //5
               [length / 2, width / -2, height / 2], //6
               [length / 2,  width / -2, height / -2], //7
           ],
           faces = [
               [0, 1, 2, 3], [7, 0, 3, 4], [1, 6, 5, 2], [7, 6, 1, 0], [4, 5, 6, 7], [3, 2, 5, 4]
           ]
		);
};

module muscle() {
	rotate([loft, 0, 0]) translate([0, 0, length*-5]) scale([1, 0.1, 1]) sphere(r = length*5, $fn = 256);
};

module muscle() {
		polyhedron(
			points = [
				[length / 2, width / -2, height / -2], //0
				[length / 2, width / -2, muscle_top_toe +3], //1
				[length / -2, width / -2, muscle_top_heel +3], //2
				[length / -2, width / -2, height / -2], //3
				[length / -2, width / -2 - muscle_width_heel, height / -2], //4
				[length / -2, width / -2 -3, muscle_top_heel], //5
				[length / 2, width / -2 -3, muscle_top_toe], //6
				[length / 2, width / -2 - muscle_width_toe, height / -2], //7
			],
			faces = [
				[0, 1, 2, 3], [7, 0, 3, 4], [1, 6, 5, 2], [7, 6, 1, 0], [4, 5, 6, 7], [3, 2, 5, 4]
			]
		);
};

module sole() {
	translate([0, 0, 500 - height/2 + 1]) rotate([90, 0, 0]) cylinder(h = 100, r = 500, center = true, $fn = 1000);
	
	
};

module outline() {
	hull() {
			translate([(length/2)-height*2/3/2, 0, height/-2+1.5]) rotate([90, 0, 0]) linear_extrude(100, center=true) egg_outline(height*2/3, height-6, $fn=256);
			translate([(length/-2), 0, height/-2+0.5]) rotate([90, 0, 0])
			linear_extrude(100, center=true) square(size = height*1/2-3);
	};
};

module shank() {
	translate([length/-2-10, 4, 0]) rotate([0, -90+lie, 0]) union() {
		translate([0, 0, -12.5]) cylinder(h = 15, r1 = 0, r2=13.75/2-2, center = true, $fn = 256);
	};
};

module hosel() {
	difference() {
		translate([length/-2-10, 4, 0]) rotate([0, -90+lie, 0])
		translate([0, 0, 15]) cylinder(h = 40, r = 13.75/2, center = true, $fn = 256);
		translate([length/-2-10, 4, 0]) rotate([0, -90+lie, 0])
		translate([0, 0, 15]) cylinder(h = 42, r = hole_size/2, center = true, $fn = 256);
	};
};

module heel_cut() {
	intersection() {
		translate([(length/-2)+11, 0, height/-2+1.8]) rotate([90, 0, 0]) linear_extrude(100, center=true) egg_outline(height*2/3, height*2/3-3, $fn=256);

		translate([(length/-2+5) + 6.5 - (height*2/3*2/3)/2, 0, height/-2+1.592 - 5.5]) cube([22, 40, 22], center = true);
	};
};

module heel_form() {
	difference() {
			translate([(length/-2+5) + 6.5 - (height*2/3*2/3)/2, 0, height/-2+1.592 - 5.5]) cube([20, 35, 20], center = true);
			heel_cut();
	};
};

module blade() {
	difference() {
		intersection() {
			intersection() {
				rotate([loft, 0, 0]) {
					union() {
						face();
						muscle();
					};
				};
			outline();
			};
		sole();
		};
	heel_form();
	};
};

module weld() {
	hull() {
		intersection () {
			translate([length/-2, 0, 0]) cube([1, 40, 60], center = true);
			blade();
	};
		shank();
	};
};

module pieces() {
	union() {
		blade();
		weld();
	};
}

union() {
	minkowski() {
			pieces();
		sphere(r=2, $fn=30);
	};
	hosel();
};
```

kintel, 2025-04-29:
Could you fill out the rest of the bug template?

kintel, 2025-05-05:
Closing due to lack of engagement. Please reopen if you can provide the missing info,

closed 2025-05-05 by kintel