openscad/openscad #5868: Grid Option in Export PDF does not work properly

JonMcLin, 2025-04-29:
**Describe the bug**
"Grid" does not disable and state cannot be changed when "Scale" is disabled in Export PDF Options dialog.   This results in a PDF with a grid but no scale.  (This was not the intended behavior when originally implemented; see below.)   This results in unexpected behavior, because with all options disabled the grid may still be produced.  WORKAROUND: First disable Grid while Scale is enabled, and then disable Scale.

**To Reproduce**
Steps to reproduce the behavior:
1. Produce any 2D geometry (eg, square(100) is sufficient) and render.
2. Select "File"->"Export"->"Export PDF"
3. Enable "Scale"
4. Enable "Grid"
5. Disable "Scale"
6. Observe that "Grid" remains enabled but is grayed out such that it cannot be changed (both expected, this was the original design intent when I created the dialog box.)
7. Export the PDF.
8. Open the PDF - observe the geometry AND the grid, but no scale along the edges.

**Expected behavior**
The Export PDF option dialog provides a Boolean option to include a scale, and a suboption under "Scale" is a Boolean to include a grid (and another enum to set the size of that grid).  The intended behavior is that the grid will only be produced when "Scale" is selected, and this shows up in the dialog box by the "Grid" options being grayed-out when "Scale" is deselected.

**Code reproducing the issue**
Trivial - see "to reproduce"

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 24.04
 - System: AMD Ryzen 7 5700U
 - OpenSCAD Version:  2025.04.19.snap from Ubuntu Nightly

**Library & Graphics card information**
NA

**Additional context**
I'm pretty sure I tested this behavior with the original implementation, but I will go back and look.  As I recall in the original implementation the grid was produced inside the block which implemented the scale.

JonMcLin, 2025-04-29:
Yes, here's the issue in export PDF:
Grid simply needs to be moved inside the scale block like Scale Message is.

**Current:**
    // Scale
    if (options->showScale) {
    	draw_axes(cr, Mlx,Mrx,Mty,Mby);
    	// Scale Message
    	if (options->showScaleMsg) draw_text(about.c_str(), cr, Mlx+1, Mty-1, 5.);
    }
    // Grid
    if (options->showGrid) draw_grid(cr, Mlx,Mrx,Mty,Mby, options->gridSize);

**Proposed:**
    // Scale
    if (options->showScale) {
    	draw_axes(cr, Mlx,Mrx,Mty,Mby);
    	// Scale Message
    	if (options->showScaleMsg) draw_text(about.c_str(), cr, Mlx+1, Mty-1, 5.);
        // Grid
        if (options->showGrid) draw_grid(cr, Mlx,Mrx,Mty,Mby, options->gridSize);
    }

t-paul, 2025-05-01:
Closed via #5869.

closed 2025-05-01 by t-paul