openscad/openscad #5873: Precision should be binary, not decimal

wiseguy720, 2025-05-01:
**Is your feature request related to a problem? Please describe.**
From what documentation I've seen, OpenScad uses 10^-8 as a precision. Computers are binary; 10^-8 cannot be precisely represented.  Rounding off to the closest 10^-8 results in round-off error when represented in floating point.

**Describe the solution you'd like**
Precision/rounding should be 2^-27, which is slightly less than 10^8 and it can be precicely represented in IEEE 754.

**Describe alternatives you've considered**
I have considerd writing my own funtion to round everything to the closest 2^-27 or 2^-26, but if the internals of OpenScad re-round to 10^-8, it will still introduce round-off error on top of my round-off.

**Additional context**
Add any other context or screenshots about the feature request here.

kintel, 2025-05-01:
Could you clarify where 10^-8 is used as precision and what effect that has on your code?
If this is just a general investigation into internals, I suggest discussing it on the mailing list or IRC (https://openscad.org/community.html), as we want GitHub tickets to be a lot more clear and actionable.

wiseguy720, 2025-05-01:
With some effort, I belive I could create a convoluted test case, but I can see about discussing this in the mailing list instead.

closed 2025-05-01 by wiseguy720