openscad/openscad #5877: Packaging for GNU guix: lib3mf not found

nomike, 2025-05-03:
Hi!

I'm trying to package the latest version as openscad-nightly in GNU guix.

CMake fails however with not being able to find lib3mf.

I'm not yet very familiar with CMake, but so far I was able to figure out that [if (PC_LIB3MF_FOUND)](https://github.com/openscad/openscad/blob/716690492a551b6458697c9c0c77dfadbd190f4d/cmake/Modules/FindLib3MF.cmake#L28) is evaluating to `true`. But [if (Lib3MF_FOUND)](https://github.com/openscad/openscad/blob/716690492a551b6458697c9c0c77dfadbd190f4d/cmake/Modules/FindLib3MF.cmake#L35) evaluates to false.

With lib3mf changing it's casing between version 1.x and 2.x, I wouldn't be surprised if there is a wrong casing somewhere.

Do you have any idea about what the issue could be?

Cheers
nomike

This is the directory structure of lib3mf:

```plaintext
$ find /gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/share
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/share/doc
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/share/doc/lib3mf-2.2.0
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/share/doc/lib3mf-2.2.0/LICENSE
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/C
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/C/lib3mf_types.h
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/C/lib3mf.h
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/CppDynamic
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/CppDynamic/lib3mf_dynamic.h
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/CppDynamic/lib3mf_abi.hpp
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/CppDynamic/lib3mf_dynamic.hpp
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/CppDynamic/lib3mf_types.hpp
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/CSharp
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/CSharp/Lib3MF.cs
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/Cpp
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/Cpp/lib3mf_abi.hpp
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/Cpp/lib3mf_implicit.hpp
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/Cpp/lib3mf_types.hpp
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/CDynamic
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/CDynamic/lib3mf_types.h
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/CDynamic/lib3mf_dynamic.h
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/CDynamic/lib3mf_dynamic.cc
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/Pascal
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/Pascal/Unit_Lib3MF.pas
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/Go
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/Go/lib3mf_impl.go
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/Go/lib3mf.go
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/Python
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/Python/Lib3MF.py
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/NodeJS
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/NodeJS/lib3mf_nodewrapper.h
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/NodeJS/lib3mf_types.h
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/NodeJS/lib3mf_dynamic.h
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/NodeJS/binding.gyp
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/NodeJS/lib3mf_nodeaddon.cc
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/NodeJS/lib3mf_dynamic.cc
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include/Bindings/NodeJS/lib3mf_nodewrapper.cc
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/lib
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/lib/lib3mf.so.2.2.0.0
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/lib/lib3mf.so
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/lib/lib3mf.so.2
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/lib/pkgconfig
/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/lib/pkgconfig/lib3mf.pc
```

kintel, 2025-05-03:
Perhaps dump the contents of `/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/lib/pkgconfig/lib3mf.pc` in case that file has some issues

kintel, 2025-05-03:
Hard to say, could you also print the contents of PC_LIB3MF_VERSION, PC_LIB3MF_LIBRARIES and PC_LIB3MF_INCLUDE_DIRS immediately before calling find_package_handle_standard_args ?

nomike, 2025-05-03:
I added a helper function which dumps all variables an called it right after the call to [find_package_handle_standard_args(Lib3MF](https://github.com/openscad/openscad/blob/716690492a551b6458697c9c0c77dfadbd190f4d/cmake/Modules/FindLib3MF.cmake#L29).

I did a `grep -i lib3mf` on the output and this is what I got:
```
-- CMAKE_CURRENT_FUNCTION_LIST_FILE=/tmp/guix-build-openscad-nightly--2025.04.28.535d96f.drv-0/source/cmake/Modules/FindLib3MF.cmake
-- CMAKE_CURRENT_LIST_FILE=/tmp/guix-build-openscad-nightly--2025.04.28.535d96f.drv-0/source/cmake/Modules/FindLib3MF.cmake
-- CMAKE_CXX_IMPLICIT_INCLUDE_DIRECTORIES=/gnu/store/3gdzif6iaixha7vf0mx3mskxddw5rp5r-ghostscript-9.56.1/include;/gnu/store/2in27cci7by6sspjk4jwdp197i6592lg-flex-2.6.4/include;/gnu/store/pf2s6ki3j7z2q1v2pqndnmy7bv0b6yjb-gettext-minimal-0.21/include;/gnu/store/9ayvahxd8lh1cj50nig90ihwgidkqgkq-imagemagick-6.9.13-5/include;/gnu/store/bs0apm2aa01qq0vz4qbf3miqc7l6zclb-procps-4.0.3/include;/gnu/store/s6hivi3vwc1cyyw853y9c8xwahlqym5j-python-3.11.11/include;/gnu/store/wav1xw8c236aym2089mdwi3qnb79dzrf-xorg-server-21.1.15/include;/gnu/store/dmganah69s71vpdmvfd4i2883zp5j6j8-cairomm-1.18.0/include;/gnu/store/424c967mkwcyaq58lgcq0p4kv9bg0bpd-clipper2-1.5.2/include;/gnu/store/7ah6i5829f5ha6bdfzcj1gf21115xyyd-freetype-2.13.0/include;/gnu/store/5csw0jn2lfkzavkq1fjjl7s7bdgmzjjk-glib-2.82.1/include;/gnu/store/n2rphbkasv05gahnjknamyqrvckiqcvb-hidapi-0.14.0/include;/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/include;/gnu/store/pgmi9wndca71p35fzd8y4sfvx66nllwr-libglvnd-1.7.0/include;/gnu/store/p1syin32hllbjwqgqdcyj11hywyxdyd7-libspnav-0.2.3/include;/gnu/store/y9550nxs1p4k1afpib8mvry422192cjq-manifold-3.0.1/include;/gnu/store/q4vlygzdi3d3hlv2z7j9mkg8iq7p3hra-mimalloc-2.1.2/include;/gnu/store/az1ldjkkw15vssd9mdx804pspsq7516l-qtsvg-5.15.15/include;/gnu/store/zz6inps5b5yfcdmdr39931n147988rg3-tbb-2021.6.0/include;/gnu/store/di4520fl1q7lkxahblsvpd2079zhx0q4-boost-1.83.0/include;/gnu/store/3rzplnnipk5p1i8gafid3h46q5834k7z-cgal-5.6.1/include;/gnu/store/y8rpgm5pq2wzdxa9wkfhwd8l3pisi7wv-double-conversion-3.1.5/include;/gnu/store/cnk2cxb1ggcbmrg75fyjagxk6l6y4qhi-eigen-3.4.0/include;/gnu/store/36pl6iy1dv6xzidg6n05qpf4wpnlhs06-fontconfig-minimal-2.14.0/include;/gnu/store/djarc5py303kgnn130x2i15j8zjiabd2-glew-2.2.0/include;/gnu/store/kka705681m1hq98b9jz98vxk9s5qd4ld-gmp-6.3.0/include;/gnu/store/4l66m7h6laqdfzcpdkb2j9a32pb1iidx-harfbuzz-8.3.0/include;/gnu/store/dqym4cqjc0pgwhiaqim3gwz18yh5fqrc-libxml2-2.9.14/include;/gnu/store/m25xmj1x8a6pw8bhrck8kn08zff1w3hi-libzip-1.9.2/include;/gnu/store/pz4xkjl28w0pqf8nhgv68xix17ap6dsx-mpfr-4.2.1/include;/gnu/store/ii4x1zy4nwp0y3mqkjz2rsmhsqb68q4n-opencsg-1.4.2/include;/gnu/store/kyw12v6pcn5acgzpqjqmkgdvhacks9bq-qscintilla-2.13.4/include;/gnu/store/amm14yf7r8rx1kdrbvlphm8w3i75gipi-qtbase-5.15.15/include;/gnu/store/qaqn7kh1g7f1jskpfrs3vhpmp8k0i5l3-qtmultimedia-5.15.15/include;/gnu/store/56aq6sdx35f7rsxq8jq9ypafk0dhd3p3-bzip2-1.0.8/include;/gnu/store/544ga5n6nq8y2961ihigra6n6a0r99nn-file-5.45/include;/gnu/store/vrl2r0ps24k1a6yap2chjylv5rhbndvr-gawk-5.3.0/include;/gnu/store/fbaw0sb21gv02qq7gs9wg5y5wlpdgzih-xz-5.4.5/include;/gnu/store/lc6r1mh0jlcwixjziiw1zsx1my478rv1-make-4.4.1/include;/gnu/store/ylgkrq9j1gyfig2y66srmgsbyh9w20d9-binutils-2.41/include;/gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/include/c++;/gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/include;/gnu/store/zvlp3n8iwa1svxmwv4q22pv1pb1c9pjq-glibc-2.39/include;/gnu/store/gfxk7da5fdns0jgh499qy4snjp9rh02r-xorgproto-2024.1/include;/gnu/store/zv420d7jscbcwkzndxb9n9dma4xhmf43-pixman-0.42.2/include;/gnu/store/a4x6jq2c8vh6n58x7r18j31n32v1rkxf-mesa-24.3.2/include;/gnu/store/79vyia401wsj77nw5yrnhyypbgmpnp9r-libxcvt-0.1.3/include;/gnu/store/f6by3dsa7y0pcxxsz32nsd49md8j7yv2-libpciaccess-0.18.1/include;/gnu/store/gh9cgprk2x0zq7xw44dal0v8s3wixpj3-cairo-1.18.2/include;/gnu/store/wg97vaz6bnj0g9hr1pq7vqwbxgd971vd-libsigc++-3.6.0/include;/gnu/store/1prv14v6jfnzzg7szm57690b7fr6sx33-zlib-1.3/include;/gnu/store/qj9jhgqn2b9g7yxdhyw6ki1vz241apkd-libpng-1.6.39/include;/gnu/store/2p8j6npwa2k59d8lbhlqzvffn0437x8l-util-linux-2.37.4-lib/include;/gnu/store/gwn3p1r5ghlapv9yjad0mk2n23la7j8z-pcre2-10.42/include;/gnu/store/s6iqwc5sqjrk76kzslqc1n1wlcvfyqkw-libffi-3.4.4/include;/gnu/store/8bs3gia5rknqsw840rrwm2na4200ybc5-expat-2.5.0/include;/gnu/store/spmp7bbr5svf4m6acrncqaz8r0bdk5xr-glu-9.0.2/include;/gnu/store/8v7qj9333g2ydgk5yarpv748xqx2glh7-icu4c-73.1/include;/gnu/store/crv2h3nal4a9b9hsrrznsffw13izqf6r-graphite2-1.3.13/include;/gnu/store/svghsdn1x3fyili8ij791nn25zba9bpf-linux-libre-headers-5.15.49/include;/gnu/store/x81gxj50x2a9h11svxnkzvwzghc2gzzn-libxxf86vm-1.1.6/include;/gnu/store/4y6bw7c0xl96x0j0wym07vi1qa356nya-libxshmfence-1.3.3/include;/gnu/store/7yrbwc38rwnngv0ysc53qp35f78k92jk-libxfixes-6.0.1/include;/gnu/store/fqlf0wcypg2q7drhbqrr7nw4rkn74mh0-libxdamage-1.1.6/include;/gnu/store/ak5m1c67g8avfa9q05q9k2cqwzzcznwl-libx11-1.8.10/include;/gnu/store/q1f4p6wk68zp1nzyzjng832bih2p5mwv-libvdpau-1.5/include;/gnu/store/sw0rl8sfz01j723s03yndrnk14xllfpd-libdrm-2.4.124/include;/gnu/store/4c99ncr205cp7cb0419plv9vikhvar0s-libxrender-0.9.12/include;/gnu/store/fixj7jfkisczxaibjhlnc763vqp5x3hd-libxext-1.3.6/include;/gnu/store/nw27myjl7crh9rallpqx27wdlmf54qn4-libxcb-1.17.0/include;/gnu/store/x1mms0gyjks84665606bcjsnzri3mr5k-libxdmcp-1.1.5/include;/gnu/store/gvax5yl1b2i83lw5ak6r0gi28m4mxqqa-libxau-1.0.12/include;/gnu/store/86fc8bi3mciljxz7c79jx8zr4wsx7xw8-gcc-11.4.0/include/c++/x86_6 [... cut ...]
-- _pkgconfig_STATIC_LIBRARY_DIRS=/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/lib
-- lib3mf_DIR=lib3mf_DIR-NOTFOUND
-- pkgcfg_lib_PC_LIB3MF_3mf=/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/lib/lib3mf.so
-- pkgcfg_lib_PC_LIB3MF_z=/gnu/store/1prv14v6jfnzzg7szm57690b7fr6sx33-zlib-1.3/lib/libz.so
-- pkgcfg_lib_PC_LIB3MF_zip=/gnu/store/m25xmj1x8a6pw8bhrck8kn08zff1w3hi-libzip-1.9.2/lib/libzip.so
-- prefix_result=/gnu/store/myqqm9q4zybyl39plgw355ghrng7wka4-lib3mf-2.2.0/lib
```

kintel, 2025-05-03:
`PC_LIB3MF_INCLUDE_DIRS` is empty

..but `PC_LIB3MF_INCLUDEDIR` is set, although we don't check it. Not sure exactly why - need to read up on pkg-config.
..but in the meantime, try patching the CMake macro locally to substitute it.

kintel, 2025-05-05:
@nomike Did you figure this out?

nomike, 2025-05-06:
There is this call:
[`pkg_check_modules(PC_LIB3MF ${_Lib3MF_PC_NAME} QUIET)`](https://github.com/openscad/openscad/blob/716690492a551b6458697c9c0c77dfadbd190f4d/cmake/Modules/FindLib3MF.cmake#L27)

The variable `${_Lib3MF_PC_NAME}` is set to `lib3mf` in this line:
[set(_Lib3MF_PC_NAME lib3mf)](https://github.com/openscad/openscad/blob/716690492a551b6458697c9c0c77dfadbd190f4d/cmake/Modules/FindLib3MF.cmake#L22)

The [CMake documentation](https://cmake.org/cmake/help/latest/module/FindPkgConfig.html#command:pkg_check_modules) lists this method signature:

```c
pkg_check_modules(<prefix>
                  [REQUIRED] [QUIET]
                  [NO_CMAKE_PATH]
                  [NO_CMAKE_ENVIRONMENT_PATH]
                  [IMPORTED_TARGET [GLOBAL]]
                  <moduleSpec> [<moduleSpec>...])
```

Further on it says:

> Changed in version 3.18: Include directories specified with -isystem are stored in the <XXX>_INCLUDE_DIRS variable. Previous versions of CMake left them in <XXX>_CFLAGS_OTHER.

According to the build log, the CMake version being used with my quix build is 3.24.2, so that note should already apply.

The manual also states:

> There are some special variables whose prefix depends on the number of `<moduleSpec>` given. When there is only one `<moduleSpec>`, `<YYY>` will simply be <prefix>, but if two or more `<moduleSpec>` items are given, `<YYY>` will be `<prefix>`_`<moduleName>`.
> `<YYY>`_INCLUDEDIR
>     include directory of the module

So all in all:
I do not understand why `PC_LIB3MF_INCLUDE_DIRS` is empty, but I do fully understand why `PC_LIB3MF_INCLUDEDIR` is set.

So what I've done in my guix build is to substitute the former with the latter and it works.

You might want to change this upstream as well.

kintel, 2025-05-06:
Thanks for the confirmation; I'll try to figure this out and see if it's safe to switch.

nomike, 2025-05-29:
Did you manage to have a look at this already? ^_^

nomike, 2025-07-16:
I just merged a fix for this into the main Branch of PythonSCAD.

https://github.com/pythonscad/pythonscad/pull/54

You peeps probably want to cherry pick that commit. :-)