openscad/openscad #5880: "qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat" on guix

nomike, 2025-05-04:
I'm still working on getting a current nightly build running on GNU guix.
I managed to get it built but have some issues with the tests.

When I run openscad, I get presented with the wolcome dialog but once I click "New", openscad crashes.

I see the following on the console:

```plaintext
$ /tmp/guix-build-openscad-nightly--2025.04.28.535d96f.drv-0/build/openscad         
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Gtk-Message: 01:58:32.997: Failed to load module "canberra-gtk-module"
Gtk-Message: 01:58:32.998: Failed to load module "canberra-gtk-module"
Could not initialize localization (application path is '/tmp/guix-build-openscad-nightly--2025.04.28.535d96f.drv-0/build').
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Loaded design '/home/nomike/coding/3d prints/my-designs/squeeze-bottle-stand/squeeze-bottle-stand.scad'.
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Could not initialize GLX
[1]    2697848 IOT instruction (core dumped)  QT_QPA_PLATFORM=xcb 
```
The guix package definition: [openscad-nomike.scm.gz](https://github.com/user-attachments/files/20025546/openscad-nomike.scm.gz)

The build log: 
[openscad-nomike-build.log.gz](https://github.com/user-attachments/files/20025548/openscad-nomike-build.log.gz)

There are a few issues here, but I think the GLX is the most serious one, so I wanted to start tackling this.

Any Idea on that one?

t-paul, 2025-05-04:
Your definition depends on `libglvnd` for OpenGL which is just a dispatcher/proxy library. Do you have an actual OpenGL driver in scope when the application starts (e.g. from Mesa)?

As you define a specific commit to build, it would be useful to also set for `cmake`
```
-DOPENSCAD_VERSION=...
-DOPENSCAD_COMMIT=...
```

CI on Circle-CI for example is using for the AppImage
```
export OPENSCAD_COMMIT=$(git log -1 --pretty=format:%h)
export OPENSCAD_VERSION="$(date +%Y.%m.%d).ai${CIRCLE_BUILD_NUM}"
```

As this recipe is likely meant to rebuild at any time, it would probably be better relying on the commit date for the version tag though, keeping it consistent with the latest code change:
```
git log -1 --date=format:"%F" --format="%ad"
```

nomike, 2025-05-04:
Thanks a lot for the advice.

I've updated the package definition:

```scheme
(define-public openscad-nightly
  (let ((commit "535d96fe28931d92ca4af7cdddb18dd006fb4125")
        (revision "2025.04.28"))
    (package
      (name "openscad-nightly")
      (version (git-version "" revision commit))
      (source
       (origin
         (method git-fetch)
         (uri (git-reference
               (url "https://github.com/openscad/openscad")
               (commit commit)
               (recursive? #t)))
         (sha256
          (base32 "0srfpjw5y6nqxypfzjzjkdckzb40r1ylhlzp52wr29yi7ahsmp83"))
         (file-name (git-file-name name version)))

       )
      (build-system cmake-build-system)
      (arguments
       (list
        #:configure-flags
        #~(list "-DCMAKE_BUILD_TYPE=Release"
                "-DUSE_BUILTIN_OPENCSG=ON"
                "-DMANIFOLD_PYBIND=OFF"
                "-DMANIFOLD_TEST=OFF"
                "-DENABLE_TESTS=OFF"
                "-DEXPERIMENTAL=ON"
                "-DSNAPSHOT=ON"
                "-DSUFFIX=nightly"
                "-DUSE_BUILTIN_CLIPPER2=OFF"
                (string-append "-DOPENSCAD_VERSION=" #$revision)
                (string-append "-DOPENSCAD_COMMIT=" #$commit)
                "-DENABLE_GLX=OFF")
        #:phases
        #~(modify-phases %standard-phases
            ; (delete 'check)
            (add-after 'unpack 'patch-source
              (lambda* (#:key inputs #:allow-other-keys)
                ;; https://github.com/openscad/openscad/issues/5877
                (substitute* "cmake/Modules/FindLib3MF.cmake"
                  (("PC_LIB3MF_INCLUDE_DIRS")
                   "PC_LIB3MF_INCLUDEDIR"))
                (substitute* "CMakeLists.txt"
                  ;; Use the system sanitizers-cmake module.
                  (("\\$\\{CMAKE_SOURCE_DIR\\}/submodules/sanitizers-cmake/cmake")
                   (string-append (assoc-ref inputs "sanitizers-cmake")
                                  "/share/sanitizers-cmake/cmake")))))
            (add-after 'unpack 'patch-source
              (lambda* (#:key inputs #:allow-other-keys)
                (substitute* "cmake/Modules/FindLib3MF.cmake"
                  (("PC_LIB3MF_INCLUDE_DIRS")
                   "PC_LIB3MF_INCLUDEDIR"))
                (substitute* "CMakeLists.txt"
                  ;; Use the system sanitizers-cmake module.
                  (("\\$\\{CMAKE_SOURCE_DIR\\}/submodules/sanitizers-cmake/cmake")
                   (string-append (assoc-ref inputs "sanitizers-cmake")
                                  "/share/sanitizers-cmake/cmake")))))
            )))

      (inputs (list boost
                    cairomm
                    cgal
                    clipper2
                    double-conversion
[... 94 lines cut ...]
dynamics is used by FreeCAD 1.0.0 for its new Assembly workbench.")
      (license license:lgpl2.1+))))
```

However, I'm now still receiving the same error:

```plaintext
/tmp/guix-build-openscad-nightly--2025.04.28.535d96f.drv-10/build/openscad-nightly 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Gtk-Message: 15:29:23.647: Failed to load module "canberra-gtk-module"
Gtk-Message: 15:29:23.647: Failed to load module "canberra-gtk-module"
Could not initialize localization (application path is '/tmp/guix-build-openscad-nightly--2025.04.28.535d96f.drv-10/build').
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Could not initialize GLX
[1]    3793683 IOT instruction (core dumped)  
```

I'm barely familiar with `EGL`, `GLX`, `mesa`, `vulcan`, etc. yet (but I'm learning). So for now I have added `mesa` as a dependency and removed `libglvnd`.

What I don't understand is that it still complains about not being able to initialize GLX even though I explicitely specified `-DENABLE_GLX=OFF`. ¯\_(ツ)_/¯

Here is the new build-log: [openscad-nomike-build.log.gz](https://github.com/user-attachments/files/20027891/openscad-nomike-build.log.gz)

nomike, 2025-05-04:
I'm running guix on top of Ubuntu 24.10, which is using Wayland.

In Ubuntu I can see both `qtwayland5` and `qt6-wayland` being installed.
In the guix package definition the inputs contain `qtwayland-5`.

However, I'm not getting this error:

```plaintext
QT_QPA_PLATFORM=wayland /tmp/guix-build-openscad-nightly--2025.04.28.535d96f.drv-10/build/openscad-nightly
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

[1]    3894430 IOT instruction (core dumped)  QT_QPA_PLATFORM=wayland 
```

I will also write to guix-devel@gnu.org for some advice on this as it seems like it´s an issue specific to guix.

nomike, 2025-05-06:
Danny, a friend of mine and one of the guix core developers, helped me troubleshoot that issue and it was quite a challenge.

Since Danny and I were chatting over matrix in German, I'm trying my best to translate the relevant parts to English and bring it to a form which is somehow understandable.

I got the broader picture, but I'm still not understanding all details. Still I'm trying my best to transliterate our findings:

openscad has issues finding the GLX shared object:

```plaintext
$ strace -f  /gnu/store/vd9x6q48zriwgr63hrw2vcm616rsmggr-openscad-nightly--2025.04.28.535d96f/bin/openscad-nightly 2>&1 |grep '[.]so'
[...]
[pid 28191] openat(AT_FDCWD, "/gnu/store/6lqzf52q96pkp6phijips98w53nczz4w-libglvnd-1.7.0/lib/libGLX_mesa.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[pid 28191] openat(AT_FDCWD, "/gnu/store/hw6g2kjayxnqi8rwpnmpraalxi0djkxc-glibc-2.39/lib/libGLX_mesa.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[pid 28191] openat(AT_FDCWD, "/gnu/store/1bbzk022hdrv8y5mvi8kvw07bddy2bk6-libx11-1.8.10/lib/libGLX_mesa.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[pid 28191] openat(AT_FDCWD, "/gnu/store/hw6g2kjayxnqi8rwpnmpraalxi0djkxc-glibc-2.39/lib/libGLX_mesa.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[pid 28191] openat(AT_FDCWD, "/gnu/store/vd9x6q48zriwgr63hrw2vcm616rsmggr-openscad-nightly--2025.04.28.535d96f/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 33
[pid 28191] openat(AT_FDCWD, "/gnu/store/6lqzf52q96pkp6phijips98w53nczz4w-libglvnd-1.7.0/lib/libGLX_indirect.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[pid 28191] openat(AT_FDCWD, "/gnu/store/hw6g2kjayxnqi8rwpnmpraalxi0djkxc-glibc-2.39/lib/libGLX_indirect.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[pid 28191] openat(AT_FDCWD, "/gnu/store/1bbzk022hdrv8y5mvi8kvw07bddy2bk6-libx11-1.8.10/lib/libGLX_indirect.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[pid 28191] openat(AT_FDCWD, "/gnu/store/hw6g2kjayxnqi8rwpnmpraalxi0djkxc-glibc-2.39/lib/libGLX_indirect.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
```
It also showed this though:

```plaintext
openat(AT_FDCWD, "/gnu/store/6lqzf52q96pkp6phijips98w53nczz4w-libglvnd-1.7.0/lib/[libGLX.so](http://libglx.so/).0", O_RDONLY|O_CLOEXEC) = 3
```

We found the part in `libglvnd` where the [`inidrect` fallback](https://github.com/NVIDIA/libglvnd/blob/606f6627cf481ee6dcb32387edc010c502cdf38b/src/GLX/libglxmapping.c#L49) comes from.

In mesa 17.1 [GLVND support for EGL was added](https://www.phoronix.com/news/Mesa-GLVND-EGL).

In the [mesa source](https://cgit.freedesktop.org/mesa/mesa/commit/?h=17.1&id=ce562f9e3fab769d64b0e5453ec2b4f8710a31ce) you can find:

```automake
AC_ARG_ENABLE([libglvnd],
    [AS_HELP_STRING([--enable-libglvnd],
        [Build GLX and EGL for libglvnd @<:@default=disabled@:>@])],
    [enable_libglvnd="$enableval"],
    [enable_libglvnd=no])
AM_CONDITIONAL(USE_LIBGLVND, test "x$enable_libglvnd" = xyes)
```

So Danny suggested to either recompile mesa with libglvnd support or remove EGL support from openscad.
For the time being I chose the latter option, but I might still try getting EGL to work.

openscad looks fine with GLX only, but maybe that additional layer of XWayland causes a performance penalty when rendering?

So what I ended up doing:

1. Switch from `cmake-build-system` to `qt-build-system`
2. Set compile options `-DENABLE_GLX=ON` and `-DENABLE_EGL=OFF`
3. remove `libglvnd` from the inputs

Which leads me to the following package definition:

```scheme
(define-public openscad-nightly
  (let ((commit "d1351d6282abfd239cdd0c657f755d8c4a123ff8")
        (version "2025.05.02")
        (revision "0"))
    (package
      (name "openscad-nightly")
      (version (git-version version revision commit))
      (source
       (origin
         (method git-fetch)
         (uri (git-reference
               (url "https://github.com/openscad/openscad")
               (commit commit)
               (recursive? #t)))
[... 89 lines cut ...]
                           python
                           xorg-server-for-tests))
      (synopsis "Script-based 3D modeling application")
      (description
       "OpenSCAD is a 3D Computer-aided Design (CAD) application.  Unlike an
  interactive modeler, OpenSCAD generates 3D models from a script, giving you
  full programmatic control over your models.")
      (home-page "https://openscad.org/")
      (license license:gpl2+))))
```

There are only two warnings left:

```plaintext
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Gtk-Message: 04:26:34.829: Failed to load module "canberra-gtk-module"
```

The one about wayland seems to not matter at all and the one about canberra is weird, as I have specified libcanberra as a package input.

As far as I understood this library is used for notification sounds, but despite the error, the "ping" when F6-rendering has finished is audible.

I will have a look though with `strace`.

The issue with the locale is gone too, most likely it didn't like me calling the binary from the build directory directly instead of installing it properly first.

kintel, 2025-05-06:
Just FYI: The GLX vs. EGL setting in OpenSCAD only influences cmd-line rendering. We prefer EGL as it allows you to perform offscreen rendering without first setting up an X11 server - good for things like ssh sessions and CI testing.

Debugging this in the context of OpenSCAD can be a bit heavy. While developing the current multi-platform offscreen rendering infra, I build a stand-alone offscreen renderer playground which may be easier to reason about: https://github.com/kintel/offscreen

nomike, 2025-05-06:
Regarding that canberra issue:

In `strace` you can see that openscad couldn't find `libcanberra-gtk-module.so`:

```plaintext
access("/gnu/store/n8d05d0kh201gn49cby1y8rzj6300m7z-gtk+-3.24.43/lib/gtk-3.0/3.0.0/linux/modules/libcanberra-gtk-module.so", F_OK) = -1 ENOENT (No such file or directory)
access("/gnu/store/n8d05d0kh201gn49cby1y8rzj6300m7z-gtk+-3.24.43/lib/gtk-3.0/3.0.0/linux/modules/libcanberra-gtk-module.la", F_OK) = -1 ENOENT (No such file or directory)
access("/gnu/store/n8d05d0kh201gn49cby1y8rzj6300m7z-gtk+-3.24.43/lib/gtk-3.0/3.0.0/modules/libcanberra-gtk-module.so", F_OK) = -1 ENOENT (No such file or directory)
access("/gnu/store/n8d05d0kh201gn49cby1y8rzj6300m7z-gtk+-3.24.43/lib/gtk-3.0/3.0.0/modules/libcanberra-gtk-module.la", F_OK) = -1 ENOENT (No such file or directory)
access("/gnu/store/n8d05d0kh201gn49cby1y8rzj6300m7z-gtk+-3.24.43/lib/gtk-3.0/linux/modules/libcanberra-gtk-module.so", F_OK) = -1 ENOENT (No such file or directory)
access("/gnu/store/n8d05d0kh201gn49cby1y8rzj6300m7z-gtk+-3.24.43/lib/gtk-3.0/linux/modules/libcanberra-gtk-module.la", F_OK) = -1 ENOENT (No such file or directory)
access("/gnu/store/n8d05d0kh201gn49cby1y8rzj6300m7z-gtk+-3.24.43/lib/gtk-3.0/modules/libcanberra-gtk-module.so", F_OK) = -1 ENOENT (No such file or directory)
access("/gnu/store/n8d05d0kh201gn49cby1y8rzj6300m7z-gtk+-3.24.43/lib/gtk-3.0/modules/libcanberra-gtk-module.la", F_OK) = -1 ENOENT (No such file or directory)
```

`openscad-nightly` depends on a few qt-5 inputs which in turn all depend on `qtbase-5`.
`qtbase-5` in turn has a dependency on `gtk+` for supporting gtk+ themes.

`gtk` on guix seems to be built without the `libcanberra-gtk-module`. But as openscad is a qt application anyway, it doesn't really matter I guess.

I will have another look perhaps, but this just went down a lot on my list of priorities.

EGL support on the other hand seems to be more serious. I will give this another shot.