openscad/openscad #5890: State of Libre Graphics presentation at Libre Graphics Meeting

MA-Barbe, 2025-05-09:
Hi there!,

Sorry if this is the wrong place. I had trouble finding an email or a direct contact, please let me know if there is somewhere better for this.

I'm MA, I'm helping with the annual [LibreGraphics Meeting](https://libregraphicsmeeting.org/) (LGM) conference and am in charge of the State of Libre Graphics presentation. This opening presentation at LGM is an update from the many software, curation, publication and umbrella projects of our Libre Graphics community. I would love to be able to include OpenSCAD in the presentation!

The submission repository and instructions is located [here](https://github.com/libregraphicsmeeting/state-of-lg-2025). Basically, we're just looking for a description, what cool things the OpenSCAD team has been up to this year, and anything cool to look forward to in the future! Bonus points, I thought might be fun if the slides for OpenSCAD or whatever were also created using OpenSCAD. To give time for assembly, please have the slides in before May 21st, preferably before, the earlier the easier for me.

Of course, I understand writing slides takes time, so if no one wants to make slides, but would still like to be included, that's completely fine, I don't mind making them! Just let me know a 2-3 sentence description of OpenSCAD, and give or point to a change-log and roadmap.

Furthermore, if anyone is interested and can make it to all or any of the days May 28th-31st in Nuremberg, Germany, we would be happy to have you for LGM! Registration is free.

t-paul, 2025-05-11:
I suspect with no reply yet, it's way too late to get anything prepared. If you have some time for that, it might be possible to collect some bullet points together via chat on IRC.

https://web.libera.chat/?channel=#openscad (no registration needed, it's all volunteers, so replies may take some time)

MA-Barbe, 2025-05-12:
Ok thanks! I got to figure out my IRC (newbie problems), but will do! As soon as I stop being overwhelmed by like 10 other things randomly popping up...