openscad/openscad #5891: double-click on file to open crashes on start

tcurdt, 2025-05-11:
**Describe the bug**

On macOS when the OpenSCAD is closed I double-click on a scad file which opens OpenSCAD - which immediately crashes.

When I open OpenSCAD, then double-click on a file it opens the file as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Close OpenSCAD
2. Double-click on any `.scad` file
3. macOS shows the crash report

https://gist.github.com/tcurdt/d77f86db94732ec63ca3cb19c9271fdc

**Expected behavior**

It should just start OpenSCAD and open the file.
This used to work just fine. Last version I had installed that worked was 2024.05.24 (git 8a890a9e1)

So this must have been introduced between 8a890a9e1 and 72c9919d6.

**Code reproducing the issue**

NA

**Screenshots**

NA

**Environment and Version info (please complete the following information):**

 - OS: macOS 15.3.2
 - System: M1 pro
 - OpenSCAD Version 2025.05.09 (git 72c9919d6) (dev build)

**Library & Graphics card information**

NA

**Additional context**

`open foo.scad` unsurprisingly gives the same result.

`/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD foo.scad` surprisingly works fine.

tcurdt, 2025-05-11:
Restarting "Finder" did not help.

kintel, 2025-05-11:
Works fine here (macOS 14 on M1) using the same binary as you.
I suspect that you have some preferences, or other persistent settings enabled which I don't, which triggers this issue.

Since I cannot reproduce, it would be helpful if you could try a couple of things to see if it either changes the crash log or fixes the crash:
* Open different types of .scad designs
* Disable all features
* Enable/disable auto-reload
* View/hide different sets of windows

t-paul, 2025-05-11:
Random question: Where is the file stored? Is that on a NAS or anything other than a "normal" disk volume?

tcurdt, 2025-05-11:
> Works fine here (macOS 14 on M1) using the same binary as you.

Darn. OK.

Some things I can answer right away:

- Happens on all `.scad` files I tried. I tried many.
- The file is on the Desktop but I also tried other folders. All "normal".

One thing I am wondering is the permissions angle.

Although looking at the stack trace it looks like some global setting is causing an abort.

I'll try to dig a bit more.
I still have a pending system update and will then report back.

kintel, 2025-05-11:
Thx, I'll try to reason about it too when I have some time. We did recently change code related to setting initialization so it's possible a bug was introduced

tcurdt, 2025-05-16:
Applied all the latest system upgrade and (obviously) restarted.
Still the same problem on 15.5 (LH97Q9VM3K)

Disabled Autoload.
Still the same probelm

I then removed all plist files (AppCleaner).
Still the same problem.

The only thing custom I have left is setting the env var via launchd

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
    <key>Label</key>
    <string>setenv.OPENSCADPATH</string>
    <key>ProgramArguments</key>
    <array>
        <string>/bin/launchctl</string>
        <string>setenv</string>
        <string>OPENSCADPATH</string>
        <string>/Users/tcurdt/Projects/openscad/libraries</string>
    </array>
    <key>RunAtLoad</key>
    <true/>
</dict>
</plist>
```
But even unloading that did not help.

I scrolled through the diff https://github.com/openscad/openscad/compare/8a890a9..72c9919 ... but at least nothing jumped at me as worth digging into.

I am running out of ideas. I will try to keep updating to the latest builds.

Anything else I should try?

kintel, 2025-05-16:
I'll try if I can reproduce on macOS 15. Perhaps time to upgrade :)

tcurdt, 2026-09-26:
This seems to be really fixed now. Just for the records.