openscad/openscad #5892: Render project using GPU?

atcasanova, 2025-05-13:
**Is your feature request related to a problem? Please describe.**
Nope

**Describe the solution you'd like**
I read that OpenSCAD would render previews using GPU, but it seems to use CPU to render the preview. Why can't we use GPU aswell to render the final project? (i've just started using it today, i'm almost sure someone already asked this but for some reason i didn't find it)

atcasanova, 2025-05-13:
Yeah nevermind just found the nightlies lol

atcasanova, 2025-05-13:
OOOOOPS even nighlies doesnt seem to use my external GPU, just my intel onboard one

kintel, 2025-05-13:
We will render previews using the same GPU used to render the GUI. The GUI framework we use (Qt) will auto-select a GPU and give us a window to render into, so we have very limited opportunities to farm out anything to other GPUs.

atcasanova, 2025-05-13:
> We will render previews using the same GPU used to render the GUI. The GUI framework we use (Qt) will auto-select a GPU and give us a window to render into, so we have very limited opportunities to farm out anything to other GPUs.

isn't it possible to write code in Qt that will detect available NVIDIA GPUs and pick them instead of intel ones for example?

kintel, 2025-05-13:
Everything is possible, but not very convenient. I think the best option would be to connect your screen to your external GPU - then you'll likely use that GPU for everything on that screen.

jordanbrown0, 2025-05-14:
>Why can't we use GPU as well to render the final project?

Rendering and preview are very different.  The previewer does black magic.  It does *not* actually figure out what the resulting shape is(!).  It throws triangles at the graphics subsystem, and the graphics subsystem figures out which pixels are in front and shows them.

The rendering engine actually figures out which edges are added and removed by the various Boolean operations.  The GPU is indeed a very powerful computer, and in theory could perhaps be used for rendering, but in a very different way than it is used for previewing.  I suspect that there's a PhD thesis to be had there.