openscad/openscad #5893: Can final render be used for previews as well?

nomike, 2025-05-13:
I'm using a fairly recent build and have switched to the new manifold rendering backend.

I have recently published a quite complex design. When I open it in OpenSCAD, the preview is lagging as hell, even though I set $fn to 20.

For the F6 render with manifold, I set $fn to 64, it doesn't seem to take longer to render than with the preview render and once it's rendered, moving the camera is blazingly fast. If even keeps the colors from the preview.

Would it be possible to have a configuration option to also use the final render for previews?

I assume the distinction between preview and final render was made to have somewhat bearable performance, but with the new manifold backend, this no longer seems to be necessary (at last for the majority of cases, maybe there are some edge cases for which it makes sense to keep the current workflow).

If not, the preview render could be abolished completely, maybe the function could be kept as a stub but all it does is to call the final render function. Then over time it could be slowly phased out from the code-base, F5 could be degraded to an alias to F6 and after a few years, we will tell stories to our grandchildren about how we had to press F6 for the final render when we were young.

Any thought on this?

kintel, 2025-05-13:
Someone started on this in #5314 but it stalled.

MichaelAtOz, 2025-10-18:
4. Preview in a separate thread (#1195) & make cancellable. Manifold can take some time for complex models.

jordanbrown0, 2025-10-19:
I'd make it explicit (your # 3).  Adaptive schemes like # 1 and # 2 seem like they will lead to confusion.