openscad/openscad #5896: Build doesn't fail if nettle is not found but -DENABLE_PYTHON=ON

nomike, 2025-05-13:
**Describe the bug**
When defining -DENABLE_PYTHON=ON during building but not having the nettle dependency available, the build does not fail.

This is due to this code in [/CMakeLists.txt](https://github.com/openscad/openscad/blob/72c9919d63116f8e711f3566ae34e9eb63a2d6e6/CMakeLists.txt#L468):

```cmake
if(ENABLE_PYTHON)
  find_package(Python REQUIRED COMPONENTS Interpreter Development)
  find_package(Nettle 3.4)
  if(NOT Nettle_FOUND)
    message(WARNING "Nettle not found, disabling python support.")
    set(ENABLE_PYTHON OFF CACHE BOOL "" FORCE)
  else()
    message(STATUS "Python ${Python_VERSION} enabled, using Nettle ${Nettle_VERSION}")
    target_include_directories(OpenSCAD PRIVATE ${Python_INCLUDE_DIRS})
    target_link_libraries(OpenSCAD PRIVATE ${Python_LIBRARIES})
    target_include_directories(OpenSCAD PRIVATE ${Nettle_INCLUDE_DIRS})
    target_link_libraries(OpenSCAD PRIVATE ${Nettle_LIBRARIES})
    add_custom_target(OpenSCADPython ALL COMMAND ${CMAKE_COMMAND} -E create_symlink openscad${SUFFIX_WITH_DASH} "openscad-python")
  endif()
endif()
```
IMHO `find_package(Nettle)` should use the `REQUIRED` flag as well. If the packager sets `-DENABLE_PYTHON=ON`, the build should fail if that module could not be built.

And is there a specific reason why there is a hard dependency on version 3.4? In guix we currently have version 3.9.1. I could of course package the older version, but I need to understand first why this is necessary.

t-paul, 2025-05-13:
Not a bug, Python so far is very much optional. But it's probably ok to change the behavior to fail as we have the dedicated ON/OFF switch.

nomike, 2025-05-13:
Ah...and I realized that `find_package` without `STRICT` defines a mininum version. So that's good.

BTW: What's the reason for not using pkg-config? Are there any issues with building om other platforms? It it just a legacy thing?

PR #5897 (2025-06-12): Make Nettle required when enabling Python support
Currently if `-DENABLE_PYTHON=ON` is set, but the required Nettle dependency is not found, python support is just disabled.

IMHO the build should fail in such a case so that the user configuring the build has the choice to either add Nettle or disable Python support instead of the CMake script deciding that for them.

Fixes #5896.
closed 2025-06-12 by t-paul