openscad/openscad #5900: Automatic Reload and Preview happens ervertime I type something

Nexusnui, 2025-05-14:
**Describe the bug**
With every character I type I the preview is generated and I get warnings/error messages, since the statement is incomplete. This also shifts the text to the right evertime an error occurs.

**To Reproduce**
Steps to reproduce the behavior:
1. Activate Design->Automatic Reload and Preview
2. Type Something

**Expected behavior**
Automatic Reload and Preview, should happen, when saving the file like in previous versions.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Framework 13 i7 Intel 11th Gen
 - OpenSCAD Version 2025.05.14 (git 83cb944e7) installed with exe installer from OpenScad page.

**Additional context**
This also happened on a version prior.

Nexusnui, 2025-05-14:
I did test it in a older version I have installed and it does not happen there.

kintel, 2025-05-14:
Duplicate of #5795

closed 2025-05-14 by kintel

Nexusnui, 2025-05-14:
Did not see that one.