openscad/openscad #5903: Nightly build does not start (black window)

markusb, 2025-05-15:
**Describe the bug**
Nightly build does not start (black window)

**To Reproduce**
Steps to reproduce the behavior:
Start openscad in a terminal:
`$ openscad-nightly`

**Expected behavior**
Openscad start up and displays the UI

**Code reproducing the issue**

**Screenshots**
    $ openscad-nightly 
    QOpenGLWidget: Failed to create context
    QOpenGLWidget: Failed to create context
    qt.qpa.backingstore: composeAndFlush: QOpenGLContext creation failed
    qt.qpa.backingstore: composeAndFlush: makeCurrent() failed

**Environment and Version info (please complete the following information):**
OS: Ubuntu 24.04.2 LTS (Noble Numbat)
AMD x86 PC
OpenSCAD Version: 2025.05.14.nightly (git 7fc32e3)

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

$ openscad-nightly --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2025.05.14.nightly (git 7fc32e3)
System information: Linux 6.8.0-58-generic #60-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 14 18:29:48 UTC 2025 x86_64 Ubuntu 24.04.2 LTS 16 CPUs 31.26 GB RAM
User Agent: OpenSCAD/2025.05.14.nightly (git 7fc32e3) (Linux x86_64; Ubuntu 24.04.2 LTS)
Compiler: GCC "13.2.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 5.6, Cartesian<Gmpq>, Extended_cartesian<Gmpq>
Python Version: Python 3.12.3
OpenCSG version: OpenCSG 1.8.1
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 5.15.13
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.80.0
lodepng version: 20230410
libzip version: 1.7.3
fontconfig version: 2.15.0
freetype version: 2.13.2
harfbuzz version: 8.3.0
cairo version: 1.18.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output, python-engine
Application Path: /usr/bin
Documents Path: /home/markus/.local/share
User Documents Path: /home/markus/Documents
Resource Path: /usr/share/openscad-nightly
User Library Path: /home/markus/.local/share/OpenSCAD/libraries
User Examples Path: /home/markus/.local/share/OpenSCAD/examples
User Config Path: /home/markus/.config/OpenSCAD
Backup Path: /home/markus/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/markus/.local/share/OpenSCAD/libraries
  /usr/share/openscad-nightly/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /home/markus/.local/share/fonts
  /usr/share/gnome/fonts
  /usr/share/cinnamon/fonts
  /home/markus/.local/share/flatpak/exports/share/fonts
  /var/lib/flatpak/exports/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /var/lib/snapd/desktop/fonts
  /home/markus/.fonts
  /usr/share/texmf/fonts/opentype/public/lm
  /usr/share/texmf/fonts/opentype/public/lm-math
  /usr/share/texmf/fonts/opentype/public/tex-gyre-math
  /usr/share/texmf/fonts/opentype/public/tex-gyre
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/markus/Library/Fonts
  /usr/share/fonts/X11
  /usr/share/fonts/cMap
  /usr/share/fonts/cmap
  /usr/share/fonts/opentype
  /usr/share/fonts/truetype
  /usr/share/fonts/type1
  /usr/share/fonts/woff
  /usr/share/fonts/xscreensaver
  /usr/share/fonts/X11/Type1
  /usr/share/fonts/X11/encodings
  /usr/share/fonts/X11/misc
  /usr/share/fonts/X11/util
  /usr/share/fonts/cmap/adobe-cns1
  /usr/share/fonts/cmap/adobe-gb1
  /usr/share/fonts/cmap/adobe-japan1
  /usr/share/fonts/cmap/adobe-japan2
  /usr/share/fonts/cmap/adobe-korea1
  /usr/share/fonts/opentype/cabin
  /usr/share/fonts/opentype/cantarell
  /usr/share/fonts/opentype/font-awesome
  /usr/share/fonts/opentype/linux-libertine
  /usr/share/fonts/opentype/malayalam
  /usr/share/fonts/opentype/mathjax
  /usr/share/fonts/opentype/noto
  /usr/share/fonts/opentype/opendyslexic
  /usr/share/fonts/opentype/porson
  /usr/share/fonts/opentype/urw-base35
  /usr/share/fonts/truetype/Gargi
  /usr/share/fonts/truetype/Gubbi
  /usr/share/fonts/truetype/Nakula
  /usr/share/fonts/truetype/Navilu
  /usr/share/fonts/truetype/Sahadeva
  /usr/share/fonts/truetype/Sarai
  /usr/share/fonts/truetype/abyssinica
  /usr/share/fonts/truetype/ancient-scripts
  /usr/share/fonts/truetype/baskerville
  /usr/share/fonts/truetype/crosextra
  /usr/share/fonts/truetype/dejavu
  /usr/share/fonts/truetype/droid
  /usr/share/fonts/truetype/femkeklaver
  /usr/share/fonts/truetype/font-awesome
  /usr/share/fonts/truetype/fonts-beng-extra
  /usr/share/fonts/truetype/fonts-deva-extra
  /usr/share/fonts/truetype/fonts-gujr-extra
  /usr/share/fonts/truetype/fonts-guru-extra
  /usr/share/fonts/truetype/fonts-kalapi
  /usr/share/fonts/truetype/fonts-orya-extra
  /usr/share/fonts/truetype/fonts-telu-extra
  /usr/share/fonts/truetype/fonts-tiresias
  /usr/share/fonts/truetype/fonts-yrsa-rasa
  /usr/share/fonts/truetype/freefont
  /usr/share/fonts/truetype/gentium
  /usr/share/fonts/truetype/gentium-basic
  /usr/share/fonts/truetype/glyphicons
  /usr/share/fonts/truetype/kacst
  /usr/share/fonts/truetype/kacst-one
  /usr/share/fonts/truetype/lao
  /usr/share/fonts/truetype/lato
  /usr/share/fonts/truetype/liberation
  /usr/share/fonts/truetype/liberation2
  /usr/share/fonts/truetype/libreoffice
  /usr/share/fonts/truetype/lohit-assamese
  /usr/share/fonts/truetype/lohit-bengali
  /usr/share/fonts/truetype/lohit-devanagari
  /usr/share/fonts/truetype/lohit-gujarati
  /usr/share/fonts/truetype/lohit-kannada
  /usr/share/fonts/truetype/lohit-malayalam
  /usr/share/fonts/truetype/lohit-oriya
  /usr/share/fonts/truetype/lohit-punjabi
  /usr/share/fonts/truetype/lohit-tamil
  /usr/share/fonts/truetype/lohit-tamil-classical
  /usr/share/fonts/truetype/lohit-telugu
  /usr/share/fonts/truetype/lyx
  /usr/share/fonts/truetype/malayalam
  /usr/share/fonts/truetype/noto
  /usr/share/fonts/truetype/open-sans
  /usr/share/fonts/truetype/padauk
  /usr/share/fonts/truetype/pagul
  /usr/share/fonts/truetype/porson
  /usr/share/fonts/truetype/quicksand
  /usr/share/fonts/truetype/roboto
  /usr/share/fonts/truetype/samyak
  /usr/share/fonts/truetype/samyak-fonts
  /usr/share/fonts/truetype/sinhala
  /usr/share/fonts/truetype/teluguvijayam
  /usr/share/fonts/truetype/tibetan-machine
  /usr/share/fonts/truetype/tiresias
  /usr/share/fonts/truetype/tlwg
  /usr/share/fonts/truetype/ttf-dejavu
  /usr/share/fonts/truetype/ttf-khmeros-core
  /usr/share/fonts/truetype/tuffy
  /usr/share/fonts/truetype/ubuntu
  /usr/share/fonts/type1/gsfonts
  /usr/share/fonts/type1/texlive-fonts-recommended
  /usr/share/fonts/type1/urw-base35
  /usr/share/fonts/woff/opendyslexic
  /usr/share/fonts/X11/encodings/large
  /usr/share/fonts/truetype/roboto/unhinted
  /usr/share/fonts/truetype/roboto/unhinted/RobotoTTF

libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
Unable to initialize EGL: EGL_NOT_INITIALIZED
Unable to create GLX context using glXCreateContextAttribsARB()
Unable to create GLX context using glXCreateNewContext()
Segmentation fault (core dumped)

**Additional context**

kintel, 2025-05-15:
It kind of looks like you don't have OpenGL drivers installed. Does any simple graphics program, like `glxgears`, work?

markusb, 2025-05-15:
Yep, you are right. My OpenGL was broken. A reboot did solve it, though. Wonderful Nvidia drivers...

However, why did openscad crash, instead of saying fix your openGL?

kintel, 2025-05-16:
Having broken OpenGL is so rare that we haven't spent time making that situation more robust

closed 2025-05-16 by kintel