openscad/openscad #5904: View > Center keybind does not function on Windows

charlesmaher, 2025-05-16:
The keybind <kbd>ctrl</kbd>+<kbd>shift</kbd>+<kbd>0</kbd> does not center the view, currently the only way is with the mouse or pressing <kbd>alt</kbd>, <kbd>v</kbd>, <kbd>n</kbd>

**To Reproduce**
Steps to reproduce the behavior:
1. Use right click to move the view away from the center
2. Press <kbd>ctrl</kbd>+<kbd>shift</kbd>+<kbd>0</kbd>

**Expected behavior**
For the view to return to the center, as it does when the View > Center menu item is used.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: 12th-gen Intel i5, x86_64
 - OpenSCAD-2025.05.15-x86-64.zip, downloaded from openscad.org development snapshots

<details>
<summary>Screenshot of menu item</summary>

[image]

</details>

charlesmaher, 2025-05-16:
I'm not sure what the issue is, other similar shortcuts work, such as <kbd>ctrl</kbd>+<kbd>shift</kbd>+<kbd>V</kbd> to "View All"

Here's the relevant code if it helps.
https://github.com/openscad/openscad/blob/168b3ec9905cb541a80099ef2848d3bdfe280ac8/src/gui/MainWindow.ui#L1261-L1268

jordanbrown0, 2025-05-16:
Windows seems to default to stealing Ctrl+Shift+0 for language switching.  (Maybe more than +0).  Perhaps we should pick a different binding.

I think finding things in Settings is considered a game at Microsoft, so they rearrange things every once in a while to make it more challenging.

However, on my particular Windows 11:
* Settings
* Time & Language
* Typing
* Advanced Keyboard Settings
* Input language hot keys
* Advanced Key Settings (tab)
* Between input languages (ignore the fact that it says "Left Alt + Shift")
* Change Key Sequence...
* In the right half, select "Not Assigned".
* OK
* OK

Here's screen snippets:

[image]

[image]

[image]

[image]

[image]

[image]

jordanbrown0, 2025-05-16:
But now I can't get it to fail again.

charlesmaher, 2025-05-16:
I can confirm that your workaround fixes it for me too, but then breaks again once the computer is rebooted. Mine was already set to "Not Assigned" so it's definitely a Microsoft bug.

jordanbrown0, 2025-05-16:
I haven't rebooted to check, but some of the stuff I read talked about the problem coming back on reboot so I'm not surprised.

So it seems like our alternatives are:
* It's a Windows bug, and we should tell people to talk to MS.
* Pick a different key binding.

jordanbrown0, 2026-05-08:
See also #2977 User Defined Key Bindings.