openscad/openscad #5909: OpenSCAD reports "Python is not enabled" when run with "-o" on a python file

nomike, 2025-05-16:
**Describe the bug**

When running OpenSCAD on a python file with the "-o" option, an error is printed to the terminal and the file is not rendered.

**To Reproduce**

Steps to reproduce the behavior:
1. Create a simple python file and store it as `cube.py`:

```python
from openscad import*

c = cube(10)
show(c)
```
2. Launch OpenSCAD with this command: `openscad cube.py`
    The GUI opens and the file is rendered es expected.

3. Launch OpenSCAD with this command: `openscad cube.py -o cube.stl`
    Openscad will fail with:

```plaintext
Python is not enabled
ERROR: Parser error: syntax error in file ../../tmp/cube.py, line 1
Can't parse file '/home/nomike/tmp/cube.py'!
```

**Expected behavior**

The python file should be rendered as expected.

** Additional Notes **

I tried adding `--enable=python-engine`, but it resulted in the same error.
But even if it were to work with that parameter, using the `-o` option should not influence feature toggle.

**Environment and Version info (please complete the following information):**

```plaintext
$ openscad --info
OpenSCAD Version: 2025.05.08-1.72c9919 (git 72c9919d63116f8e711f3566ae34e9eb63a2d6e6)
System information: Linux 6.11.0-25-generic #25-Ubuntu SMP PREEMPT_DYNAMIC Fri Apr 11 23:29:18 UTC 2025 x86_64 Ubuntu 24.10 12 CPUs 45.28 GB RAM
User Agent: OpenSCAD/2025.05.08-1.72c9919 (git 72c9919d63116f8e711f3566ae34e9eb63a2d6e6) (Linux x86_64; Ubuntu 24.10)
Compiler: GCC "11.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 5.6.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>
Python Version: Python 3.12.2
OpenCSG version: OpenCSG 1.8.1
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 5.15.15
QScintilla version: 2.13.4
InputDrivers: 
GLib version: 2.82.1
lodepng version: 20230410
libzip version: 1.9.2
fontconfig version: 2.14.0
freetype version: 2.13.0
harfbuzz version: 8.3.0
cairo version: 1.18.2 (runtime: 1.18.4)
lib3mf version: 2.2.0
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output, python-engine
Application Path: /gnu/store/79smfsprf3mg7n2bqjn8q2k66rlq3vl5-openscad-2025.05.08-1.72c9919/bin
Documents Path: /home/nomike/.local/share
User Documents Path: /home/nomike/Documents
Resource Path: /gnu/store/79smfsprf3mg7n2bqjn8q2k66rlq3vl5-openscad-2025.05.08-1.72c9919/share/openscad
User Library Path: /home/nomike/.local/share/OpenSCAD/libraries
User Examples Path: /home/nomike/.local/share/OpenSCAD/examples
User Config Path: /home/nomike/.config/OpenSCAD
Backup Path: /home/nomike/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/nomike/.local/share/OpenSCAD/libraries
  /gnu/store/79smfsprf3mg7n2bqjn8q2k66rlq3vl5-openscad-2025.05.08-1.72c9919/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /gnu/store/mjg1gq1w6csggjhy5wh6njyzqpic4avq-font-dejavu-2.37/share/fonts
  /home/nomike/.local/share/fonts
  /home/nomike/.local/share/flatpak/exports/share/fonts
  /var/lib/flatpak/exports/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /var/lib/snapd/desktop/fonts
  /gnu/store/79smfsprf3mg7n2bqjn8q2k66rlq3vl5-openscad-2025.05.08-1.72c9919/share/fonts
  /gnu/store/fvfzcrgkn4jvqafhrzm8a6kwxzdjpz72-glib-2.82.1/share/fonts
  /home/nomike/.fonts
  /home/nomike/.guix-home/profile/share/fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/nomike/Library/Fonts
  /gnu/store/mjg1gq1w6csggjhy5wh6njyzqpic4avq-font-dejavu-2.37/share/fonts/truetype
  /usr/share/fonts/X11
  /usr/share/fonts/cMap
  /usr/share/fonts/cmap
  /usr/share/fonts/fonts-go
  /usr/share/fonts/opentype
  /usr/share/fonts/truetype
  /usr/share/fonts/type1
  /usr/share/fonts/woff
  /home/nomike/.guix-home/profile/share/fonts/opentype
  /home/nomike/.guix-home/profile/share/fonts/truetype
  /usr/share/fonts/X11/Type1
  /usr/share/fonts/X11/encodings
  /usr/share/fonts/X11/misc
  /usr/share/fonts/X11/util
  /usr/share/fonts/cmap/adobe-cns1
  /usr/share/fonts/cmap/adobe-gb1
  /usr/share/fonts/cmap/adobe-japan1
  /usr/share/fonts/cmap/adobe-japan2
  /usr/share/fonts/cmap/adobe-korea1
  /usr/share/fonts/opentype/artemisia
  /usr/share/fonts/opentype/asana-math
  /usr/share/fonts/opentype/cabin
  /usr/share/fonts/opentype/cantarell
  /usr/share/fonts/opentype/comic-neue
  /usr/share/fonts/opentype/complutum
  /usr/share/fonts/opentype/didot
  /usr/share/fonts/opentype/ebgaramond
  /usr/share/fonts/opentype/font-awesome
  /usr/share/fonts/opentype/freefont
  /usr/share/fonts/opentype/inter
  /usr/share/fonts/opentype/ipaexfont-gothic
  /usr/share/fonts/opentype/ipaexfont-mincho
  /usr/share/fonts/opentype/ipafont-gothic
  /usr/share/fonts/opentype/ipafont-mincho
  /usr/share/fonts/opentype/linux-libertine
  /usr/share/fonts/opentype/lobster
  /usr/share/fonts/opentype/lobstertwo
  /usr/share/fonts/opentype/malayalam
  /usr/share/fonts/opentype/neohellenic
  /usr/share/fonts/opentype/noto
  /usr/share/fonts/opentype/olga
  /usr/share/fonts/opentype/porson
  /usr/share/fonts/opentype/roboto
  /usr/share/fonts/opentype/solomos
  /usr/share/fonts/opentype/stix
  /usr/share/fonts/opentype/stix-word
  /usr/share/fonts/opentype/urw-base35
  /usr/share/fonts/truetype/Gargi
  /usr/share/fonts/truetype/Gubbi
  /usr/share/fonts/truetype/Nakula
  /usr/share/fonts/truetype/Navilu
  /usr/share/fonts/truetype/Sahadeva
  /usr/share/fonts/truetype/Sarai
  /usr/share/fonts/truetype/adf
  /usr/share/fonts/truetype/andika
  /usr/share/fonts/truetype/annapurna
  /usr/share/fonts/truetype/arphic
  /usr/share/fonts/truetype/baskerville
  /usr/share/fonts/truetype/charis
  /usr/share/fonts/truetype/clear-sans
  /usr/share/fonts/truetype/comfortaa
  /usr/share/fonts/truetype/croscore
  /usr/share/fonts/truetype/crosextra
  /usr/share/fonts/truetype/dejavu
  /usr/share/fonts/truetype/droid
  /usr/share/fonts/truetype/ebgaramond
  /usr/share/fonts/truetype/font-awesome
  /usr/share/fonts/truetype/fonts-beng-extra
  /usr/share/fonts/truetype/fonts-deva-extra
  /usr/share/fonts/truetype/fonts-gujr-extra
  /usr/share/fonts/truetype/fonts-guru-extra
  /usr/share/fonts/truetype/fonts-kalapi
  /usr/share/fonts/truetype/fonts-orya-extra
  /usr/share/fonts/truetype/fonts-telu-extra
  /usr/share/fonts/truetype/fonts-yrsa-rasa
  /usr/share/fonts/truetype/freefont
  /usr/share/fonts/truetype/gentium
  /usr/share/fonts/truetype/gentium-basic
  /usr/share/fonts/truetype/gentiumplus
  /usr/share/fonts/truetype/gentiumplus-compact
  /usr/share/fonts/truetype/lato
  /usr/share/fonts/truetype/liberation
  /usr/share/fonts/truetype/libreoffice
  /usr/share/fonts/truetype/lohit-assamese
  /usr/share/fonts/truetype/lohit-bengali
  /usr/share/fonts/truetype/lohit-devanagari
  /usr/share/fonts/truetype/lohit-gujarati
  /usr/share/fonts/truetype/lohit-kannada
  /usr/share/fonts/truetype/lohit-malayalam
  /usr/share/fonts/truetype/lohit-oriya
  /usr/share/fonts/truetype/lohit-punjabi
  /usr/share/fonts/truetype/lohit-tamil
  /usr/share/fonts/truetype/lohit-tamil-classical
  /usr/share/fonts/truetype/lohit-telugu
  /usr/share/fonts/truetype/lyx
  /usr/share/fonts/truetype/malayalam
  /usr/share/fonts/truetype/msttcorefonts
  /usr/share/fonts/truetype/noto
  /usr/share/fonts/truetype/open-sans
  /usr/share/fonts/truetype/pagul
  /usr/share/fonts/truetype/paratype
  /usr/share/fonts/truetype/roboto
  /usr/share/fonts/truetype/samyak
  /usr/share/fonts/truetype/samyak-fonts
  /usr/share/fonts/truetype/teluguvijayam
  /usr/share/fonts/truetype/tuffy
  /usr/share/fonts/truetype/ubuntu
  /usr/share/fonts/type1/texlive-fonts-recommended
  /usr/share/fonts/type1/urw-base35
  /usr/share/fonts/woff/ebgaramond
  /usr/share/fonts/X11/encodings/large
  /usr/share/fonts/opentype/roboto/slab
  /usr/share/fonts/truetype/roboto/unhinted
  /usr/share/fonts/truetype/roboto/unhinted/RobotoTTF

GL context creator: EGL (new)
EGL version: 1.5
PNG generator: lodepng

GLAD version: 2.0.4
OpenGL Version: 4.6 (Compatibility Profile) Mesa 24.3.2
GL Renderer: AMD Radeon Graphics (radeonsi, renoir, LLVM 18.1.8, DRM 3.59, 6.11.0-25-generic)
GL Vendor: AMD
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

gsohler, 2025-05-16:
please add

--trust-python

and it will work fine

nomike, 2025-05-16:
> please add
> --trust-python and it will work fine

Thanks a lot. That works.

I consider this to be the equivalent of the "Do you trust this script?" dialog presented in the GUI, so it's also fine from a user-interface point of view.

closed 2025-05-16 by nomike