openscad/openscad #5913: Special variable $arguments for debugging purposes

caufang, 2025-05-17:
Currently there is no way (that I'm aware of) to simply see the values of all parameters passed to a module/function.
You have to create an `echo()` explicitly specifying each of the parameters, which can get annoying.

I propose a special variable `$arguments`/`$parameters` that provides a structure of key=>value pairs of the current parameters, similar to what `textmetrics()` returns, or imported JSON.

I have no idea how complicated this would be in the program code, and apologize if this is an burdensome request.

t-paul, 2025-05-17:
I suggest to rephrase this without the reference to special variables. Those have too many issues attached that it's very unlikely to happen this way.

As for the feature itself, how would that be used? Do you have any examples of awkward code that could be improved with that feature? There is indeed not much debug support, but it feels like this is a very special case that would only help in rare cases.

jordanbrown0, 2025-05-18:
I've been thinking about this a bit, and see a couple of uses.

- Varargs. You want a module or function that takes an unlimited list of parameters, like max() or echo().
- In conjunction with a spread operator, a way to "wrap" an underlying function or module, accepting a list of arguments and passing them transparently through to the underlying function or module without needing to know the full list, and without the troubles associated with the need to *not* pass omitted parameters.
- A desire to pass a name-value list where the names are not specified in advance - like echo().

One problem is OpenSCAD's split personality, positional vs named parameters.  Is $arguments (or whatever) a list, or an object?

Another is that OpenSCAD's argument processing is even less structured than that.  Note that echo() has named arguments that are also order-sensitive, and that echo() allows the same name to be specified more than once.

You can simulate a lot of this with other data structures, of course, passing lists or, with #4478, objects instead of distinct arguments.  But those schemes make such functions be second class citizens.

yahbluez, 2025-05-18:
Having argc, argv would be very useful to write much cleaner and more stable code. It enhances the universal use of modules. 

And to have a look at the special handling in echo()
what about a key=val representation.

module foo(){
    echo(argc=argc, argv=argv, kwargs=kwargs);
}

foo(1, 2, bla=4, "fish", blub="blob");

OUTPUT:
argc=5, argv=[1, 2 , "bla=4", "fish", "blub=blob"], kwargs={"bla": 4, "blub":, "blob"}

generally i like to see more json than CSV in openscad to handle external data.
Adding features is the easy way to stay backward compatible.

caufang, 2025-05-18:
I was thinking it would be best to keep this feature request as concise as possible.

Variadic arguments would be nice, but I don't want to try to have too many things in this same ticket. If variadics are to be implemented, I think it would be best if they were explicitly defined as `...arg_name` rather than arbitrarily defined at call-time.

I was thinking that `$arguments` would only give you the key+values of defined parameters for your module/function, and not follow the way that `echo()` works as that is a bit of a special case.

I don't think named vs positional would be important, because from within your module everything is named.

Example code for my idea:

```
test(10, 20, 5);

module test(width, height, radius = 10) {
  
  echo($arguments);
  
  echo($arguments.width);
  
}
```
Console output:
```
ECHO: { height = 20; radius = 5; width = 10; }
ECHO: 10
```
The order of the properties here is because I was using a sample JSON file to get proper console formatting, and the JSON importer appears to alpha-sort the property names.

yahbluez, 2025-05-18:
> ECHO: { height = 20; radius = 5; width = 10; }

we should not break the openscad syntax
 so why ';' instead of the usual ','

and why another data representation that is **incompatible** with the rest of the world for no need. 

Just your idea but in clean JSON:

{ "height": 20,  "radius": 5,  "width": 10 }

handling external data get's more and more important for openscad and going the JSON way the DIR way  makes it easy to handle whatever data comes.

caufang, 2025-05-18:
I just copied the output from the OpenSCAD console when echoing out a piece of imported JSON.

The console output syntax is not my idea, it's just how OpenSCAD outputs structured data.

Here is what I wrote to simulate my idea:

arguments_example.json :
```json
{"width": 10, "height": 20, "radius": 5}
```

arguments_example.scad
```
$arguments = import("./arguments_example.json");

test(10, 20, 5);

module test(width, height, radius = 10) {
  
  echo($arguments);
  
  echo($arguments.width);
  
}
```

jordanbrown0, 2025-05-18:
@caufang ...

>If variadics are to be implemented, I think it would be best if they were explicitly defined as ...arg_name rather than arbitrarily defined at call-time.

You're right that for varargs a Python-style
```
module foo(fixed1, fixed2, ...varargs)
```
scheme would be better than a JavaScript-style magic variable.

>I was thinking that $arguments would only give you the key+values of defined parameters for your module/function,  [...] I don't think named vs positional would be important, because from within your module everything is named.

Yes... until you start talking about varargs, when the picture gets much murkier.  And if you're not talking about varargs, if you're only talking about the arguments that are explicitly defined, then the value of the feature gets really really small, because it only gives you access to something you already had access to.

>not follow the way that echo() works as that is a bit of a special case.

Yes, it is, in four distinct ways:
- It accepts an arbitrary number of arguments.
- Its named arguments are order-sensitive; it prints them in the order presented.
- It can take arbitrary named arguments, not just those in its declaration.
- It can take the same name argument more than once, and keeps them distinct.

@yahbluez ...
> Having argc, argv would be very useful to write much cleaner and more stable code.

Use cases would be helpful here.  In what way would it make your program cleaner and more stable?

Note:  argc is not needed, because `len(argv)` would work.

caufang, 2025-05-18:
Maybe it would be better/simpler to instead of a special variable that is readable, to instead have a `echo_args()` function.

I'm thinking `echo_args()` would output the the module/function name, along with the parameters, similar to how `echo()` works.

I think adding a single parameter to the function to allow for "pretty printing", just having each argument show on a separate line.

I was thinking that `echo_args()` would more function like a pre-execution macro, essentially being replaced at runtime with a call to `echo()`.

Here is my idea, though ideally there would be cleaner output in the console:

```scad
/**
 * @param bool pretty
 *   'true' means to output each argument on a separate line.
 */
module echo_args(pretty = false) {}

test(10, 20, 5);

module test(width, height, radius = 10) {
  
  echo_args();
  
  echo("module test(", width = width, height = height, radius = radius, ")");
  
  
  echo_args(true);
  
  echo("module test(", 
    "\n", width = width, 
    "\n", height = height, 
    "\n", radius = radius, 
  "\n)");
}
```

caufang, 2025-05-18:
Current console output:

```
ECHO: "module test(", width = 10, height = 20, radius = 5, ")"
ECHO: "module test(", "
", width = 10, "
", height = 20, "
", radius = 5, "
)"
```

marcuserronius, 2025-06-09:
> Another is that OpenSCAD's argument processing is even less structured than that. Note that echo() has named arguments that are also order-sensitive, and that echo() allows the same name to be specified more than once.

In the spirit of KISS, maybe a simpler solution could use the less-cooked argument data (which could also result in echo()-like capability and retain parameter positions) and make it a special function that takes an optional argument:
```
module test(width, height, radius) {
  echo(argv());
  echo(argv("width"));
}
test(10,20,5);
// ECHO: [["width",10],["height",20],["radius",5]]
// ECHO: 10
```
If we're considering hypothetical future implementation of variadic functions and modules, this could be extended with numeric indices for unnamed parameters. Put these together (with a hypothetical print() function and spreading syntax), and we get:
```
function my_echo(...)
  [for(param=argv()) print(param[0],":", param[1], " ")]
my_echo(this=10, that=20, 30, this=40)
// output: this:10 that:20 2:30 this:40

module MyCylinder(h=1, r1=1, r2=1, center, ...) {
  r1 = argv("d1") ? argv("d1")/2 :
    argv("d") ? argv("d")/2 :
    argv("r") ? argv("r") : 
    r1
  // or with a little extension to return just the first match:
  r1 = argv("d1","d") ? argv("d1","d")/2 : argv("r1","r")
}
```

yahbluez, 2025-06-10:
@marcuserronius 

We all ready started with textmetrics() to make use of todays way to structure data.
While a key value list is better than nuts it would be much more straightforward
to use JSON with all his advantages and remove a barrier for all than new users
which are common using this in any other language.

look how tiny the difference is that can make a huge step forward:

module test(width, height, radius) {
  echo(argv());
  echo(argv("width"));
  
  echo(argv.width);
}
test(10,20,5);
// ECHO: {"width":10,"height":20,"radius":,5}
// ECHO: 10
// ECHO: 10

marcuserronius, 2025-06-19:
I really like the object/dictionary proposal from #4478. My one objection (which is a very small, probably inconsequential one) is that I don't see how it would allow echo-compatible arguments, which I think could be useful. Unless I'm reading it wrong, objects wouldn't support multiple keys with the same name, and it's unclear if they'd preserve key ordering. Enabling echo-compatible arguments would be nice, I can think of a couple good use cases for them. Nothing that couldn't be done another way, but access to that could help make some modules or functions a little simpler to use and understand. One example off the top of my head where this would be useful:
'''
// Module for simple compound rotations; treats arguments as rotate() operations to be performed sequentially
module Turn(...args) {
  // slice off the first one
  first = args[0];
  remains = [for(i=[1:len(args)-1]) args[i]];
  key = first[0];
  value = first[1];
  // which axis
  axis = key=="x"? [1,0,0]: 
    key=="y"? [0,1,0]:
    key=="z"? [0,0,1]:
    [0,0,0];
  if(len(remains)==0) {
    rotate(axis*value) children();
  } else {
    Turn(...remains) rotate(axis*value) children();
  }
}

Turn(z=60, x=90, z=90) cylinder(10, 4, $fn=3); // turn 60° around z, then 90° around x, then 90° around z again
'''
One could calculate or use trial and error to turn this into a single 'rotate([x,y,z])' command, but it would be considerable extra work, as well as difficult for others to parse what's happening; and using multiple single axis rotations gets too wordy rather quickly. Sure you could make it take a list of operations, ie. 'Turn([["z",60],["x",90],["z",90]])', but that's significantly less readable to me.

Still, this is only in my "that would be nice" category, and I have no idea how difficult it would be to implement it with those options.

marcuserronius, 2025-06-19:
I really like the object/dictionary proposal from #4478. My one objection (which is a very small, probably inconsequential one) is that I don't see how it would allow echo-compatible arguments, which I think could be useful. Unless I'm reading it wrong, objects wouldn't support multiple keys with the same name, and it's unclear if they'd preserve key ordering. Enabling echo-compatible arguments would be nice, I can think of a couple good use cases for them. Nothing that couldn't be done another way, but access to that could help make some modules or functions a little simpler to use and understand. One example off the top of my head where this would be useful:
```
// Module for simple compound rotations; treats arguments as rotate() operations to be performed sequentially
module Turn(...args) {
  // slice off the first one
  first = args[0];
  remains = [for(i=[1:len(args)-1]) args[i]];
  key = first[0];
  value = first[1];
  // which axis
  axis = key=="x"? [1,0,0]: 
    key=="y"? [0,1,0]:
    key=="z"? [0,0,1]:
    [0,0,0];
  if(len(remains)==0) {
    rotate(axis*value) children();
  } else {
    Turn(...remains) rotate(axis*value) children();
  }
}

// rotate 60° around z, then 90° around x, then 90° around z again
Turn(z=60, x=90, z=90) cylinder(10, 4, $fn=3); // turn 60° around z, then 90° around x, then 90° around z again
```
One could calculate (or use trial and error) to turn this into a single `rotate([x,y,z])` command, but it would be considerable extra work, as well as difficult for others to parse what's happening. The alternative, using multiple single axis rotations, gets too wordy rather quickly. Sure, you could make it take a list of operations, ie. `Turn([["z",60],["x",90],["z",90]])`, but that's significantly less readable to me.

Still, this is only in my "that would be nice" category, and I have no idea how difficult it would be to implement it with those options.

yahbluez, 2025-06-20:
@marcuserronius if we use a JSON for object instead of reinventing the wheel there is no problem to have a echo() like way to take parameters. There is nothing wrong in using a well established Object notation. It solves any issue shown in this thread and opens up the door for openscad to get use of external data. Also have in mind that the python addon in openscad dev version can use json in any thinkable way.

jordanbrown0, 2025-06-20:
JSON per se doesn't help or hurt the ability to represent echo-like arguments, because JSON is only a textual representation of a data structure and does not discuss the internal representation of the data structure or its semantics.

There are three ways in which echo-like arguments are ... difficult:
* order of named arguments matters.  echo(x=1, y=2) produces different results from echo(y=2, x=1).
* named arguments need not be unique.  echo(a=1, a=2) is perfectly legal and outputs two values.
* ordering of named and unnamed arguments matters. echo(3, x=4) produces different results from echo(x=4, 3).

JSON does not directly offer a data structure that mixes named and unnamed values, so can't direct represent that last case.

JSON per se does not prohibit duplicate names, but it also does not guarantee that they are allowed. The canonical instance of JSON, the JavaScript language itself, does not allow duplicate names in an object. 

JSON per se does not address whether order of named values is preserved. The text at json.org says "unordered"; the ECMA standard leaves the question to implementations. Until 2015, JavaScript objects were explicitly unordered; after 2015 there is a specific order that isn't all that obvious.  (But I think the 2015 ordering is FIFO for all "echo" cases.)

So, anyhow, "just use JSON" doesn't directly answer the question of how to represent echo-like arguments lists, or even whether it's possible to represent them as data structures.

So what would I do?

I would start by offering mechanisms that return the named arguments as an object and the unnamed arguments as a list, preserving the order of the unnamed arguments but not the order of the named arguments or the order between the two.  Those would provide convenient access for the most common cases.  If need be, I would then provide a third mechanism that lets you inspect the argument list in all of its gory detail, one element at a time, in caller order, with and without names - probably as a list of [name, value] pairs where name is `undefined` for an unnamed argument.

yahbluez, 2025-06-20:
@jordanbrown0 

{
  "arguments": [
    {
      "type": "arg",
      "value": <value>
    },
    {
      "type": "kwarg",
      "key": <key>,
      "value": <value>
    }
  ]
}

maybe I did not see a point but this JSON structure handles any of this points.

arguments holds every parameter in his given order,
each argument is either a arg,  just a value
or a kwarg with a key and a value.

Did I miss something?
I think any input of any function could be represented in that way.
There is no limit about the parameters, order is preserved, 
and keys can be used several times with several values because the order matters.

yahbluez, 2025-06-20:
"named arguments need not be unique. echo(a=1, a=2) is perfectly legal and outputs two values."

Is this echo() only? Do other openscad functions, modules allow that?

I do not understand how that may work in a function that did not just use all parameters in the order they came in like echo() did.

module foo(bar=2, bar=3){echo(bar);} foo(); => ECHO: 2
The second bar_3 is ignored, tested with the nightly snap.

So if the multiple use of a keyword and the order of multiple keywords only happens in echo() that reduces the issue a lot and would make the JSON object even more easy.

jordanbrown0, 2025-06-21:
`echo()` is magic and does things that a conventional OpenSCAD module cannot do:
* accepts an arbitrary number of named and unnamed arguments
* accepts named arguments with arbitrary (not declared by the module) names
* preserves order of named arguments
* preserves order of mixtures of named and unnamed arguments
* allows duplicated names

Maybe I was reading stuff into the request, but it seemed like a goal would be to allow one to write one's own wrapper around echo().

We would need a way to tell OpenSCAD "let this module take an unrestricted list of arguments", and then one or more ways to report that list.

The data structures are fairly easy:
* Reporting a list of unnamed arguments is easy:  a list.
* Reporting a list of named arguments, with no duplicates and order not maintained, is easy:  an object.
* Reporting a mixed list of named and unnamed arguments, with duplicates, with order maintained, is not hard:  a list of [name, value] pairs.  This can represent anything, but is not at all convenient so should not be the primary representation.

The question is how you retrieve those data structures.

If I had to take a stab at these, I would borrow from Python and say that the declaration could include `*argv`, to get a list of the unnamed arguments, `**argobj` to get an object containing all of the named arguments, or (adding to Python) `***rawargv` to get a list of [name,value] pairs.  Absent `*` or `***`, it would be an error to pass more unnamed arguments than are declared.  Absent `**` or `***`, it would be an error to pass undeclared named arguments.  Absent `***`, it would be an error to duplicate names.  Similarly, in a call `*argv` would pass the contents of a list as a series of unnamed arguments, `**argobj` would pass the contents of the object as a set of named arguments, and `***rawargv` would turn a list of [name,value] pairs into a series of named and unnamed arguments.

There are other schemes that would work - in particular, there could be one or more functions.  In theory there could be magic variables, but I think I'd recommend against that.

yahbluez, 2025-06-21:
I fully understand that changing the internal data structure for parameters in an existing application is a task that needs a lot of thinking, discussing and planning upfront to find a way to enhance the possibilities without breaking stuff.

Your view helped me a lot to see the issues facing such a task. 

My opinion / path to a solution:

Let echo() do what echo() did, no need to change anything, no need to borrow anything from the (ugly) way echo() handel his very own parameter list.

Let enhance the internal parameter data structure of all other function / module declarations.
I **highly highly** recommend to make use of the power of JSON for this step, there is not a single con but many pro for this idea.

Unfortunately I cannot offer the time to help active to write that code, I do not have a deep view into openscad source.
But I'm willing to discuss, motivate and give my knowledge to help for a solution. Seeing from the outside of the box into problems often shows ways one deeply involved can't see.

The/A way such a change of parameter handling could happen:

I guess there is a central part in openscad that handle parameters for functions and modules and one that handle echo().
In this part of code the extension to JSON has to happen first. (we ignore echo())
In a way that the calling stuff can still get what it is used to get but newer or updates of internal or external functions / modules may get access to the new JSON parameter object.

**That way nothing old break.**

Around / included with this solution new JSON handling stuff like the object we already have with textmetrics() can be added.

Step todo, in the code that handle the parameters, additional code is needed to create and fill an JSON object like this:

```
{
  "args": [<value>, <value>, ...],
  "kwargs": {
    "<key>": <value>,
    "<key>": <value>,
    ...
  },
  "keys": [<key>, <key>, ...]
}
```

Let us name that object "this" and be a visibly object in the scope of the function / module.
(Any name works i just use this to have a name now while talking about the object)

At this nothing changes for any older function / module beside that they have another object "this" in her scope.

Using the same method used for the textmetrics object inside the function / module the new object could be used as:

this.args => a list of all values in the order the are given [...]
this.kwargs.key => the value of the named parameter "key"
this.keys => a list of all named parameters in the order they are given.

**Example**

```
module foo(1, 2, bar=5, bug="Hello", 11, some="else" ){
    echo(this.args);
    echo(this.keys);

    for(key = keys){
        echo(this.kwargs.key);
    }
}

ECHO: [1, 2, 11]
ECHO: ["bar", "bug", "some"]

ECHO: 5
ECHO: "Hello"
ECHO: "else"

```

Moving to the dynamic of an internal JSON structure, where we have tons of  libs to use them in any language, opens up the door to whatever idea is added in the future and also will help a lot for the new embedded python part. For all of our jungsters is using JSON a natural way to handle data and that may motivate more to come to openscad. 

I understood that any kind if data handling trough openscad harms somehow the "don't be able to harm" rule, but that rule get's a little bit obsolete with the python embedding. 

Imagine how powerful and universal  modules can be written if they can take or return a JSON object.
That opens the door to have modules that do not stop function if newer versions add stuff or change parameters.
I see tons of overhead coding in BOSL2 to circle around this limitations today.

Never mind that's enough for now.
Thank you for reading.

jordanbrown0, 2025-06-21:
It really isn't as complicated as you think.  The internal data structures are plenty good enough for all of these purposes.  The magic of echo() is that it *doesn't* do a bunch of processing that everything else does.  Argument lists are *already* stored as arrays of name-value pairs, with the names being optional.

We just need the language features to expose those lists to OpenSCAD programs *without* doing the checking that is currently done.

The real roadblock is that I don't want to spend more effort on object/dictionary-related stuff until PR#4478 gets integrated.

yahbluez, 2025-06-21:
Great to read that the implementation may not be a big deal.

Not so great to see that an 3y old PR is still blocking so much development.

Maybe just adding the JSON object and the most basic way to use them like we already have with textmetrics() brings a new drive into the PR4478?

There is nothing wrong and it is a wise move to adept well developed features from other languages. 
Languages that uses objects since their very beginning like python where everything is object.
The python way to handle lists is awesome and a python dictionary is the same as an JSON object.
Many good stuff and rocks to stand on can be adopted from python. 

And the parameters are an easy fast to establish backward compatible way to enter this step and at the same time demonstrate the way for the PR and just starts with the first step.

If we have a json object as universal parameter access method more stuff and discussions how to go further with object will come up fast and this is also a red line to solve PR4478.

yahbluez, 2025-06-23:
@marcuserronius 

```
{
  "": 23,
  "": 42,
  "id": "bar",
  "id": "baz"
}

```

This is not a valid JSON and would break things instead of making them better.

As i told, do not reinvent the wheel use existing well proved technics. 
The two minimal more complex JSON structures i showed up are valid and can handle even echo().