openscad/openscad #5915: Documentation has `let()` as a mathematical function

jordanbrown0, 2025-05-18:
The documentation (cheat sheet and [user manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Mathematical_Functions#let)) has let() as a mathematical function.  It isn't particularly mathematical, and it isn't exactly a function.

**Background**
There are three magic pseudo-functions:  `let()`, `echo()`, and `assert()`.  They are magic in that they take an extra "child" argument *after* the function-style invocation.

**Issue**
These are not normal functions, and should not be lumped in with the normal functions.  They should get their own section and their own discussion, much like the list-comprehension elements have.

In addition they are not particularly "mathematical", and so doubly shouldn't be in that section.

**Action**
I think they need to be pulled into their own section, in both the User Manual and the cheat sheet.  I'm happy to do that work, but wanted to get a little consensus first.

bitbasher, 2025-07-18:
i am in the process of doing just that .. the incomplete draft of the new section is called [Syntactic Modules](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Syntactic_Modules) 'cause i think let() etc look and work more like modules than anything else.
but i am happy to hear from anyone who can replace that section title as being pretty lame

jordanbrown0, 2025-07-19:
They aren't modules either, so I would not use that word.  Perhaps the best word would be "operators".

This is a reference manual, so I would minimize "why you might want to do this" text.  Describe what it does, give an example, and let the reader figure out what they want to use it for.  Anything more belongs in tutorial material, not in a reference manual.

jordanbrown0, 2026-09-19:
Unstale.