openscad/openscad #5916: `let()` statement is no longer needed; documentation should downplay it

jordanbrown0, 2025-05-18:
The [User Manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Conditional_and_Iterator_Functions#Let_Statement) has a section on the `let()` statement.  The `let()` statement is no longer needed; you can just do assignments.  The manual should describe it, so that a reader can understand historical programs, but should point out that it is unnecessary.

That is, the example:
```
for (i = [10:50])
{
    let (angle = i*360/20, r= i*2, distance = r*5)
    {
        rotate(angle, [1, 0, 0])
        translate([0, distance, 0])
        sphere(r = r);
    }
}
```
can now be written as
```
for (i = [10:50])
{
    angle = i*360/20;
    r= i*2;
    distance = r*5;

    rotate(angle, [1, 0, 0])
        translate([0, distance, 0])
        sphere(r = r);
}
```

The one thing that `let()` can do that assignment can't do is to create a scope:
```
let(a=1) { echo(a=a); }
let(a=2) { echo(a=a); }
```
but that's a somewhat unusual use case and can be done using any *other* module, e.g. union() or group().

**Action**
I'm happy to make the change, but wanted some consensus first.

I am *not* suggesting that `let()` be formally deprecated and generate warnings, though that might be something to do in the future to reduce clutter in the language.

nophead, 2025-05-18:
Let is needed in functions.

jordanbrown0, 2025-05-18:
`let()` in expressions (which includes in functions) is not a statement and is a different section of the manual.  (And see #5915 about how they aren't really documented right.)

There have historically been some oddities in how assignments in children work, but I think they've been addressed.  Regardless, any such oddities can be addressed using any other scope-creating constructure; you don't need let() for it.

jordanbrown0, 2025-05-18:
let() in expressions is documented [here](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Mathematical_Functions#let).

nophead, 2025-05-18:
I used to use assign() but that was deprecated so I switched to for() then let was added so I use that.  Here is an example.
```
module camera_housings()
    layout([for(c = cameras) pcb_length(camera_pcb(c))], 15, false) let(c = cameras[$i])
        if(pcb_screw(camera_pcb(c)))
            camera_fastened_assembly(c, 3);
```

I don't think $variables work with an ordinary assignment

jordanbrown0, 2025-05-18:
I didn't say that it wouldn't work, but rather that it was never necessary.

```
module camera_housings()
    layout([for(c = cameras) pcb_length(camera_pcb(c))], 15, false) {
        c = cameras[$i];
        if(pcb_screw(camera_pcb(c)))
            camera_fastened_assembly(c, 3);
    }
```

> I don't think $variables work with an ordinary assignment

Sure they do.
```
$fn = 4;
echo($fn);
circle(10);
```

jordanbrown0, 2025-05-18:
Ah, I hadn't caught the $i in there.

Yes, that would be hit by the old behavior of assignments in children, where the child assignments were executed before the parent executes.  But that's been changed; the child assignments are now executed when children() is called.

And even in the old way, you could protect them by wrapping them in any scope-creating construct, like union():
```
module camera_housings()
    layout([for(c = cameras) pcb_length(camera_pcb(c))], 15, false) union() {
        c = cameras[$i];
        if(pcb_screw(camera_pcb(c)))
            camera_fastened_assembly(c, 3);
    }
```

jordanbrown0, 2025-05-18:
I've removed my comment about $ variables working in assignments; I thought you were talking about the left side of an assignment and you were talking about using them on the right side of an assignment in a child context.

caufang, 2025-05-19:
I would say making the documentation for `let()` more verbose would be a good idea.

The example for `assign()` above it has the equivalent code without using `assign()`, so doing it for `let()` makes sense.

If we're going to show equivalence, I would want to boil down the example code as much as possible, removing unnecessary syntax that might confuse the specific functionality the documentation is trying to highlight:

Using `let()`
```scad
for (i = [10:50])
let (angle = i * 360 / 20, r= i * 2, distance = r * 5)
rotate(angle, [1, 0, 0])
translate([0, distance, 0])
sphere(r = r);
```

Possibly include an example that shows you can have multiple lines in `let()`, for readability purposes:
```scad
for (i = [10:50])
let (
  angle = i * 360 / 20,
  r= i * 2,
  distance = r * 5,
)
rotate(angle, [1, 0, 0])
translate([0, distance, 0])
sphere(r = r);
```

Without `let()`
```scad
for (i = [10:50])
{
  angle = i * 360 / 20;
  r= i * 2;
  distance = r * 5;
  
  rotate(angle, [1, 0, 0])
  translate([0, distance, 0])
  sphere(r = r);
}
```

I personally prefer to use `let()` over variable assignment in scopes as I only use braces when absolutely needed, as it helps me visually keep sequential operations on objects separate from other objects.

bitbasher, 2025-07-18:
The use of let inside a vector comprehension might still be a necessary use case .. one cannot use union { } or things like that to create scope, and the iterator variables defined inside the ()s of the for statement ..

but my understanding of how to use let in this use case is far from complete so perhaps it is not an issue

jordanbrown0, 2025-07-19:
The `let()` statement is not the same as the `let()` kind-of-magic-operator in an expression, which in turn is not the same as the `let()` in the context of a list comprehension.

jordanbrown0, 2026-09-20:
Unstall.