openscad/openscad #5919: Normals of some faces flip into incorrect directions when working with linear_extrude and squares

Wixely, 2025-05-20:
**Describe the bug**
Some normals of some faces flip into incorrect directions when working with linear_extrude and squares.

**To Reproduce**
Use the following below code to reproduce the scenario.
Press F12 to show Thrown Together view
Make any change such as deleting one line or changing the order of the objects and observe the normals flip.
Render and observe

**Expected behavior**
All extruded squares should behave equally and have their normals correctly facing outward

**Code reproducing the issue**
```

{
linear_extrude(height = 50, center = false)
    { 
    translate([400, 80, 0]) square([40, 20]);
    translate([480, 140, 0]) square([20, 20]);
    translate([240, 160, 0]) square([120, 20]);
    translate([260, 180, 0]) square([40, 20]);
    translate([320, 180, 0]) square([140, 20]);
    translate([80, 200, 0]) square([140, 20]);
    translate([280, 200, 0]) square([20, 20]);
    translate([360, 200, 0]) square([20, 20]);
    translate([440, 200, 0]) square([20, 40]);
    translate([280, 220, 0]) square([60, 20]);
    translate([100, 240, 0]) square([20, 20]);
    translate([140, 240, 0]) square([80, 20]);
    translate([300, 240, 0]) square([20, 20]);
    translate([340, 240, 0]) square([20, 20]); // affected square
    translate([380, 240, 0]) square([80, 20]);
    translate([320, 260, 0]) square([80, 20]);
    translate([460, 260, 0]) square([40, 20]);
    } 
}
```

**Screenshots**

[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 11 [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]
 - System: AMD 5950X
 - OpenSCAD Version 2021.01 https://files.openscad.org/OpenSCAD-2021.01-x86-64.zip

jordanbrown0, 2025-05-20:
Reproduced on 2021.01.
Not reproduced on 2025.04.23.

2021.01 is quite old.  Please try a recent development snapshot.
https://openscad.org/downloads.html#snapshots

Wixely, 2025-05-20:
Thank you Jordan, I didn't see the 2025 version until now.

closed 2025-05-20 by Wixely