openscad/openscad #5920: View/Thrown Together does not put checkmark in menu

jordanbrown0, 2025-05-20:
**Describe the bug**
View/Thrown Together and View/Preview are a mode selection.  They used to (when there was also Surfaces and Wireframe) show a check mark next to the currently selected one.  (But Surfaces and Wireframe didn't.)  Today, they both show icons and they do not show any indication of which mode is selected.

**To Reproduce**
Steps to reproduce the behavior:
1.  Create a model that is different in Thrown Together, e.g. `difference() { cube(10); sphere(10); }`.  This model is only so that you can see that you are in fact changing modes.
2. Select View/Preview.  Observe that you get preview display, a cube with a sphere cut away.  Observe that there is no indication that View/Preview is selected.
3. Select View/Thrown Together.  Observe that you get TT display, a cube plus a green sphere.  Observe that there is no indication that View/Thrown Together is selected.

**Expected behavior**
Menu should have a checkbox next to the selected option.

**Screenshots**
Preview:

[image]

Thrown Together:

[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version 2025.05.16

kintel, 2025-05-21:
They have checkmarks on macOS/Qt6.
Which version of Qt is this? Could it be a Qt5 vs. Qt6 issue.

jordanbrown0, 2025-05-21:
Qt version: 5.15.11

jordanbrown0, 2025-05-21:
2021 doesn't put checkboxes next to Surfaces or Wireframe, so I suspect that you can't have both a checkbox and an icon.

kintel, 2025-05-22:
[image]

jordanbrown0, 2025-05-22:
Qt6 on MSYS2 on Windows has a subtle marking that appears equivalent to the check mark.  Note the boxes around the icons for Thrown Together, Show Axes, Show Scale Markers, and Perspective.

[image]

I don't immediately find any other Windows applications with this UI element.

It's getting to be hard to find MS programs (which presumably reflect Windows best practices) that use traditional menus, but those few that I found, that had modal menu items, used checkboxes *or* icons, not both.  The one example that I found that might plausibly use both is Paint's View/Thumbnail option, which has an icon and no indication that it's active.  In non-MS applications that use traditional menus (which are also getting hard to find), I found checkboxes without icons.

The icons add very little.  If dropping them restores the checkboxes, I'd call that a win.