openscad/openscad #5921: Drastically different render times, seemingly identical computers

thestumbler, 2025-05-22:
Running MacOS 15.5 on two different M1 ARM computers, a MBP and Mac mini. I am running the same OpenSCAD snapshot on both computers (2025-1-27), and configured identically as best I can tell. When I load the same project on both computers, the render times are quite different:

MacBook Pro ~ 12 sec
Mac mini ~ 58 sec

History... I have been primarily using the MBP for this design. I only recently tried to use it on the Mini. It's possible that back in Jan / Feb with I configured the MBP to run the snapshot version of OpenSCAD, I did something to tweak the system, something I can no longer remember.

Any ideas what things could cause such different render times?

-Chris

thestumbler, 2025-05-22:
well, it may just be the number of cores. The Mini has 4 performance cores and 4 efficiency cores, but the MBP has 8 performance cores and 2 efficiency cores. I wasn't thinking my laptop was more powerful than the desktop computer. I suspect this closes my question.
-Chris

closed 2025-05-22 by thestumbler

kintel, 2025-05-22:
The difference sounds too large to be simply about 2 more performance cores. Perhaps you have different settings for e.g. Manifold vs. CGAL backend. Also keep in mind that there may be caching involved.
Previews generally don't give great rendering time numbers, best to measure geometry performance is to F6-render after clearing caches.