openscad/openscad #5923: OpenScad unresponsive after start on Apple Silicon M4

renestraub, 2025-05-23:
**Describe the bug**
OpenSCAD becomes unresponsive immediately after start on Apple Silicon M4

**To Reproduce**
- Download and install OpenSCAD 2025.05.16 on Mac Mini M4 Pro
- Start OpenSCAD
- Application is unresponsive immediately. Cursor changes to Spinning Wheel of Death

**Expected behavior**
Application is working, accepting user interaction, not blocking

**Code reproducing the issue**
-

**Screenshots**
-

**Environment and Version info (please complete the following information):**
 - OS: macOS 15.5 (24F74)
 - System: Mac Mini M4
 - OpenSCAD Version: 2025.05.16, Download from https://openscad.org/downloads.html

**Library & Graphics card information**
/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2025.05.16 (git c76900f9a)
System information: macOS 15.5.0 arm64 Mac16,11 12 CPUs 48.00 GB RAM 
User Agent: OpenSCAD/2025.05.16 (git c76900f9a) (macOS 15.5.0 arm64 Mac16,11)
Compiler: Clang "15.0.0 (clang-1500.3.9.4)"
MinGW build: No
Debug build: No
Boost version: 1_88
Eigen version: 3.4.0
CGAL version, kernels: 6.0.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>
OpenCSG version: OpenCSG 1.7.0
Clipper2 version: 1.5.2
Manifold version: 3.0.1
Qt version: 6.5.5
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.83.0
lodepng version: 20230410
libzip version: 1.9.2
fontconfig version: 2.14.1
freetype version: 2.12.1
harfbuzz version: 6.0.0
cairo version: 1.18.0
lib3mf version: 2.4.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /Applications/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/straub/Documents
User Documents Path: /Users/straub/Documents
Resource Path: /Applications/OpenSCAD.app/Contents/Resources
User Library Path: /Users/straub/Documents/OpenSCAD/libraries
User Examples Path: /Users/straub/Documents/OpenSCAD/examples
User Config Path: /Users/straub/Library/Application Support/OpenSCAD
Backup Path: /Users/straub/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /Users/straub/Documents/OpenSCAD/libraries
  /Applications/OpenSCAD.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /System/Library/Assets/com_apple_MobileAsset_Font3
  /System/Library/Assets/com_apple_MobileAsset_Font4
  /Users/straub/.local/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /Users/straub/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /Users/straub/Library/Fonts
  /usr/X11/lib/X11/fonts/100dpi
  /usr/X11/lib/X11/fonts/75dpi
  /usr/X11/lib/X11/fonts/OTF
  /usr/X11/lib/X11/fonts/Speedo
  /usr/X11/lib/X11/fonts/TTF
  /usr/X11/lib/X11/fonts/Type1
  /usr/X11/lib/X11/fonts/cyrillic
  /usr/X11/lib/X11/fonts/encodings
  /usr/X11/lib/X11/fonts/misc
  /usr/X11/lib/X11/fonts/util
  /System/Library/Fonts/Supplemental
  /usr/X11/lib/X11/fonts/encodings/large

GL context creator: CGL (new)
PNG generator: Core Foundation

GLAD version: 2.0.4
OpenGL Version: 2.1 Metal - 89.4
GL Renderer: Apple M4 Pro
GL Vendor: Apple
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

**Additional context**
-

kintel, 2025-05-29:
I just verified that that specific works on both macOS 14.7 and 15.5, so there seems to be something about your system that causes OpenSCAD to freeze.

Some questions which may help limit the scope of the search:
* Have you used OpenSCAD before on this computer or is this your first install?
* Does the latest official release (2021.01) work
* Could you spot check a couple of other snapshots from https://files.openscad.org/snapshots/ ?
* Could you try launching OpenSCAD from the terminal to see if any messages show up? In a Terminal: `/Volumes/OpenSCAD/OpenSCAD.app/Contents/MacOS/OpenSCAD`

renestraub, 2025-05-29:
Hello Kintel,

regarding your questions:

* Have you used OpenSCAD before on this computer or is this your first install?
I have used OpenSCAD 2021 on this computer (and for many years on a x64 Mac). Just the Apple Silicon release doesn't work.

* Does the latest official release (2021.01) work
Yes, all ok

* Could you spot check a couple of other snapshots from https://files.openscad.org/snapshots/ ?
I tried 3 snapshots in the last 4 weeks (when I moved to a new M4 Mac Mini). They all have exactly the same behavior.

* Could you try launching OpenSCAD from the terminal to see if any messages show up? In a Terminal: /Volumes/OpenSCAD/OpenSCAD.app/Contents/MacOS/OpenSCAD

Yes, same behavior. Here is the output of a session with full debug

/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD --debug all
Debug on. --debug=all
Camera: Camera()
openscad: Application location detected as /Applications/OpenSCAD.app/Contents/MacOS
Camera: Camera()
Camera: Camera()
qt.qpa.fonts: Populating font family aliases took 104 ms. Replace uses of missing font family "DejaVu Sans" with one that exists to avoid this cost. 
Preferences: SettingsReader R: editor/lineWrapVisualizationBegin = 'Text'
Preferences: SettingsReader R: editor/backspaceUnindents = 'true'
Preferences: SettingsReader R: editor/indentStyle = 'Spaces'
Preferences: SettingsReader R: printing/octoPrintFileFormat = 'asciistl'
Preferences: SettingsReader R: printing/localAppFileFormat = 'asciistl'
Preferences: SettingsReader R: advanced/toolbarExport3D = 'asciistl'
Preferences: SettingsReader R: advanced/toolbarExport2D = 'dxf'
QSettingsCached: QSettings::setValue(): view/hideCustomizer = 'true'
QSettingsCached: QSettings::setValue(): view/hideEditorToolbar = 'false'
QSettingsCached: QSettings::setValue(): view/hide3DViewToolbar = 'false'
QGLView: GLAD: Loaded OpenGL 2.1
QSettingsCached: QSettings::setValue(): view/hideEditor = 'false'
QSettingsCached: QSettings::setValue(): view/hideConsole = 'false'
QSettingsCached: QSettings::setValue(): view/hideAnimate = 'false'
QSettingsCached: QSettings::setValue(): view/hideViewportControl = 'false'
QSettingsCached: QSettings::setValue(): view/hideErrorLog = 'false'
QSettingsCached: QSettings::setValue(): view/hideFontList = 'false'
Fontconfig warning: ignoring UTF-8: not a valid region tag
QSettingsCached: QSettings::setValue(): recentFileList = ''
FALLBACK (log once): Fallback to SW vertex for line stipple
FALLBACK (log once): Fallback to SW vertex processing, m_disable_code: 2000
FALLBACK (log once): Fallback to SW vertex processing in drawCore, m_disable_code: 2000
^C^C

A properly working 2021 (x64) installation shows this output

/Applications/OpenSCAD-2021.01.app/Contents/MacOS/OpenSCAD --debug all
Debug on. --debug=all
Camera: Camera()
Camera: Camera()
Camera: Camera()
Preferences: SettingsReader R: 3dview/showWarningsIn3dView = '' => 'true'
Preferences: SettingsReader R: 3dview/mouseCentricZoom = '' => 'true'
Preferences: SettingsReader R: editor/indentationWidth = '' => '4'
Preferences: SettingsReader R: editor/tabWidth = '' => '4'
Preferences: SettingsReader R: editor/lineWrap = '' => 'Word'
Preferences: SettingsReader R: editor/lineWrapIndentationStyle = '' => 'Fixed'
Preferences: SettingsReader R: editor/lineWrapIndentation = '' => '4'
Preferences: SettingsReader R: editor/lineWrapVisualizationBegin = 'Text' => 'Text'
Preferences: SettingsReader R: editor/lineWrapVisualizationEnd = '' => 'Border'
Preferences: SettingsReader R: editor/showWhitespaces = '' => 'Never'
Preferences: SettingsReader R: editor/showWhitespacesSize = '' => '2'
Preferences: SettingsReader R: editor/autoIndent = '' => 'true'
Preferences: SettingsReader R: editor/backspaceUnindents = 'true' => 'true'
Preferences: SettingsReader R: editor/indentStyle = 'Spaces' => 'Spaces'
Preferences: SettingsReader R: editor/tabKeyFunction = '' => 'Indent'
[... 69 lines cut ...]
Preferences: SettingsReader R: input/axisDeadzone4 = '' => '0.1'
Preferences: SettingsReader R: input/axisDeadzone5 = '' => '0.1'
Preferences: SettingsReader R: input/axisDeadzone6 = '' => '0.1'
Preferences: SettingsReader R: input/axisDeadzone7 = '' => '0.1'
Preferences: SettingsReader R: input/axisDeadzone8 = '' => '0.1'
Preferences: SettingsReader R: input/axisDeadzone9 = '' => '0.1'
Preferences: SettingsReader R: input/joystickNr = '' => '0'
QSettingsCached: QSettings::setValue(): recentFileList = ''
QSettingsCached: QSettings::setValue(): view/hideEditor = 'false'
QSettingsCached: QSettings::setValue(): view/hideEditor = 'false'
QSettingsCached: QSettings::setValue(): view/hideCustomizer = 'false'
QSettingsCached: QSettings::setValue(): view/hideCustomizer = 'true'
QSettingsCached: QSettings::setValue(): view/hideConsole = 'false'
QSettingsCached: QSettings::setValue(): view/hideConsole = 'false'
QSettingsCached: QSettings::setValue(): view/hideErrorLog = 'false'
QSettingsCached: QSettings::setValue(): view/hideErrorLog = 'false'
QSettingsCached: QSettings::setValue(): view/hideEditor = 'false'
QSettingsCached: QSettings::setValue(): view/hideConsole = 'false'
QSettingsCached: QSettings::setValue(): view/hideErrorLog = 'false'
QSettingsCached: QSettings::setValue(): view/hideCustomizer = 'true'
QSettingsCached: QSettings::setValue(): view/hideEditorToolbar = 'false'
QSettingsCached: QSettings::setValue(): view/hide3DViewToolbar = 'false'
FALLBACK (log once): Fallback to SW vertex for line stipple
FALLBACK (log once): Fallback to SW vertex processing, m_disable_code: 2000
FALLBACK (log once): Fallback to SW vertex processing in drawCore, m_disable_code: 2000

renestraub, 2025-06-03:
It's a GUI issue. The command line works as expected. For two models I could create a PNG with success.
I don't have a XCode dev environment setup yet, but can give it a try. Do you have some build/test instructions for me? 

```bash
% /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD ring1.scad -o ring.png
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Geometries in cache: 16
Geometry cache size in bytes: 37200
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.091

% /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD case1.scad -o case1.png
Compiling design (CSG Products normalization)...
Normalized CSG tree has 11 elements
Geometries in cache: 33
Geometry cache size in bytes: 864096
CGAL Polyhedrons in cache: 2
CGAL cache size in bytes: 20774848
Total rendering time: 0:00:04.968
```

kintel, 2025-06-03:
Btw., how long did you wait before giving up? There is some initialization that needs to run (like font lookup), which could take some time, but there is usually a small progress window for this.

renestraub, 2025-06-04:
I could almost get it to compile with these steps

```
 1079  git clone https://github.com/openscad/openscad.git
 1080  cd openscad
 1081  git submodule update --init --recursive
 1082   source scripts/setenv-macos.sh
 1083   ./scripts/macosx-build-dependencies.sh
 1084  cmake -B build -DEXPERIMENTAL=1
 1085  cmake --build build
```

Unfortunately it fails at compiling OpenSCAD src/io/export_pdf.cc

```
straub@Mac-mini openscad % cmake --build build
[  1%] Built target Clipper2
[  7%] Built target manifold
[  9%] Built target OpenSCAD_autogen_timestamp_deps
[  9%] Built target OpenSCAD_autogen
[ 13%] Built target mimalloc-static
[ 13%] Building CXX object CMakeFiles/OpenSCAD.dir/src/io/export_pdf.cc.o
/Users/straub/openscad/src/io/export_pdf.cc:207:52: error: unknown type name 'cairo_pdf_metadata_t'
  207 | void add_meta_data(cairo_surface_t *surface, const cairo_pdf_metadata_t metadata, const std::string& value, const std::string& value2 = "") {
      |                                                    ^
1 error generated.
```

Any ideas?

kintel, 2025-06-04:
You'll have to `rm -r build` to get rid of your partial build

renestraub, 2025-06-05:
Does this backtrace of a crash in lldb help?

```
admin@Mac-mini openscad % lldb build/OpenSCAD.app/Contents/MacOS/OpenSCAD
(lldb) target create "build/OpenSCAD.app/Contents/MacOS/OpenSCAD"
Current executable set to '/Users/admin/prj/openscad/build/OpenSCAD.app/Contents/MacOS/OpenSCAD' (arm64).
(lldb) r
Process 73759 launched: '/Users/admin/prj/openscad/build/OpenSCAD.app/Contents/MacOS/OpenSCAD' (arm64)
Process 73759 stopped
* thread #1, queue = 'WMClientWindowManager', stop reason = EXC_BREAKPOINT (code=1, subcode=0x18870077c)
    frame #0: 0x000000018870077c libxpc.dylib`_xpc_api_misuse + 92
libxpc.dylib`_xpc_api_misuse:
->  0x18870077c <+92>: brk    #0x1

libxpc.dylib`string_builder_new:
    0x188700780 <+0>:  pacibsp 
    0x188700784 <+4>:  stp    x20, x19, [sp, #-0x20]!
    0x188700788 <+8>:  stp    x29, x30, [sp, #0x10]
Target 0: (OpenSCAD) stopped.
(lldb) bt
* thread #1, queue = 'WMClientWindowManager', stop reason = EXC_BREAKPOINT (code=1, subcode=0x18870077c)
  * frame #0: 0x000000018870077c libxpc.dylib`_xpc_api_misuse + 92
    frame #1: 0x00000001886de33c libxpc.dylib`xpc_connection_set_target_uid + 216
    frame #2: 0x0000000269a8ec58 WindowManagement`-[WMClientWindowManager _createXPCConnection] + 1004
    frame #3: 0x0000000269a8f834 WindowManagement`-[WMClientWindowManager _xpcConnection] + 116
    frame #4: 0x0000000269a8ce48 WindowManagement`__31-[WMClientWindowManager stages]_block_invoke + 36
    frame #5: 0x000000018883885c libdispatch.dylib`_dispatch_client_callout + 16
    frame #6: 0x000000018882e7a8 libdispatch.dylib`_dispatch_lane_barrier_sync_invoke_and_complete + 56
    frame #7: 0x0000000269a8cd90 WindowManagement`-[WMClientWindowManager stages] + 300
    frame #8: 0x000000018d706734 AppKit`__54-[NSWMWindowCoordinator initializeStageFramesIfNeeded]_block_invoke + 32
    frame #9: 0x000000018883885c libdispatch.dylib`_dispatch_client_callout + 16
    frame #10: 0x0000000188821a28 libdispatch.dylib`_dispatch_once_callout + 32
    frame #11: 0x000000018d7066e0 AppKit`-[NSWMWindowCoordinator initializeStageFramesIfNeeded] + 304
    frame #12: 0x000000018c9f9aa8 AppKit`-[NSWindow _commonInitFrame:styleMask:backing:defer:] + 692
    frame #13: 0x000000018c9f9494 AppKit`-[NSWindow _initContent:styleMask:backing:defer:contentView:] + 848
    frame #14: 0x000000018cb4b7cc AppKit`-[NSPanel _initContent:styleMask:backing:defer:contentView:] + 48
    frame #15: 0x000000018c9f9138 AppKit`-[NSWindow initWithContentRect:styleMask:backing:defer:] + 48
    frame #16: 0x000000018cb4b780 AppKit`-[NSPanel initWithContentRect:styleMask:backing:defer:] + 48
    frame #17: 0x000000018cc4b4d0 AppKit`-[NSWindow initWithContentRect:styleMask:backing:defer:screen:] + 24
    frame #18: 0x00000001086a5980 libqcocoa.dylib`-[QNSPanel initWithContentRect:styleMask:backing:defer:screen:platformWindow:] + 232
    frame #19: 0x0000000108680e00 libqcocoa.dylib`QCocoaWindow::createNSWindow(bool) + 1100
    frame #20: 0x0000000108679830 libqcocoa.dylib`QCocoaWindow::recreateWindowIfNeeded() + 1248
    frame #21: 0x0000000108679248 libqcocoa.dylib`QCocoaWindow::initialize() + 820
    frame #22: 0x0000000107ab666c QtGui`QWindowPrivate::create(bool) + 428
    frame #23: 0x0000000106c9efd4 QtWidgets`QWidgetPrivate::create() + 1020
    frame #24: 0x0000000106c9cba4 QtWidgets`QWidget::create(unsigned long long, bool, bool) + 304
    frame #25: 0x0000000106cafc1c QtWidgets`QWidgetPrivate::setVisible(bool) + 784
    frame #26: 0x0000000106e4a244 QtWidgets`QDialogPrivate::setVisible(bool) + 240
    frame #27: 0x0000000106e498e0 QtWidgets`QDialog::exec() + 160
    frame #28: 0x0000000100cf6578 OpenSCAD`gui(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>&, std::__1::__fs::filesystem::path const&, int, char**, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) + 1332
    frame #29: 0x000000010060a004 OpenSCAD`main + 14480
    frame #30: 0x0000000188636b98 dyld`start + 6076
```

kintel, 2025-06-05:
To get a better stack trace, you could try a real debug build:
```
cmake -B build -DEXPERIMENTAL=1 -DCMAKE_BUILD_TYPE=Debug
```

kintel, 2025-06-05:
Anyway, this looks a bit suspicious: `frame #0: 0x000000018870077c libxpc.dylib_xpc_api_misuse + 92`
It kind of looks like macOS is detecting some bad XPC connection API usage. Do you happen to have any special system software running, e.g. custom network firewalls, custom antivirus or other system security tools?

It also looks like the hang might be occuring during the Welcome Screen startup. We can try to disable that for debugging, but a better stack trace would help identify it.

renestraub, 2025-06-28:
@johnbiggs : Thanks a lot for the update. 2025.06.23 also works on my Mini M4. Tested with several constructions.
Now it would be nice to know what did the trick.

kintel, 2025-06-28:
Good that it works now! It's curious why this was an issue earlier. We did upgrade a number of 3rd party package on June 3-4. Apart from that we haven't changed much lately..