openscad/openscad #5931: Measuring tool doesn't appear to function correctly on non-primitive shapes

anfractuosity, 2025-05-29:
**Describe the bug**

I can use the measuring tool on a cube (`cube(30, 30, 30)`) but not non-primitive shapes.

**To Reproduce**

Try code below and try to measure hollow area, which appears not possible currently.

**Expected behavior**

Able to measure inside of hollow area of cube.

**Code reproducing the issue**

```
		difference()
		{
			bat_len = 85;
			bat_wid = 21;
			bat_hig = 21 + 3;
			wall = 2;
			cube([bat_len + wall, bat_hig + wall, bat_wid + wall]);
			translate([wall, wall, wall])
			cube([bat_len - wall, bat_hig, bat_wid - wall]);
		}
```

**Environment and Version info (please complete the following information):**
 - OS: Arch Linux
 - System: x86-64
 - OpenSCAD Version: 2025.05.16.ai25460

UBaer21, 2025-05-30:
it seems to work but not all edges/points can be selected easy.

[image]

I think it would be helpful if measurements could be restricted to an axis, e.g. finding the bottom/wall thickness in Z on a point.

anfractuosity, 2025-05-30:
On Arch, I couldn't seem to hover over any edge with the hollow cube, whereas I could for a simple cube strangely

gsohler, 2025-05-30:
Measurement performance(speed)  could be greatly improved when "preparing"
the Data after each view change into a quad-tree for fast lookup
But this would only work for vertices (not for edges or faces)

Unless I get a clean vision how we keep current functionality, i will not
start with a PR