openscad/openscad #5944: Can't build macOS release locally.

lgarron, 2025-05-31:
**Describe the bug**
I can't build the macOS release so I can use it locally.

(I can't use the debug build because it segfaults without any useful message when I tried to open a non-trivial project in it.)

**To Reproduce**
I've tried [running the following](https://github.com/openscad/openscad/blob/167476e5753304efbbd3d1fbe82cb7f52c828259/.circleci/config.yml#L210-L213):

```bash
# bash
export NUMCPU=$(($(sysctl -n hw.ncpu) * 3 / 2))
export VERSIONDATE=$(date "+%Y.%m.%d")
time ./scripts/release-common.sh -d $VERSIONDATE snapshot
```

This fails with:

```
CMake Error at CMakeLists.txt:975 (find_package):
  By not providing "FindClipper2.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Clipper2",
  but CMake did not find one.

  Could not find a package configuration file provided by "Clipper2" with any
  of the following names:

    Clipper2Config.cmake
    clipper2-config.cmake

  Add the installation prefix of "Clipper2" to CMAKE_PREFIX_PATH or set
  "Clipper2_DIR" to a directory containing one of the above files.  If
  "Clipper2" provides a separate development package or SDK, be sure it has
  been installed.
```

I tried running `scripts/macosx-build-dependencies.sh` (even though dependencies are already installed from Homebrew). This results in:

```
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.
```

My installed version of `CMake` (i.e. the output of `cmake --version`) is 4.0.2, so I don't know how to make sense of this error.

**Expected behavior**
I can compile the code?

**Code reproducing the issue**
See above.

**Screenshots**
N/A (see commandline output above)

**Environment and Version info (please complete the following information):**
 - OS: macOS 15.5
 - System: 16" M1 MacBook PRo
 - OpenSCAD Version: commit fd99288ad77142cc120b5fe3f00207776b7f5d25

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

kintel, 2025-06-01:
Basically, there are two ways of building on macOS: 1. Using homebrew dependencies and 2. Using self-build dependencies.

1. is good for local execution and testing, but the resulting binaries are not distributable. This is how we automatically run OpenSCAD tests on GitHub
2. is good for building actual snapshot binaries, but comes at the cost of building your own dependencies. Important note: This does not yet work with CMake 4, see #5940.

Starting with 1. is recommended:
* ` ./scripts/macosx-build-homebrew.sh`
* `cmake -B build -DEXPERIMENTAL=1`
* `cmake --build build`

lgarron, 2025-06-04:
> 1. is good for local execution and testing, but the resulting binaries are not distributable. This is how we automatically run OpenSCAD tests on GitHub
> 2. 
> Starting with 1. is recommended:
> 
> * ` ./scripts/macosx-build-homebrew.sh`
> * `cmake -B build -DEXPERIMENTAL=1`
> * `cmake --build build`

Yeah, this works for me for a super simple test file. But depending on when I've tried to rebuild it I've had it run extremely slow, segfault, hang, or just fail to start.

> 2\. is good for building actual snapshot binaries, but comes at the cost of building your own dependencies. Important note: This does not yet work with CMake 4, see [Make CMake 4 work everywhere #5940](https://github.com/openscad/openscad/issues/5940).

Okay, I'm able to avoid the CMake 4 error by running:

```
env CMAKE_POLICY_VERSION_MINIMUM=3.5 ./scripts/macosx-build-dependencies.sh
```

But then it crashes with:

```
Building boost 1.88.0 ...
boost_1_88_0/libs/math/doc/graphs/horner_vs_estrin_accuracy.png: truncated bzip2 input: Unknown error: -1
tar: Error exit delayed from previous errors.
```

Is there a recommended way to build using this approach? Maybe in a fresh VM? Or in GitHub Actions in a fork?

lgarron, 2025-06-04:
> To deal with performance issues, try to build in release mode: `cmake -DCMAKE_BUILD_TYPE=Release ...`

Thanks! This seems to be significantly more stable and performant than a debug build for me. Took me a while to figure out where to put that flag, but this worked:

```shell
./scripts/macosx-build-homebrew.sh
cmake -B build -DCMAKE_BUILD_TYPE=Release -DEXPERIMENTAL=1
cmake --build build
```

(I tried building in GitHub Actions as well, but that one has been stuck waiting on a runner for 1½ hours and counting.)

kintel, 2025-06-04:
Note: GitHub actions won't necessarily produce a binary you can use, as it may depend on the exact OS version and CPU that it's built on. That's what I mean by "resulting binaries are not distributable". Also, free action quota may be low :/

lgarron, 2025-06-04:
> Note: GitHub actions won't necessarily produce a binary you can use, as it may depend on the exact OS version and CPU that it's built on. That's what I mean by "resulting binaries are not distributable". 

Ah, good to know. I guess we'll see if I find out.

> Also, free action quota may be low :/

I do have a billing method hooked up, I guess I need to read up on how to make sure it actually gets used.

In any case, thanks so much! I now have a local build with https://github.com/openscad/openscad/pull/5942 that I can use to make my 3D printing projects a lot easier.

I guess I'd be alright with closing this since I don't need any more info at the moment, but I think it might be useful to put this information in an easier-to-find place for anyone who also wants a useful local build. The README currently:

1. requires reading but also skipping through multiple sections to piece together exactly what commands to run[^1],
2. places the Homebrew install as the second suggestion even though you're told me it's recommended, and
3. does not give me enough information to figure out how to perform a release build (or when it would be appropriate to build one).

[^1] By contrast, it would really be nice to have a single code block marked "if you have Homebrew, copy and run this to do everything you need for a full build". (Especially since GitHub will add a convenient button to copy the code block.)

kintel, 2025-06-04:
It's significant work to maintain waterproof instructions per platform, especially on macOS which seems to be a moving target, so we haven't found the time to prioritize it. ..but tweaking the README would make sense.