openscad/openscad #5958: Naming conflict with pythonscad

nomike, 2025-06-06:
Both OpenSCAD and pythonscad produce executables named `openscad` and `openscad-python`.

This makes it very hard (guix) or even impossible (Ubuntu, Debian, Fedora,....probably almost all other Linux distributions) to install both at the same time.

I already filed an [incident with the pythonscad project](https://github.com/pythonscad/pythonscad/issues/19) to have the executable renamed to `pythonscad` there.

I do however question why there is an `openscad-python` exectuable with OpenSCAD and whether it makes any difference which one I call.

Because if there is no difference, OpenSCAD could probably ditch the `openscad-python` executable.

I guess this was just inadvertently taken over when merging parts of pythonscad into the OpenSCAD tree.

As both projects are that tightly connected, it might be better in the long run to fully combine those and not develop them separately, but that's a discussion for another time.

kintel, 2025-06-06:
Considering pythonscad is a fork of OpenSCAD (and possibly limited in lifetime until features can be merged), installing both on the same system using package managers may not be an ideal scenario..

nomike, 2025-06-06:
Ah. Okay.

Well, it is usually anyway the fork which changes something relation to the name when forking (e.g. MySQL → MariaDB, OpenOffice → LibreOffice, memtest86

closed 2025-06-06 by nomike

gsohler, 2025-06-06:
Apparantly  we are stuck again here:

I dont know, which features of pythonscad are interesting to python-enabled openscad. For many  features I am sure, you would not want them

On the other hand: before putting any further effort yourself, your biggest goal is to make venv working, but I am not the correct person with the required knowledge.

My strategy to improve  openscad's python is clearly to enable the windows build, because it can extend readch best. but as our cmakeList.txt greatly differ and small pr chunks can be better reviewed, i prepared a targetting pr some time ago.

https://github.com/openscad/openscad/pull/5697

Happy Weekend ;)

t-paul, 2025-06-06:
> I dont know, which features of pythonscad are interesting to python-enabled openscad. For many features I am sure, you would not want them

Why? Which ones? It's impossible to tell if you don't ask.

In general, I think the main goal should be to *share* features so all front-end languages can benefit from the features. Otherwise if features are well separated, I don't see a huge problem, but again, not possible to tell unless it's been talked about.

> On the other hand: before putting any further effort yourself, your biggest goal is to make venv working, but I am not the correct person with the required knowledge.

I knew nothing about it when I started. I tried to help, but that seems to go nowhere. I still think some solution is required but I guess we need to pause that and see how things go.

> My strategy to improve openscad's python is clearly to enable the windows build, because it can extend readch best. but as our cmakeList.txt greatly differ and small pr chunks can be better reviewed, i prepared a targetting pr some time ago.

There is currently GSoC project looking to get a native (Visual Studio) build going. This could be another option for a possible Windows release build but we'll have to see how that project goes.
With the current build process, I don't see how a CMake change will help with enabling Python. If it turns out we need to self-compile Python (like for the AppImage) that needs to be MXE.

kintel, 2025-06-06:
@nomike There is a setting in CMakeLists already called `SUFFIX` - it's used to be able to install both release and nightly on the same system: `openscad` and `openscad-nightly`. I believe it's Linux-only and it might not be extended to Python support yet.

nomike, 2025-06-06:
I see a couple of topics in need of being addressed:

1. It's not good that there is this a direct hard dependency between pythonscad and the specific python version. It's mainly an issue for Windows and mac users though, as Linux has proper package management which ensures such problems don't exist. (BTW: I don't have a lot of experience developing natively in Windows, but I have Windows 11 on my work laptop, so I can help out in resolving specific issues there, I don't see myself in the role of driving a native visual studio build or something). (*)

2. IMHO it's important to get a new official release of OpenSCAD out in the near future. Most people still use the 2021.01 version as that's the one packaged with their distribution. I'm not sure why there is such a huge gap between now and the last release, but I can only speculate that OpenSCAD has become a victim of feature creep. Regardless, as far as I know there are efforts on making the code stable to be able to create a new release soon. So IMHO this should have some level of priority.

3. As far as I can see, we need more unit tests targeting the python implementation. That's something I surely can help out with. Shouldn't be that hard. After all, especially if we're going for a stable public release, it's important to ensure stability of the product (BTW: I love how many unit-tests there are for OpenSCAD. Someone files a bug? Write a test for it. That's great! A lot of projects don't do that properly IMHO).

4. Regarding the "venv" topic: Can someone point me to a discussion thread somewhere where I can read up on that topic. Maybe I have some ideas when I know exactly what's the deal.

5. On the topic of the feature gap between OpenSCAD and pythonscad: In pythonscad you can do stuff like this: `cube([10, 20, 10]).up(20)` which is the equivalent to `translate([0, 0, 20]) cube([10, 20, 10]);`. Much more concise and easy to write IMHO.

While there is python support in OpenSCAD now, this feature is still not there. It's a convenience function only, I know, one could still write `cube([10, 20, 10]).transate([0, 0, 20])` in OpenSCAD-python, but there might be other nice (future) ideas which could make code more readable and easier to write, where OpenSCAD can benefit from.

Other things like `osimport` which exists in pythonscad but not yet in OpenSCAD are more crucial. How can use python mode if I can't import external STL, SVG, etc. files?

Or for example I was recently thinking about something like this:

```python

def some_complex_object():
    # construct a complex geometry somehow
    return complex_object

sco = some_complex_object()
base = cube([sco.dimmensions.x, sco.dimmensions.y, 10)

render = base | sco.translate(base.dimmensions.z)
```

While I love the possibilities of OpenSCAD, the DSL is sometimes a bit hard to deal with due to it's limitations (e.g. the [multidimensional array parser](https://github.com/nomike/funcutils/blob/67ab897b1a24342babd24a197fd480f6e0fb59e6/list.scad#L77) I once wrote. It works, but it boy was this painful to write. And the code isn't what I would call maintainable.

So having awesome python support in OpenSCAD would be a huge benefit and open it for a much wider audience.

6. Judging from the outside it looks like there is some friction between the two projects in terms of merging features in between them. We should maybe think about ways to reduce it, to spend more energy and time on new features or stabilizing existing ones. 

All in all I think OpenSCAD is a great piece of software which I have spent countless hours with already and I appreciate all the development which has gone into it. And I love that there is emerging python support as well and fresh ideas coming up.

7. It looks like I'm slowing becoming the package manager for both projects on GNU guix and I'm more than happy to continue doing that.

(*) BTW: What's the idea behind this GSoC project and native visual studio builds? Is there an issue with using cmake on Windows? Blender for example uses cmake as well and that should be quite similar in terms of needed libraries as OpenSCAD. I'm not trying to question this project, I merely want to learn and understand what the pain points are in this regard.

---
Sorry for the long post, here's a potato: 🥔

kintel, 2025-06-06:
Let's take everything _except_ discussing executable naming to IRC or the mailing list, or another venue. @nomike I still think `-DSUFFIX=mysuffix` should give you some mileage.

nomike, 2025-06-07:
I managed to introduce a variable `EXECUTABLE_NAME` which defaults to "OpenSCAD".

This is used to specify the name of executables generated, and I have just now compiled "LolCAD" and the test suite is running.

I can clean this up tomorrow and file a pull requests if you think it looks good.

You can have a look here: https://github.com/nomike/openscad/tree/cmake-parametrize-appname

nomike, 2025-06-07:
> 100% tests passed, 0 tests failed out of 2549

nomike, 2025-06-08:
> There are also multiple places in code that need to be unified and made dependent on the cmake setting.

Sure, setting the binary name is only a start.
And I did not take care of Windows and mac for now.