openscad/openscad #5964: 3mf export terminates program

UBaer21, 2025-06-08:
**Describe the bug**
program crash without error or warning

**To Reproduce**
render and export as 3mf (color)

**Expected behavior**
no crash

**Code reproducing the issue**
`color(alpha=1)cube();`

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - System:  Intel PC 64-bit
 - OpenSCAD Version 2025.06.05

kintel, 2025-06-08:
We have a test case for this. Does the test case also crash? https://github.com/openscad/openscad/blob/master/tests/data/scad/3D/features/color-tests.scad

kintel, 2025-06-08:
Also, considering alpha defaults to 1, does this also crash `color() cube();` ?

UBaer21, 2025-06-08:
@kintel  the test case crashes my local install (from website), also` color()cube();`  the culprit seems to be the color selection for export.

[image]
if selecting "no colors" ↦ no crash.

t-paul, 2025-06-08:
Hmm, as it's not crashing on Linux for me, this sounds like some variable initialization issue or similar. Maybe I can have a look once the Windows development stuff is progressing well...

kintel, 2025-06-08:
Getting Windows on lib3mf v2 may have higher precedence, but may be a bit annoying in the context of MXE

kintel, 2025-06-08:
Meh, the Windows tests run on lib3mf v2 as well