openscad/openscad #5965: issues in Windows 11 Build duing setup

bitbasher, 2025-06-08:
**Environment**
Compiling with MSYS2 on Windows 11

**Issue**
i am following the NEW directions for the Windows Build at: 
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Building_on_Microsoft_Windows

( full log of my build is attached
[openscad-build-prob-2.txt](https://github.com/user-attachments/files/20645138/openscad-build-prob-2.txt)
)

1. install msys2-x86_64-20250221
2. start it up in admin mode
3. download and run the dependencies script with no issues
4. clone the repository
5. cd into the repo and recurse to get manifold etc
6. mkdir build ; cd build
7. cmake .. -G"MSYS Makefiles" -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Release -DEXPERIMENTAL=ON -DSNAPSHOT=ON

and paused here because i saw these things in the log

1. USE_GLAD/USE_GLEW not specified: Defaulting to glew
2. Visual Leak Detector (VLD) is not found.
3. CMake Warning (dev) at C:/bin/Msys2/ucrt64/lib/cmake/CGAL/CGAL_SetupBoost.cmake:20 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

and using that command : 
 cmake --help-policy CMP0167
CMP0167
-------

.. versionadded:: 3.30

The ``FindBoost`` module is removed.

CMake 3.29 and below provide a ``FindBoost`` module, but it needs constant
updates to keep up with upstream Boost releases.  Upstream Boost 1.70
and above provide a ``BoostConfig.cmake`` package configuration file.
``find_package(Boost CONFIG)`` finds the upstream package directly,
without the find module.

CMake 3.30 and above prefer to not provide the ``FindBoost`` module
so that ``find_package(Boost)`` calls, without the ``CONFIG`` or
``NO_MODULE`` options, find the upstream ``BoostConfig.cmake`` directly.
This policy provides compatibility for projects that have not been ported
to use the upstream Boost package.

The ``OLD`` behavior of this policy is for ``find_package(Boost)`` to
load CMake's ``FindBoost`` module.  The ``NEW`` behavior is for
``find_package(Boost)`` to search for the upstream ``BoostConfig.cmake``.

This policy was introduced in CMake version 3.30.
It may be set by ``cmake_policy()`` or ``cmake_minimum_required()``.
If it is not set, CMake warns, and uses ``OLD`` behavior.

kintel, 2025-06-08:
tl;dr it's all harmless.

> 1. USE_GLAD/USE_GLEW not specified: Defaulting to glew

This is just an informational message. Nothing to be worried about.

> 2. Visual Leak Detector (VLD) is not found.

This seems to be harmless. It looks like some Visual C++-related service. Not sure why it shows up..

> 2. CMake Warning (dev) at C:/bin/Msys2/ucrt64/lib/cmake/CGAL/CGAL_SetupBoost.cmake:20 (find_package):>    Policy CMP0167 is not set: The FindBoost module is removed. 

This is a weakness in a 3rd party library (CGAL). It's harmless and not Windows-related: https://github.com/CGAL/cgal/issues/8914

bitbasher, 2025-06-08:
i have no opinion on the first two .. just thought someone should take a look

the one about FindBost tho .. that looks like someone should update the code to not use the deprecated feature.

anyway, thanks for taking a look

kintel, 2025-06-08:
Yeah, the FindBoost issue is fixed upstream in CGAL, we're waiting for their next release before we can utilize it in OpenSCAD.