openscad/openscad #5966: Color rendering: color() without arguments exports as black to 3mf

kintel, 2025-06-08:
`color() cube()` exports as a black cube, while it renders as a default (yellow) colored cube.

Only tested with lib3mf v2 on macOS.

Needs:
* [ ] Add test case
* [ ] Test on other platforms
* [ ] Test on lib3mf v1
* [ ] Add alpha channel test
* [ ] Test other color-capable export formats

kintel, 2025-06-08:
Note that regular default exports works fine. It's the "naked" `color()` that triggers this behavior.
..and the 3MF viewer in question is OpenSCAD itself.

kintel, 2025-06-20:
The main challenges with this issue are:
* Mapping from default to concrete color is spread out in various places
* Default alpha vs. default color is not well-defined
* Color4f inherits from Vector4f, making it hard to locate all places in the codebase where colors are used
* It's possible to do weird stuff, like default red, but concrete green and blue. This should probably not be allowed. Perhaps separate color and alpha into two fields and treat them separately?

jordanbrown0, 2025-10-14:
Here's some examples to play with.  Any answer to this question should specify what each of them will do, and why.

```
translate([0,0,0])  color(c="blue", alpha=0.5)       cube(5);
translate([10,0,0]) color(c="blue")                  cube(5);
translate([20,0,0]) color(alpha=0.5)                 cube(5);
translate([30,0,0]) color(c="blue") color(alpha=0.5) cube(5);
translate([40,0,0]) color(alpha=0.5) color(c="blue") cube(5);
translate([50,0,0]) color()                          cube(5);
```

My tentative answer would be that both `c` and `alpha` are optional.  If `c` is supplied, it sets the color.  If `alpha` is supplied, it sets the alpha.  (And if `c` includes an alpha component, it's equivalent to specifying `alpha`.  Specifying alpha both in `c` and explicitly should be a warning.)

That's pretty easy to understand.  It's not 100% compatible, but I bet that it's compatible with the vast majority (if not all) of existing practical uses.

The interaction with # and %, both above and below these color specifications, needs to be specified.  My inclination (mostly from an ease-of-explanation standpoint) is that # and % live in an alternate plane of reality, and their objects are not subject to these color specifications.

kintel, 2025-10-15:
> > 
> > I didn't really get that far. I'm tempted to say that should not be allowed.
> 
> Sorry, I meant: to close this issue, do you want to answer all open questions regarding the `color` module, or just the questions regarding the `color` module when c==undef? It sounds like the former.

Ideally we should answer all open questions. If not, it's perfectly fine to open a new issue with any stragglers, then we can assess the severity of that separately.

kintel, 2025-10-15:
> CGAL render does not show any colors. Manifold render does try for alpha and colors, but it's full of bugs i.e.:
> But, I suppose you need it to work in order to export color 3mf, right? Is that a blocker for cutting a release?

CGAL render doesn't support colors - that's expected and WAI
Transparency rendering has issues. That's an orthogonal issue: #5319
The OpenSCAD rendering issues don't influence the export capabilities.

coryrc, 2025-10-15:
> That's an orthogonal issue: [#5319](https://github.com/openscad/openscad/issues/5319) The OpenSCAD rendering issues don't influence the export capabilities.

Ah, thanks.

Okay, so I think I have a new rule regarding color and alpha.

Preview and Render view are expected to show transparency. Some export types will support transparency (i.e. PNG) and some will not (i.e. 3MF) and some will not support any color (STL). Objects with the same color but different alpha will be considered the same color in formats which support color and not transparency. Alpha of zero will still be considered the associated color (when one is present).

Does this sound agreeable?

jordanbrown0, 2025-10-15:
The default color comes from Edit/Preferences/3D View/Color scheme.  (Which in turn mostly comes from a collection of JSON files; see https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Paths .)

Out-of-range color values trigger warnings (which for many people means that they abort the run) and are undefined behavior.  I don't feel any need for their behavior to be defined.  (Subtlety:  over-large values are not exactly truncated, at least on my system; they are over-bright and will lighten otherwise shadowed faces.  If you bump that 1.5 up to 5 or so, you'll find that all shading disappears.  But I wouldn't be surprised if that's dependent on the details of the graphics pipeline.)

coryrc, 2025-10-15:
> The default color comes from Edit/Preferences/3D View/Color scheme. (Which in turn mostly comes from a collection of JSON files; see https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Paths .)

(I was being slightly obtuse because I'm not trying to specify everything, but pull out consensus)

Is everyone in agreement that "default color" means `CGAL_FACE_FRONT_COLOR` of the currently-selected theme? (and not `OPENCSG_FACE_FRONT_COLOR` or any value that can be modified by the script?)

> Out-of-range color values trigger warnings (which for many people means that they abort the run) and are undefined behavior. I don't feel any need for their behavior to be defined. (Subtlety: [...]

I'm fine with the definition being "out-of-bounds values give a warning and the behaviour is otherwise undefined" so long as that's agreeable.

coryrc, 2025-10-15:
I think the current status is:

* Preview and Manifold Render view shall show transparency. Some export types will support transparency (i.e. PNG) and some will not (i.e. 3MF) and some will not support any color (STL). Objects with the same color but different alpha will be considered the same color in formats which support color and not transparency. Alpha of zero will still be considered the associated color (when one is present).
* "Default color" means CGAL_FACE_FRONT_COLOR of the currently-selected theme.
* Out-of-bounds values shall give a warning and the behaviour is otherwise undefined.
* Modifier color change shall not affect rendering or export.

Here's some open questions to drive closure:

* If an object is not colored, is it considered to be the default color?
* If not, can an object have transparency and not a color?
* Should we also clamp out-of-bounds value to the valid range to avoid people relying on undefined behaviour? (preserving the warning) (I think this would make people not reliant on e.g. 5.0 making things really bright)
* Currently `color(alpha=x)` only changes the transparency of any *uncolored* descendants (and only if there's no higher `color()` setting color). Jordan proposed it should modify the transparency of all children. Thoughts?

jordanbrown0, 2025-10-15:
I would use the names from the color scheme files, since those are the most user-facing interface.

They seem to have three that are relevant:  opencsg-face-front, cgal-face-front, and cgal-face-2d.  In the long term I think the first two should be merged (and the implementation-specific "opencsg"/"cgal" part eliminated).  Part of the reason for merging them would be that eventually we might be able to get rid of preview as a separate concept.  I don't have a real feeling on whether 2D objects should have a separable default color.  Until we merge those two, preview should use opencsg-face-front and render should use cgal-face-front.  For most of the existing schemes they are the same, but for clearsky, tomorrow, and tomorrow-night they are different.

If an object is not colored, then I would say that it by definition has the default color.  Anything else is too weird.

I would say that `c=` **sets** the color of the subtree, and that `alpha=` **sets** the transparency of the subtree.  (And that specifying a fourth component to `c=` is equivalent to setting `alpha=`.)  I would not attempt any sort of addition or multiplication.  That's a slightly incompatible change, but I think it takes a difficult-to-explain behavior and turns it into an explainable behavior.

With respect to clamping out-of-bounds values... if we were starting from zero, with the perspective of hindsight, I'd say yes.  However, we have at least one user who is known to be relying on this behavior.  They've been strongly warned that it's at their own risk, so if for some reason it becomes desirable to change the behavior I would do it without hesitation, but changing it without a specific reason seems like breaking them just to break them.

kintel, 2025-10-15:
More nuance:
* There's a default _positive_ color and a default _negative_ color
* When exporting a default-colored object, should we export no color, or the explicit default color? How about objects with multiple default colors?

bitbasher, 2025-10-15:
> More nuance:
> 
>     * There's a default _positive_ color and a default _negative_ color

hunh? what?

>     * When exporting a default-colored object, should we export no color, or the explicit default color? 
export with the default color .. that is part of the specification of the object so .. 

>     * How about objects with multiple default colors?
how does one do this? like, a union of different colored children?

nophead, 2025-10-15:
> or being not explicitly set deemed to be all zeros .. so white?

All zeros is black, not while.

jordanbrown0, 2025-10-16:
Modifiers (# and %) only affect Preview (F5), not Render (F6).  If you need them, press F5.  Mostly you should only need F6 when you're about to export.  (But:  if we ever move toward eliminating preview as a separate concept, and always render, we'll need to rethink this relationship.  Regardless, the modifiers must not affect the shape exported.)

There are **no** OpenSCAD operations that affect only their immediate children.  (Maybe BOSL2 fakes it, but the core has no such thing.)  If you want to control each individual part of your model, that's not hard:  put "color(...)" in front of each primitive, or in front of each subassembly that's to be all the same color.

The negative color brings in a slightly different issue:  for multi-material printing, you need volumetric coloring, not face coloring.  In volumetric coloring there's no such thing as a negative color.  I don't think we're ready to go there, and I don't know exactly what volumetric coloring will look like, but I think we need to keep it in mind.

jordanbrown0, 2025-10-16:
> When exporting a default-colored object, should we export no color, or the explicit default color? How about objects with multiple default colors?

Mild preference for exporting the configured default color, so that the resulting model matches what you see in your OpenSCAD preview.

MichaelAtOz, 2025-10-16:
> For most of the existing schemes they are the same, but for clearsky, tomorrow, and tomorrow-night they are different.

However (in 2025.10.10) both F5 & F6 show the OPENCSG color.
I modified a scheme to have red & blue for OPENCSG & CGAL, only red was painted.
So we would not be changing behaviour if they were actually merged.

MichaelAtOz, 2025-10-16:
> the docs specify that the RGBA elements of a color value should all be in the range 0:1 or emit an error and stop compiling

No the docs do not specify, emit an error and stop. They specify the range.
Current behaviour is to issue a warning and display some ill-defined colour.

When I was learning OpenSCAD along long time ago, I found it was forgiving & easy to learn, as it tended to soft fail.
We have kids in Libraries all over using OpenSCAD to print my [Sheriff's badge](https://www.thingiverse.com/thing:415487), I see from the 1K remixes and comments etc I get.
They should not need to have a programming degree to use it.
But as Jordan said, it is a religious issue.

BTW did anyone notice the `color()` wiki example 2?

Example 2
----
In cases where you want to optionally set a color based on a parameter you can use the following trick:
```
 module myModule(withColors=false) {
    c=withColors?"red":undef;
    color(c) circle(r=10);
 }
```
Setting the colorname to undef keeps the default colors.

bitbasher, 2025-10-16:
> However (in 2025.10.10) both F5 & F6 show the OPENCSG color. I modified a scheme to have red & blue for OPENCSG & CGAL, only red was painted. So we would not be changing behaviour if they were [...]

did you mean : red for OpenCSG and blue for CGAL .. but CGAL was not rendered in blue? assuming your color scheme was defined properly that would be a bug no?

jordanbrown0, 2025-10-16:
>However (in 2025.10.10) both F5 & F6 show the OPENCSG color.
I modified a scheme to have red & blue for OPENCSG & CGAL, only red was painted.
So we would not be changing behaviour if they were actually merged.

You probably tried a single primitive.  Looks like there's a bug, that as with several things a single primitive gets trivial handling.  Try two primitives, so that CGAL/Manifold actually get invoked.  I just confirmed that the two get the two different colors.

(Not that I think that it's **important** that they be separate.  But they are.)

MichaelAtOz, 2025-10-16:
From: Jordan Brown ***@***.*** 
You probably tried a single primitive. Looks like there's a bug, that as with several things a single primitive gets trivial handling. Try two primitives, so that so that CGAL/Manifold actually get invoked. I just confirmed that the two get the two different colors.

 

No I happened (by chance) to have two primitives.
MY 2025.10.10 only shows Red. What did you test with?

 

Just to be sure, I did a new scheme (interesting render of white background btw)

F5

 

F6 (and in case, I tried CGAL and Manifold renders, both show this)

  _____

MichaelAtOz, 2025-10-17:
For some reason GitHub decided to discard the images, I'll edit the above on GitHub.
But for the record, both F5 & F6 were red.

MichaelAtOz, 2025-10-17:
Can anyone tell me what's wrong with the markdown above, why they are not showing inline, like Jordan's post above it?
That is what GH added using it's 'click to add files'.

coryrc, 2025-10-21:
Please close because #6309 and #6310 capture all remaining action items.

commit 6aba8901 (2025-06-20): Fix(export): Handle no-arg color() correctly in 3mf and off exports

Fixes #5966
commit 1023c382 (2025-06-20): Fix(export): Handle no-arg color() correctly in 3mf and off exports

Fixes #5966
commit 3a4246ba (2025-06-20): Fix(eval): Handle no-arg and alpha-only color() correctly

This change ensures that using color() with no arguments, or only an alpha value, is handled correctly during geometry evaluation.

Fixes #5966