openscad/openscad #5977: Escape out of Find puts focus in wrong place

jon-bondy, 2025-06-15:
**Describe the bug**
If you use <ctrl>-F to enter Find mode and then use <esc> to quit out of Find mode, focus is placed in the Console rather than back in the Editor.  This makes no sense: one cannot edit the Console.  Focus should be returned to the Editor.

**To Reproduce**
<ctrl>-F then <esc>

**Expected behavior**
Focus should return to the Editor

**Code reproducing the issue**
N/A

**Screenshots**
N/A

**Environment and Version info (please complete the following information):**
 - OS: Windows 11, fully updated
 - System: N/A
 - OpenSCAD Version: [OpenSCAD](https://www.openscad.org/) version 2024.01.14 (git 158c5b94f) 

**Library & Graphics card information**
N/A

**Additional context**
This bug has been present, and reported, for years.  It is a very simple fix.

kintel, 2025-06-16:
Same behavior on macOS.

kintel, 2025-06-16:
Ah, thx!