openscad/openscad #5988: Build failure - latest cgalutils-tess.cc fails to compile on Ubuntu 20.04.6

smartavionics, 2025-06-21:
**Describe the bug**
Building the current source, I am now seeing a compilation failure.

**To Reproduce**
Check out latest source and build

**Expected behavior**
Compilation succeeds.

**Code reproducing the issue**

**Screenshots**
```
~/3dp/openscad.git/build[master]$ make -k
[  0%] Automatic MOC and UIC for target OpenSCAD
[  2%] Built target OpenSCAD_autogen
[  6%] Built target mimalloc-static
[  7%] Built target Clipper2
[ 13%] Built target manifold
[ 13%] Building CXX object CMakeFiles/OpenSCAD.dir/src/geometry/cgal/cgalutils-tess.cc.o
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:11:71: error: ‘K’ was not declared in this scope
   11 | using Projection = CGAL::Triangulation_2_filtered_projection_traits_3<K>;
      |                                                                       ^
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:11:72: error: template argument 1 is invalid
   11 | using Projection = CGAL::Triangulation_2_filtered_projection_traits_3<K>;
      |                                                                        ^
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:25:65: error: ‘Tess_kernel’ was not declared in this scope
   25 | using Fbb = CGAL::Triangulation_face_base_with_info_2<FaceInfo, Tess_kernel>;
      |                                                                 ^~~~~~~~~~~
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:25:76: error: template argument 2 is invalid
   25 | using Fbb = CGAL::Triangulation_face_base_with_info_2<FaceInfo, Tess_kernel>;
      |                                                                            ^
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:25:76: error: template argument 3 is invalid
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:26:84: error: ‘Projection’ was not declared in this scope
   26 | using Tds = CGAL::Triangulation_data_structure_2<CGAL::Triangulation_vertex_base_2<Projection>, CGAL::Constrained_triangulation_face_base_2<Projection, Fbb>>;
      |                                                                                    ^~~~~~~~~~
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:26:94: error: template argument 1 is invalid
   26 | using Tds = CGAL::Triangulation_data_structure_2<CGAL::Triangulation_vertex_base_2<Projection>, CGAL::Constrained_triangulation_face_base_2<Projection, Fbb>>;
      |                                                                                              ^
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:26:141: error: ‘Projection’ was not declared in this scope
   26 | g Tds = CGAL::Triangulation_data_structure_2<CGAL::Triangulation_vertex_base_2<Projection>, CGAL::Constrained_triangulation_face_base_2<Projection, Fbb>>;
      |                                                                                                                                         ^~~~~~~~~~

/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:26:153: error: ‘Fbb’ was not declared in this scope
   26 | ::Triangulation_data_structure_2<CGAL::Triangulation_vertex_base_2<Projection>, CGAL::Constrained_triangulation_face_base_2<Projection, Fbb>>;
      |                                                                                                                                         ^~~

/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:26:153: error: template argument 1 is invalid
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:26:153: error: template argument 2 is invalid
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:26:156: error: template argument 1 is invalid
   26 | riangulation_data_structure_2<CGAL::Triangulation_vertex_base_2<Projection>, CGAL::Constrained_triangulation_face_base_2<Projection, Fbb>>;
      |                                                                                                                                         ^~

/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:26:156: error: template argument 2 is invalid
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:27:56: error: ‘Projection’ was not declared in this scope
   27 | using CDT = CGAL::Constrained_Delaunay_triangulation_2<Projection, Tds, CGAL::Exact_predicates_tag>;
      |                                                        ^~~~~~~~~~
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:27:68: error: ‘Tds’ was not declared in this scope
   27 | using CDT = CGAL::Constrained_Delaunay_triangulation_2<Projection, Tds, CGAL::Exact_predicates_tag>;
      |                                                                    ^~~
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:27:99: error: template argument 1 is invalid
   27 | using CDT = CGAL::Constrained_Delaunay_triangulation_2<Projection, Tds, CGAL::Exact_predicates_tag>;
      |                                                                                                   ^
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:27:99: error: template argument 2 is invalid
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:30:26: error: variable or field ‘mark_domains’ declared void
   30 | static void mark_domains(CDT& ct,
      |                          ^~~
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:30:26: error: ‘CDT’ was not declared in this scope
/home/markb/3dp/openscad.git/src/geometry/cgal/cgalutils-tess.cc:30:31: error: ‘ct’ was not declared in this scope
   30 | static void mark_domains(CDT& ct,
[... 526 lines cut ...]
GL_NV_blend_square
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_shader_interlock
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```
**Additional context**
Add any other context about the problem here.

kintel, 2025-06-21:
I'm not sure older Ubuntu will keep being able to build OpenSCAD. GitHub recently disabled Ubuntu 20.04 CI runner so we're no longer building it daily.
Some leading questions:
* Are you building from source because you're somehow unable to use our nightly binaries?
* Are you unable to upgrade to a more recent Ubuntu version?
* Is this a recent regression?

You might be on your own, but leaving this open in case someone wants to go digging..

smartavionics, 2025-06-21:
Thanks for the speedy reply. Answers...

1 - I haven't tried using your nightlies before but I just installed the snap and it appears to run OK so I will move to using that.

2 - Probably could upgrade the Ubuntu but it's not on my agenda right now.

3 - Yes, it's a recent regression.

I can understand not wanting to support builds on old platforms so I don't mind if you don't fix this issue.

Please close when it suits you.

t-paul, 2025-06-21:
The issue is actually a not updated code path for CGAL < 5.4. Due to Ubuntu 20.04. being out of standard support, it's not covered by any automatic builds and/or tests anymore.

Anyway, should be fixed by the linked commit.

commit 525e3c43 (2025-06-21): Fix build with CGAL < 5.4, e.g. on Debian/Bullseye with 5.2 (fixes #5988).
commit d4994a14 (2025-06-21): Merge pull request #5989 from openscad/fix-build-debian-bullseye

Fix build with CGAL < 5.4, e.g. on Debian/Bullseye with 5.2 (fixes #5988).